thingsboard/thingsboard-gateway #269: MQTT Connector Unable to upload telemetry data to thingsboard platform

lesliechung88, 2020-04-01:
**Describe the bug**
Clear and concise description of what the bug is.

Unable to upload telemetry data to thingsboard platform, I have tried a lot of formats, I wonder what is the correct format? Is there a document example?

I use the following mqtt format to my mqtt broker：

b.json
```
{
        "deviceName": "devicename-mqtt-gateway",
        "deviceType": "devicetype-mqtt-gateway",
        "attributes": [{"model": "T1000"}],
        "telemetry": [{"temperature": 42.0}]
}
```
use mqtt client:

`cat b.json | mosquitto_pub -h localhost -p 1885 -t  "/temperature-sensors" -u "mqttuser" -P "guqiao" -s`

The device telemetry sheet is shown below, but you cannot see "temperature"

 ```
 
  | 2020-03-30 22:28:18 | defaultBrokerEventsProduced | 0
  | 2020-03-30 22:28:18 | defaultBrokerEventsSent | 0
  | 2020-04-01 15:20:44 | eventsProduced | 0
  | 2020-04-01 15:20:44 | eventsSent | 0
  | 2020-04-01 15:20:44 | mqttbrokerconnectorEventsProduced | 0
```

mqtt.json
```
{
  "broker": {
    "name":"Default Broker",
    "host":"192.168.111.133",
    "port":1885,
    "security": {
      "type": "basic",
      "username": "mqttuser",
      "password": "guqiao"
    }
  },
  "mapping": [
    {
      "topicFilter": "v1/devices/me/attributes",
      "converter": {
        "type": "json",
        "filterExpression": "",
        "deviceNameJsonExpression": "${SerialNumber}",
        "deviceTypeJsonExpression": "${SensorType}",
        "timeout": 60000,
        "attributes": [
          {
            "type": "string",
            "key": "test_key",
            "value": "${SerialNumber}"
          }
        ]
      }
    },
    {
      "topicFilter": "/temperature-sensors/+",
      "converter": {
        "type": "json",
        "filterExpression": "",
        "deviceNameJsonExpression": "${$.serialNumber}",
        "deviceTypeJsonExpression": "${$.sensorType}",
        "timeout": 60000,
        "attributes": [
          {
            "type": "string",
            "key": "model",
            "value": "${$.model}"
          }
        ],
        "timeseries": [
          {
            "type": "double",
            "key": "temperature",
            "value": "${$.temperature}"
          }
        ]
      }
    },
    {
      "topicFilter": "/temperature-sensors/+",
      "converter": {
        "type": "json",
        "filterExpression": "",
        "deviceNameJsonExpression": "${$.sensorId}",
        "deviceTypeJsonExpression": "${$.sensorType}",
        "timeout": 60000,
        "timeseries": [
          {
            "type": "double",
            "key": "humidity",
            "value": "${$.humidity}"
          }
        ]
      }
    },
    {
      "topicFilter": "/custom-sensors/+",
      "converter": {
        "type": "custom",
        "extension": "CustomMqttUplinkConverter",
        "extension-config": {
            "temperatureBytes" : 2,
            "humidityBytes" :  2,
            "batteryLevelBytes" : 1
        }
      }
    }
  ],
  "connectRequests": [
    {
      "topicFilter": "sensors/connect",
      "deviceNameJsonExpression": "${$.SerialNumber}"
    },
    {
      "topicFilter": "sensor/+/connect",
      "deviceNameTopicExpression": "(?<=sensor\/)(.*?)(?=\/connect)"
    }
  ],
  "disconnectRequests": [
    {
      "topicFilter": "sensors/disconnect",
      "deviceNameJsonExpression": "${$.SerialNumber}"
    },
    {
      "topicFilter": "sensor/+/disconnect",
      "deviceNameTopicExpression": "(?<=sensor\/)(.*?)(?=\/disconnect)"
    }
  ],
  "attributeUpdates": [
    {
      "deviceNameFilter": "SmartMeter.*",
      "attributeFilter": "uploadFrequency",
      "topicExpression": "sensor/${deviceName}/${attributeKey}",
      "valueExpression": "{\"${attributeKey}\":\"${attributeValue}\"}"
    }
  ],
  "serverSideRpc": [
    {
      "deviceNameFilter": ".*",
      "methodFilter": "echo",
      "requestTopicExpression": "sensor/${deviceName}/request/${methodName}/${requestId}",
      "responseTopicExpression": "sensor/${deviceName}/response/${methodName}/${requestId}",
      "responseTimeout": 10000,
      "valueExpression": "${params}"
    },
    {
      "deviceNameFilter": ".*",
      "methodFilter": "no-reply",
      "requestTopicExpression": "sensor/${deviceName}/request/${methodName}/${requestId}",
      "valueExpression": "${params}"
    }
  ]
}

```

**Connector name (If bug in the some connector):**
[MQTT Connector]

**Error traceback (If available):**
```
'deviceName'
Traceback (most recent call last):
  File "<input>", line 2, in <module>
KeyError: 'deviceName'
```

**Versions (please complete the following information):**
 - OS: [Ubuntu 18.04]
 - Thingsboard IoT Gateway version 
 - Python version
https://github.com/thingsboard/thingsboard-gateway/releases/latest/download/python3-thingsboard-gateway.deb

zbeacon, 2020-04-01:
Hi @lesliechung88 

Thank you for interest in the ThingsBoard IoT gateway.

To use your configuration, you need to send message like:

```json
{
        "SerialNumber": "devicename-mqtt-gateway",
        "SensorType": "devicetype-mqtt-gateway",
        "model": "T1000",
        "temperature": 42.0
}
```

Try to send message above to your broker, to topic `/temperature-sensors/1`
devicename-mqtt-gateway - This will be interpreted as a device name.
devicetype-mqtt-gateway - This will be interpreted as a device type.
T1000 - This will be interpreted as an attribute **model** of the device.
42.0 - This will be interpreted as a telemetry **temperature** of the device.

To parse your message, you should change your configuration parameters e.g.:
"topicFilter": "/temperature-sensors",
"deviceNameJsonExpression": "${$.sensorId}",
"deviceTypeJsonExpression": "${$.sensorType},
and use the [JSONPath Expressions](https://goessner.net/articles/JsonPath/) to parse message part with telemetry and attributes.

lesliechung88, 2020-04-01:
@zbeacon thank you very much!

device attribute is ok,successful upload to tb platform:

`2020-04-01 22:26:15 | model | T1003`

**but telemetry still not found "temperature"**

device telemetry sheet echo:
```
2020-04-01 22:28:55 | defaultBrokerEventsProduced | 6
-- | -- | --
  | 2020-04-01 22:28:55 | defaultBrokerEventsSent | 6
  | 2020-04-01 22:29:56 | eventsProduced | 0
  | 2020-04-01 22:29:56 | eventsSent | 0
  | 2020-04-01 22:29:56 | mqttbrokerconnectorEventsProduced | 0
```

my mqtt pub mock test command:
`cat a.json | mosquitto_pub -h localhost -p 1885 -t  "/temperature-sensors/2" -u "mqttuser" -P "guqiao" -s
`
a.json is :
```
{
"SerialNumber": "devicename-mqtt-gateway",
"SensorType": "devicetype-mqtt-gateway",
"model": "T1003",
"temperature": 42.0
}
```

mqtt.json is：

```
{
  "broker": {
    "name":"Default Broker",
    "host":"192.168.111.133",
    "port":1885,
    "security": {
      "type": "basic",
      "username": "mqttuser",
      "password": "guqiao"
    }
  },
  "mapping": [
    {
      "topicFilter": "v1/devices/me/attributes",
      "converter": {
        "type": "json",
        "filterExpression": "",
        "deviceNameJsonExpression": "${SerialNumber}",
        "deviceTypeJsonExpression": "${SensorType}",
        "timeout": 60000,
        "attributes": [
          {
            "type": "string",
            "key": "test_key",
            "value": "${SerialNumber}"
          }
        ]
      }
    },
    {
      "topicFilter": "/temperature-sensors/+",
      "converter": {
        "type": "json",
        "filterExpression": "",
        "deviceNameJsonExpression": "${SerialNumber}",
        "deviceTypeJsonExpression": "${SensorType}",
        "timeout": 60000,
        "attributes": [
          {
            "type": "string",
            "key": "model",
            "value": "${model}"
          }
        ],
        "timeseries": [
          {
            "type": "double",
            "key": "temperature",
            "value": "${temperature}"
          }
        ]
      }
    },
    {
      "topicFilter": "/custom-sensors/+",
      "converter": {
        "type": "custom",
        "extension": "CustomMqttUplinkConverter",
        "extension-config": {
            "temperatureBytes" : 2,
            "humidityBytes" :  2,
            "batteryLevelBytes" : 1
        }
      }
    }
  ],
  "connectRequests": [
    {
      "topicFilter": "sensors/connect",
      "deviceNameJsonExpression": "${$.SerialNumber}"
    },
    {
      "topicFilter": "sensor/+/connect",
      "deviceNameTopicExpression": "(?<=sensor\/)(.*?)(?=\/connect)"
    }
  ],
  "disconnectRequests": [
    {
      "topicFilter": "sensors/disconnect",
      "deviceNameJsonExpression": "${$.SerialNumber}"
    },
    {
      "topicFilter": "sensor/+/disconnect",
      "deviceNameTopicExpression": "(?<=sensor\/)(.*?)(?=\/disconnect)"
    }
  ],
  "attributeUpdates": [
    {
      "deviceNameFilter": "SmartMeter.*",
      "attributeFilter": "uploadFrequency",
      "topicExpression": "sensor/${deviceName}/${attributeKey}",
      "valueExpression": "{\"${attributeKey}\":\"${attributeValue}\"}"
    }
  ],
  "serverSideRpc": [
    {
      "deviceNameFilter": ".*",
      "methodFilter": "echo",
      "requestTopicExpression": "sensor/${deviceName}/request/${methodName}/${requestId}",
      "responseTopicExpression": "sensor/${deviceName}/response/${methodName}/${requestId}",
      "responseTimeout": 10000,
      "valueExpression": "${params}"
    },
    {
      "deviceNameFilter": ".*",
      "methodFilter": "no-reply",
      "requestTopicExpression": "sensor/${deviceName}/request/${methodName}/${requestId}",
      "valueExpression": "${params}"
    }
  ]
}
```

closed 2020-04-03 by zbeacon

deepthi30192, 2020-12-21:
I have created a Thingsboard Gateway and a device thru it. When I try to send telemetry and attributes, I get the following error:

ERROR - [tb_utility.py] - tb_utility - 63 - No telemetry and attributes in data: {"deviceName": "SN-001", "deviceType": "Thermometer", "attributes": [], "telemetry": []}"
ERROR - [tb_gateway_service.py] - tb_gateway_service - 322 - Data from MQTT Broker Connector connector is invalid."
The command I am giving is :

mosquitto_pub -h localhost -p 1883 -t "/sensor/SN-001/data" -u "user" -P "password" -f "b.json"

The json is:

{ "deviceName": "SN-001", "deviceType" : "Thermometer",

"attributes": [
    {
      "type": "string",
      "key": "model",
      "value": "T1000"
    }
],
"telemetry": [
    {
      "type": "double",
      "key": "temperature",
      "value": 42
    }
]
}

Where am I going wrong with the message format?

saintofinternet, 2021-06-10:
### the following code is not able to update the Telemetry data on Thingsboard platform. 

i am running Thingsboard on a RPI and the data is supposed to come from a ESP8266 Wemos D1 Mini board connected to a capacitive moisture sensor. 

***********************************CODE START***************************************

#include <Arduino.h>
#include "ESP8266WiFi.h" // Enables the ESP8266 to connect to the local network (via WiFi)
#include "PubSubClient.h" // Allows us to connect to, and publish to the MQTT broker

#define SensorPin A0 

// WiFi
const char* ssid = "Hoola";
const char* wifi_password = "grandmaster";

// MQTT
// Make sure to update this for your own MQTT Broker!
const char* mqtt_server = "10.10.1.110";
const char* plant_topic = "v1/devices/me/telemetry";
//const char* plant_topic = "plant";
const char* mqtt_username = "soil1";
const char* mqtt_password = "soil1";
// The client id identifies the ESP8266 device. Think of it a bit like a hostname (Or just a name, like Greg).
const char* clientID = "phatak";

// Initialise the WiFi and MQTT Client objects
WiFiClient wifiClient;
PubSubClient client(mqtt_server, 1883, wifiClient); // 1883 is the listener port for the Broker

void connect_MQTT(){
  Serial.print("Connecting to ");
  Serial.println(ssid);

  // Connect to the WiFi
  WiFi.begin(ssid, wifi_password);

  // Wait until the connection has been confirmed before continuing
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  // Debugging - Output the IP Address of the ESP8266
  Serial.println("WiFi connected");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  // Connect to MQTT Broker
  // client.connect returns a boolean value to let us know if the connection was successful.
  // If the connection is failing, make sure you are using the correct MQTT Username and Password (Setup Earlier in the Instructable)
  if (client.connect(clientID, mqtt_username, mqtt_password)) {
    Serial.println("Connected to MQTT Broker!");
  }
  else {
    Serial.println("Connection to MQTT Broker failed...");
  }
}

void setup() { 
  Serial.begin(9600);
  connect_MQTT();
  Serial.setTimeout(2000);

  float sensorValue = analogRead(SensorPin);
  Serial.println(sensorValue);

  // PUBLISH to the MQTT Broker
  if (client.publish(plant_topic, String(sensorValue).c_str())) {
    Serial.println("Moisture sent!");
    Serial.println(plant_topic);
  }
  // Again, client.publish will return a boolean value depending on whether it succeded or not.
  // If the message failed to send, we will try again, as the connection may have broken.
  else {
    Serial.println("Moisture failed to send. Reconnecting to MQTT Broker and trying again");
    client.connect(clientID, mqtt_username, mqtt_password);
    delay(10); // This delay ensures that client.publish doesn't clash with the client.connect call
    client.publish(plant_topic, String(sensorValue).c_str());
  }
  
  delay(1000);
  //ESP.deepSleep(30e6);
  ESP.deepSleep(0.2*60e6);
}

void loop() {
} 
***********************************CODE END***************************************

### Following is the output I get on the terminal but no data reaches the Thingsboard RPI server...

--- Available ports:
---  1: COM3                 'Sierra Wireless EM7355 - Gobi(TM) 5000 HS-USB Diagnostics 901F (COM3)'
---  2: COM4                 'Sierra Wireless EM7355 - Gobi(TM) 5000 HS-USB NMEA 901F (COM4)'
---  3: COM5                 'Sierra Wireless EM7355 - Gobi(TM) 5000 HS-USB Modem 901F'
---  4: COM11                'USB-SERIAL CH340 (COM11)'
--- Enter port index or full name: 4
--- Miniterm on COM11  9600,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
�en��,�$�Y��4$lllx<�`hConnecting to Hoola
.......WiFi connected
IP address: 10.10.1.107
Connected to MQTT Broker!
758.00
Moisture sent!
v1/devices/me/telemetry