thingsboard/thingsboard-gateway #270: [HELP] Dockerized Thingsboard + Thingsboard Gateway -> Nothing is published and/or created

tmontanaro, 2020-04-01:
**Describe the issue**
I have Thingsboard (in Docker) configured with MQTT port 1883.
Then Thingsboard gateway (in Docker) configured to communicate with Thingsboard and to futher listen on an external MQTT broker set on port 1885 (you can see the configuration files at the end of the message)
The MQTT connector is set to listen for new data on topic '/topicprova/prova/'.

I expect that everytime a new data is published on that topic:
a) a new message is seen on the log
b) a new device is created on Thingsboard.
However, if I try to publish on that topic nothing happen

**Configuration** (Attach your configuration file)
1. tb_gateway.yaml
   ```
   thingsboard:
     host: 172.17.0.4
     port: 1883
     security:
       accessToken: <tokenOfTheGatewayDevice>
   storage:
     type: memory
     read_records_count: 10
     max_records_count: 1000
   connectors:
     -
       name: MQTT Broker Connector
       type: mqtt
       configuration: mqtt.json
   #
   #  -
   #    name: Modbus Connector
   #    type: modbus
   #    configuration: modbus.json
   #
   #  -
   #    name: Modbus Connector
   #    type: modbus
   #    configuration: modbus_serial.json
   #
   #  -
   #    name: BLE Connector
   #    type: ble
   #    configuration: ble.json
   #
   #  -
   #    name: OPC-UA Connector
   #    type: opcua
   #    configuration: opcua.json
   #
   #  -
   #    name: Custom Serial Connector
   #    type: serial
   #    configuration: custom_serial.json
   #    class: CustomSerialConnector
   ```
2. mqtt.json
   ```
    {
      "broker": {
        "name":"Default Broker",
        "host":"172.17.0.1",
        "port":1885,
        "security": {
          "type": "anonymous"
        }
      },
      "mapping": [
        {
          "topicFilter": "/topicprova/prova/",
          "converter": {
            "type": "json",
            "deviceNameJsonExpression": "${entityId}",
            "deviceTypeJsonExpression": "${entityType}",
            "timeout": 60000,
            "attributes": [
              {
                "type": "string",
                "key": "receivingEntityId",
                "value": "${id}"
              }
            ],
            "timeseries": [
              {
                "type": "double",
                "key": "lastUpdate",
                "value": "${lastUpdate}"
              }
            ]
          }
        }
      ],
      "connectRequests": [
        {
          "topicFilter": "sensors/connect",
          "deviceNameJsonExpression": "${$.SerialNumber}"
        },
        {
          "topicFilter": "sensor/+/connect",
          "deviceNameTopicExpression": "(?<=sensor\/)(.*?)(?=\/connect)"
        }
      ],
      "disconnectRequests": [
        {
          "topicFilter": "sensors/disconnect",
          "deviceNameJsonExpression": "${$.SerialNumber}"
        },
        {
          "topicFilter": "sensor/+/disconnect",
          "deviceNameTopicExpression": "(?<=sensor\/)(.*?)(?=\/disconnect)"
        }
      ],
      "attributeUpdates": [
        {
          "deviceNameFilter": "SmartMeter.*",
          "attributeFilter": "uploadFrequency",
          "topicExpression": "sensor/${deviceName}/${attributeKey}",
          "valueExpression": "{\"${attributeKey}\":\"${attributeValue}\"}"
        }
      ],
      "serverSideRpc": [
        {
          "deviceNameFilter": ".*",
          "methodFilter": "echo",
          "requestTopicExpression": "sensor/${deviceName}/request/${methodName}/${requestId}",
          "responseTopicExpression": "sensor/${deviceName}/response/${methodName}/${requestId}",
          "responseTimeout": 10000,
          "valueExpression": "${params}"
        },
        {
          "deviceNameFilter": ".*",
          "methodFilter": "no-reply",
          "requestTopicExpression": "sensor/${deviceName}/request/${methodName}/${requestId}",
          "valueExpression": "${params}"
        }
      ]
    }
   ```

**How I tried to send data**
`mosquitto_pub -d -h 192.168.1.215 -p 1885 -t "/topicprova/prova/" -m '{"serialNumber":"Teo", "sensorType": "Farm","id":"Teo", "lastUpdate":1585733355}'`

**Connector name (If you need help with some connector/converter):**
MQTT Connector

**Error traceback (If it was raised):**
There is no error

**Versions (please complete the following information):**
 - OS: Ubuntu 18.04
 - Thingsboard version -> unknown, the one installed with `sudo docker run -it -p 9090:9090 -p 1883:1883 -p 5683:5683/udp -d -v ~/.mytb-data:/data -v ~/.mytb-logs:/var/log/thingsboard --name mytb --restart always thingsboard/tb-postgres`
 - Thingsboard IoT Gateway version -> unknown, the one installed with `sudo docker run -it -v ~/.tb-gateway/logs:/var/log/thingsboard-gateway -v ~/.tb-gateway/extensions:/var/lib/thingsboard_gateway/extensions -v ~/.tb-gateway/config:/etc/thingsboard-gateway/config --name tb-gateway -d --restart always thingsboard/tb-gateway`

tmontanaro, 2020-04-01:
##How to reproduce the use case##

1. Install the Thingsboard Docker container
   ```
   sudo docker run -d -it -p 9090:9090 -p 1883:1883 -p 5683:5683/udp -v ~/.mytb-data:/data -v ~/.mytb-logs:/var/log/thingsboard --name mytb --restart always thingsboard/tb-postgres
   ```
2. Install the Thingsboard Gateway Docker container
   ```
   sudo docker run -d -it -v ~/.tb-gateway/logs:/var/log/thingsboard-gateway -v ~/.tb-gateway/extensions:/var/lib/thingsboard_gateway/extensions -v ~/.tb-gateway/config:/etc/thingsboard-gateway/config --name tb-gateway --restart always thingsboard/tb-gateway
   ```
3. Stop the Thingsboard Gateway
   ```
   sudo docker container ls
   <GET THE ID of the container>
   sudo docker stop <container-ID>
   ```
4. Modify the parameters to link the gateway to Thingsboard
   * Open a browser and open the Thingsboard interface
      `http://192.168.1.215:9090`
   * Login with the "Tenant Administrator user"
   * Enter the "Device" section
   * Create a new Device and put a tick in the "Is gateway" option
   * Copy the generated Token Id
5. Come back on your terminal (the one used to install the containers)
6. Modify the file `~/.tb-gateway/config/tb_gateway.yaml` (172.17.0.3 is the address of the Thingsboard container and the accessToken is the one copied at point 4)
   ```
   thingsboard:
     host: 172.17.0.3
     port: 1883
     security:
       accessToken: 6MEdYnniARHDx5TrIhV0
   storage:
     type: memory
     read_records_count: 10
     max_records_count: 1000
   connectors:
   
     -
       name: MQTT Broker Connector
       type: mqtt
       configuration: mqtt.json
   #
   #  -
   #    name: Modbus Connector
   #    type: modbus
   #    configuration: modbus.json
   #
   #  -
   #    name: Modbus Connector
   #    type: modbus
   #    configuration: modbus_serial.json
   #
   #  -
   #    name: BLE Connector
   #    type: ble
   #    configuration: ble.json
   #
   #  -
   #    name: OPC-UA Connector
   #    type: opcua
   #    configuration: opcua.json
   #
   #  -
   #    name: Custom Serial Connector
   #    type: serial
   #    configuration: custom_serial.json
   #    class: CustomSerialConnector
   ```
7. MOSQUITTO:
   * Install MOSQUITTO
      ```
      sudo apt-get update
      sudo apt-get install mosquitto
      ```
   * Stop Mosquitto
      ```
      sudo systemctl start mosquitto.service
      ```
   * Change default port
      ```
      sudo nano /etc/mosquitto/mosquitto.conf
      ```
   * Add the following line to the file
      ```
      port 1885
      ```
   * Restart Mosquitto
      ```
      sudo systemctl start mosquitto.service
      ```
8. Modify the file `~/.tb-gateway/config/mqtt.json` (172.17.0.1 is the address of the host machine) to connect to the MQTT broker run in host (parent) machine.  
   ```
     "broker": {
       "name":"Default Broker",
       "host":"172.17.0.1",
       "port":1885,
       "security": {
         "type": "anonymous"
       }
     },
   ...
   ```

## Configure Thingsboard Gateway to accept all messages arriving on the MQTT broker with topic "/topicprova/prova/"
* We want to send all data to the Thingsboard Gateway on topic "/topicprova/prova/", so that, then, it creates a new device everytime a new entity-Id arrives.

* Open the file `~/.tb-gateway/config/mqtt.json`
* Add the following topic to the mapping section:
```
   {
      "topicFilter": "/topicprova/prova/",
      "converter": {
        "type": "json",
        "deviceNameJsonExpression": "${entityId}",
        "deviceTypeJsonExpression": "${entityType}",
        "timeout": 60000,
        "attributes": [
          {
            "type": "string",
            "key": "receivedEntityId",
            "value": "${id}"
          }
        ],
        "timeseries": [
          {
            "type": "double",
            "key": "lastUpdate",
            "value": "${lastUpdate}"
          }
        ]
      }
   }
```

zbeacon, 2020-04-01:
Hi @tmontanaro 

Thank you for your interest in ThingsBoard IoT gateway.

Please set logging level in file logs.conf in configuration folder to DEBUG instead of INFO. try again and attach connector.log file, service.log and converter.log files.

tmontanaro, 2020-04-01:
Hello, after the restart (to get the logs) it seems it is working!
Even though I had already done a lot of restart, maybe this was the right time.

Sorry for that!
However, the reported procedure can help anyone else :D

zbeacon, 2020-04-01:
That's nice :)

But it looks like we have an issue with speed of processing (logging increased pauses during processing)...
Anyway, thank you for your time. Let me know if you will have any questions.

tmontanaro, 2020-04-01:
Actually, sometimes, the gateway stops to work as before, but after a restart it restarts to work.
However, another problem seems to be persistent. If I try to push the following message:
```
mosquitto_pub -d -h 192.168.1.215 -p 1885 -t "/topicprova/prova/" -m '{"entityId":"Teo", "entityType": "Farm","id":"Teo", "lastUpdate":1585733355}'
```
only the entityId (pushed, as reported in above mqtt.json, on `deviceNameJsonExpression`) is actually pushed in the new device, but neither the attributes nor the timeseries is taken from the MQTT pushed message (and nor the deviceTypeJsonExpression).
I'm trying to analyse the logs to understand what happens but it seems it is nothing reported in logs.
(I only changed the IP (172.18.0.10 of the docker container, but the previous procedure still describes the procedure I followed)

connector.log
```
''2020-04-01 15:04:24' - INFO - mqtt_connector - 140 - Default Broker connected to 172.18.0.1:1885 - successfully.'
''2020-04-01 15:04:24' - INFO - mqtt_connector - 166 - Connector "Default Broker" subscribe to v1/devices/me/attributes'
''2020-04-01 15:04:24' - INFO - mqtt_connector - 166 - Connector "Default Broker" subscribe to /temperature-sensors/+'
''2020-04-01 15:04:24' - INFO - mqtt_connector - 166 - Connector "Default Broker" subscribe to /temperature-sensors/+'
''2020-04-01 15:04:24' - DEBUG - mqtt_connector - 148 - Custom converter for topic /custom-sensors/+ - found!'
''2020-04-01 15:04:24' - INFO - mqtt_connector - 166 - Connector "Default Broker" subscribe to /custom-sensors/+'
''2020-04-01 15:04:24' - INFO - mqtt_connector - 166 - Connector "Default Broker" subscribe to /iof2020/uc51/data/orion/'
''2020-04-01 15:04:24' - INFO - mqtt_connector - 166 - Connector "Default Broker" subscribe to /topicprova/prova/'
''2020-04-01 15:04:24' - INFO - mqtt_connector - 197 - "Default Broker" subscription success to topic v1/devices/me/attributes, subscription message id = 1'
''2020-04-01 15:04:24' - INFO - mqtt_connector - 197 - "Default Broker" subscription success to topic /temperature-sensors/+, subscription message id = 2'
''2020-04-01 15:04:24' - INFO - mqtt_connector - 197 - "Default Broker" subscription success to topic /temperature-sensors/+, subscription message id = 3'
''2020-04-01 15:04:24' - INFO - mqtt_connector - 197 - "Default Broker" subscription success to topic /custom-sensors/+, subscription message id = 4'
''2020-04-01 15:04:24' - INFO - mqtt_connector - 197 - "Default Broker" subscription success to topic /iof2020/uc51/data/orion/, subscription message id = 5'
''2020-04-01 15:04:24' - INFO - mqtt_connector - 197 - "Default Broker" subscription success to topic /topicprova/prova/, subscription message id = 6'
''2020-04-01 15:04:24' - INFO - mqtt_connector - 197 - "Default Broker" subscription success to topic sensors/connect, subscription message id = 7'
''2020-04-01 15:04:24' - INFO - mqtt_connector - 197 - "Default Broker" subscription success to topic sensor/+/connect, subscription message id = 8'
''2020-04-01 15:04:24' - INFO - mqtt_connector - 197 - "Default Broker" subscription success to topic sensors/disconnect, subscription message id = 9'
''2020-04-01 15:04:24' - INFO - mqtt_connector - 197 - "Default Broker" subscription success to topic sensor/+/disconnect, subscription message id = 10'
```
2. service.log
```
''2020-04-01 15:04:24' - DEBUG - tb_utility - 61 - Load connector for mqtt class name - MqttConnector from /usr/lib/python3/dist-packages/thingsboard_gateway/connecto$
''2020-04-01 15:04:24' - DEBUG - tb_utility - 66 - ModuleSpec(name='MqttConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7f827c836978>, ori$
''2020-04-01 15:04:24' - DEBUG - tb_utility - 72 - <module 'MqttConnector' from '/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/mqtt/mqtt_connector.py'$
''2020-04-01 15:04:24' - DEBUG - tb_gateway_service - 378 - Loaded devices:
 {'Teo': 'Default Broker', 'Teo2': 'Default Broker'}'
''2020-04-01 15:04:24' - INFO - tb_utility - 58 - Extension mqtt - looking for class in /usr/lib/python3/dist-packages/thingsboard_gateway/extensions/mqtt'
''2020-04-01 15:04:24' - DEBUG - tb_utility - 66 - ModuleSpec(name='CustomMqttUplinkConverter', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7f8277$
''2020-04-01 15:04:24' - DEBUG - tb_utility - 72 - <module 'CustomMqttUplinkConverter' from '/usr/lib/python3/dist-packages/thingsboard_gateway/extensions/mqtt/custom$
```
3. converter.log -> it is empty
4. tb_connection.log
```
''2020-04-01 15:04:24' - DEBUG - tb_client - 78 - connecting to ThingsBoard'
''2020-04-01 15:04:24' - DEBUG - tb_device_mqtt - 349 - Subscribed to * with id 1'
''2020-04-01 15:04:24' - DEBUG - tb_gateway_mqtt - 155 - Subscribed to *|* with id 1'
''2020-04-01 15:04:24' - DEBUG - tb_client - 53 - (<paho.mqtt.client.Client object at 0x7f827c836438>, None, 16, "Sending CONNECT (u1, p0, wr0, wq0, wf0, c1, k60) cli$
''2020-04-01 15:04:24' - DEBUG - tb_client - 53 - (<paho.mqtt.client.Client object at 0x7f827c836438>, None, 16, 'Received CONNACK (1, 0)')'
''2020-04-01 15:04:24' - DEBUG - tb_client - 60 - Gateway connected to ThingsBoard'
''2020-04-01 15:04:24' - INFO - tb_device_mqtt - 182 - connection SUCCESS'
''2020-04-01 15:04:24' - DEBUG - tb_client - 53 - (<paho.mqtt.client.Client object at 0x7f827c836438>, None, 16, "Sending SUBSCRIBE (d0) [(b'v1/devices/me/attributes'$
[... cut ...]
''2020-04-01 15:06:27' - DEBUG - tb_client - 53 - (<paho.mqtt.client.Client object at 0x7f827c836438>, None, 16, "Sending PUBLISH (d0, q1, r0, m14), 'b'v1/devices/me/$
''2020-04-01 15:06:27' - DEBUG - tb_client - 53 - (<paho.mqtt.client.Client object at 0x7f827c836438>, None, 16, "Sending PUBLISH (d0, q1, r0, m15), 'b'v1/devices/me/$
''2020-04-01 15:06:27' - DEBUG - tb_client - 53 - (<paho.mqtt.client.Client object at 0x7f827c836438>, None, 16, 'Received PUBACK (Mid: 14)')'
''2020-04-01 15:06:27' - DEBUG - tb_client - 53 - (<paho.mqtt.client.Client object at 0x7f827c836438>, None, 16, 'Received PUBACK (Mid: 15)')'
```

closed 2020-04-02 by tmontanaro