thingsboard/thingsboard-gateway #272: [HELP]CPU usage is too high, sometimes exceeding 100%

lesliechung88, 2020-04-02:
**Describe the issue**

### The CPU usage is too high, sometimes exceeding 100%. What causes it? Can it be optimized?

only run MQTT Connector

 - OS:  Ubuntu 18.04
docker:
thingsboard/tb-gateway :latest                                     32fcac115816        2 months ago        525MB

zbeacon, 2020-04-02:
Hi @lesliechung88 

Thank you for your interest in ThingsBoard IoT gateway and feedback, I will investigate and optimize processing.

zbeacon, 2020-04-02:
Hi @lesliechung88 

I researched and added some improvements, but processing requires CPU to increase speed. When processing data, the gateway can use almost all available resources to process incoming data. After the data has been processed, the processor load will become equal to 1-2 percent.

closed 2020-04-03 by zbeacon

AngeLinuX99, 2021-11-04:
Same problem of huge use of cpu on my docker server. I have 4 thingsboard_gateway instances and CPU is ALWAYS at 100% with 2.8 version. Im going to test 2.9

AngeLinuX99, 2021-11-04:
> Same problem of huge use of cpu on my docker server. I have 4 thingsboard_gateway instances and CPU is ALWAYS at 100% with 2.8 version. Im going to test 2.9

Same CPU problem in version 2.9

abalmos, 2021-11-04:
> Hi @lesliechung88
> 
> I researched and added some improvements, but processing requires CPU to increase speed. When processing data, the gateway can use almost all available resources to process incoming data. After the data has been processed, the processor load will become equal to 1-2 percent.

@zbeacon can we please re-open this issue? The expected behavior you described is not what we are experiencing. I see a near constant 100% CPU usage regardless of input data rate, in fact even when there is none.

ferguscan, 2021-12-26:
Hi,

I am experiencing this issue, and I use Thingsboard IoT Gateway via Docker.  Will the official Docker image be updated to include this fix?  I see the image was last updated two months ago: https://hub.docker.com/r/thingsboard/tb-gateway/tags

Thanks!