thingsboard/thingsboard-gateway #274: [HELP] Dockerized Thingsboard + Thingsboard Gateway -> New device created but no data is published

tmontanaro, 2020-04-02:
**Describe the issue**
I configured Thingsboard and Thingsboard gateway to work in this way:
- new data are pushed on 1 device that is a gateway
- if the device does not exist, it will create it
- push received data on the (new if it did not exist, otherwise, existing) device.
As I understand from the official documentation, this task should be done through the Thingsboard gateway and the MQTT connector.
You can find all the steps followed to create such a scenario [here](https://github.com/thingsboard/thingsboard-gateway/issues/270#issuecomment-607145803), and the configuration in the following.

Then I tried to push new data (without creating any new converter, but by using the default one) through the following instruction:
`mosquitto_pub -d -h 192.168.1.215 -p 1885 -t "/sensor/data" -m '{"serialNumber":"1234", "sensorType":"Farm", "sensorModel":"ciao", "temp":12,"hum":34}'`

But, even though the device is created with the right name (serialNumber) and the right Type (sensorType), no data and/or attributes are inserted inside.

**Am I doing anything wrong?**

**Configuration** (Attach your configuration file)
1. tb_gateway.yaml
```
thingsboard:
  host: 172.18.0.150
  port: 1883
  remoteConfiguration: false
  security:
    accessToken: Q65vseTJreuQ8RVtqTDK
storage:
  type: memory
  read_records_count: 100
  max_records_count: 100000
#  type: file
#  data_folder_path: ./data/
#  max_file_count: 10
#  max_read_records_count: 10
#  max_records_per_file: 10000
connectors:

  -
    name: MQTT Broker Connector
    type: mqtt
    configuration: mqtt.json

#  -
#    name: Request Connector
#    type: request
#    configuration: request.json
#
#  -
#    name: Modbus Connector
#    type: modbus
#    configuration: modbus.json
#
#  -
#    name: Modbus Connector
#    type: modbus
#    configuration: modbus_serial.json
#
#  -
#    name: OPC-UA Connector
#    type: opcua
#    configuration: opcua.json
#
#  -
#    name: BLE Connector
#    type: ble
#    configuration: ble.json
#
#  -
#    name: Custom Serial Connector
#    type: serial
#    configuration: custom_serial.json
#    class: CustomSerialConnector
```

2. mqtt.json
```
{
  "broker": {
    "name":"Default Broker",
    "host":"172.18.0.1",
    "port":1885,
    "security": {
      "type": "anonymous"
    }
  },
  "mapping": [
    {
      "topicFilter": "/sensor/data",
      "converter": {
        "type": "json",
        "deviceNameJsonExpression": "${serialNumber}",
        "deviceTypeJsonExpression": "${sensorType}",
        "timeout": 60000,
        "attributes": [
          {
            "type": "string",
            "key": "model",
            "value": "${sensorModel}"
          }
        ],
        "timeseries": [
          {
            "type": "double",
            "key": "temperature",
            "value": "${temp}"
          },
          {
            "type": "double",
            "key": "humidity",
            "value": "${hum}"
          }
        ]
      }
    },
    {
      "topicFilter": "/sensor/+/data",
      "converter": {
        "type": "json",
        "deviceNameTopicExpression": "(?<=sensor\/)(.*?)(?=\/data)",
        "deviceTypeTopicExpression": "Thermometer",
        "timeout": 60000,
        "attributes": [
          {
            "type": "string",
            "key": "model",
            "value": "${sensorModel}"
          }
        ],
        "timeseries": [
          {
            "type": "double",
            "key": "temperature",
            "value": "${temp}"
          },
          {
            "type": "double",
            "key": "humidity",
            "value": "${hum}"
          }
        ]
      }
    },
    {
      "topicFilter": "/custom/sensors/+",
      "converter": {
        "type": "custom",
        "extension": "CustomMqttUplinkConverter",
        "extension-config": {
            "temperatureBytes" : 2,
            "humidityBytes" :  2,
            "batteryLevelBytes" : 1
        }
      }
    }
  ],
  "connectRequests": [
    {
      "topicFilter": "sensor/connect",
      "deviceNameJsonExpression": "${SerialNumber}"
    },
    {
      "topicFilter": "sensor/+/connect",
      "deviceNameTopicExpression": "(?<=sensor\/)(.*?)(?=\/connect)"
    }
  ],
  "disconnectRequests": [
    {
      "topicFilter": "sensor/disconnect",
      "deviceNameJsonExpression": "${SerialNumber}"
    },
    {
      "topicFilter": "sensor/+/disconnect",
      "deviceNameTopicExpression": "(?<=sensor\/)(.*?)(?=\/disconnect)"
    }
  ],
  "attributeUpdates": [
    {
      "deviceNameFilter": "SmartMeter.*",
      "attributeFilter": "uploadFrequency",
      "topicExpression": "sensor/${deviceName}/${attributeKey}",
      "valueExpression": "{\"${attributeKey}\":\"${attributeValue}\"}"
    }
  ],
  "serverSideRpc": [
    {
      "deviceNameFilter": ".*",
      "methodFilter": "echo",
      "requestTopicExpression": "sensor/${deviceName}/request/${methodName}/${requestId}",
      "responseTopicExpression": "sensor/${deviceName}/response/${methodName}/${requestId}",
      "responseTimeout": 10000,
      "valueExpression": "${params}"
    },
    {
      "deviceNameFilter": ".*",
      "methodFilter": "no-reply",
      "requestTopicExpression": "sensor/${deviceName}/request/${methodName}/${requestId}",
      "valueExpression": "${params}"
    }
  ]
}
```

**Connector name (If you need help with some connector/converter):**
MQTT Connector

**Error traceback (If it was raised):**
No error: the device is created with the right type, but neither attributes nor data are inserted in the device

**Versions (please complete the following information):**
 - OS: Ubuntu 18.04
 - Thingsboard IoT Gateway version -> the latest installed with Docker (re-downloaded 2 hours ago)
 - Thingsboard version -> the latest installed with Docker (re-downloaded 2 hours ago)

zbeacon, 2020-04-02:
Hi @tmontanaro,

Thank you for your feedback, I will try to repeat this issue tomorrow and write to you.

zbeacon, 2020-04-06:
Hi @tmontanaro,

I repeated your installation, but I use 172.18.0.1 instead of 172.18.0.150 and it works correctly.
I tested data send by following command - `mosquitto_pub -d -h localhost -p 1887 -u user -P x -t "/sensor/data" -m '{"serialNumber":"SOME DEVICE", "sensorType":"Farm", "sensorModel":"ciao", "temp":12,"hum":34}'`

The ThingsBoard created device with name "SOME DEVICE" and gots data from message correctly (I have also installed mosquitto broker (dockerized) on port 1887).

Could you try to replace 172.18.0.150 and send data again? 
To check is the gateway connected - you can go to the "ATTRIBUTES" tab of the gateway device, select "Server attributes" from drop list and check attribute value and updated time of attribute with name "active". It will true if the gateway connected successfully.
Please, let me know about results.

tmontanaro, 2020-04-06:
Hi @zbeacon,
thank you very much for your feedback!
At first: yes, the gateway is connected, as you can see from the attached picture

[image]

Then, I tried to push the same data you pushed by using the following command:
`mosquitto_pub -d -h localhost -p 1885 -u user -P x -t "/sensor/data" -m '{"serialNumber":"SOME DEVICE", "sensorType":"Farm", "sensorModel":"ciao", "temp":12,"hum":34}'`  
**(as you can notice, I didn't specify user and password)**  

but in the following image you can see that the device was actually created but without any attribute:
[image]

Thus, the only difference among my and your request to pub data is in the user and password (+ the IP addresses). Where did you take username and password?

tmontanaro, 2020-04-06:
I've also tried to change IP addresses but nothing changed!
Let's summarize again all the procedure with new addresses.
I executed the following commands:
1. install mosquitto
   `sudo apt-get install mosquitto`
2. Verify that mosquitto is working:
   * terminal 1:
     `mosquitto_sub -h localhost -p 1883 -v -t "#"`
   * terminal 2 (look in terminal 1 if you see the message):
     `mosquitto_pub -d -h localhost -p 1883 -t "/topic1" -m 'hello'`
3. Change the mosquitto port:
   * `sudo nano /etc/mosquitto/mosquitto.conf`
   * change port to 1885: `port 1885` in last line
4. Restart mosquitto:
   * `sudo systemctl start mosquitto.service`
5. Install the Thingsboard network in docker
    `sudo docker network create -d bridge thingsboard-network --gateway 172.18.0.254 --ip-range 172.18.0.0/16 --subnet 172.18.0.0/16`
6. Install the Thingsboard container in docker:
    `sudo docker run -d -it -p 9090:9090 -p 1883:1883 -p 5683:5683/udp -v ~/.mytb-data:/data -v ~/.mytb-logs:/var/log/thingsboard --name mytb --restart always --net thingsboard-network --ip 172.18.0.1 thingsboard/tb-postgres`
7. Install the Thingsboard Gateway container in docker:
   `sudo docker run -d -it -v ~/.tb-gateway/logs:/var/log/thingsboard-gateway -v ~/.tb-gateway/extensions:/var/lib/thingsboard_gateway/extensions -v ~/.tb-gateway/config:/etc/thingsboard-gateway/config --name tb-gateway --restart always --net thingsboard-network --ip 172.18.0.2 thingsboard/tb-gateway`
8. Stop the `tb-gateway` container
9. Change tb-gateway settings:
   `sudo nano .tb-gateway/config/tb_gateway.yaml`  
   ```
    thingsboard:
      host: 172.18.0.1
      port: 1883
      remoteConfiguration: false
      security:
        accessToken: Q65vseTJreuQ8RVtqTDK
    storage:
      type: memory
      read_records_count: 100
      max_records_count: 100000
    connectors:
      -
        name: MQTT Broker Connector
        type: mqtt
        configuration: mqtt.json
   ```
10. Change mqtt connector settings:
   `sudo nano .tb-gateway/config/mqtt.json`
   ```
    {
      "broker": {
        "name":"Default Broker",
        "host":"172.18.0.254",
        "port":1885,
        "security": {
          "type": "anonymous"
        }
      },
      "mapping": [
        {
          "topicFilter": "/sensor/data",
          "converter": {
            "type": "json",
            "deviceNameJsonExpression": "${serialNumber}",
            "deviceTypeJsonExpression": "${sensorType}",
            "timeout": 60000,
            "attributes": [
              {
                "type": "string",
                "key": "model",
                "value": "${sensorModel}"
              }
            ],
            "timeseries": [
              {
                "type": "double",
                "key": "temperature",
                "value": "${temp}"
              },
              {
                "type": "double",
                "key": "humidity",
                "value": "${hum}"
              }
            ]
          }
        },
      ...
   ```
11. Restart the `tb-gateway` container.
12. Push a data on the MQTT broker:
   `mosquitto_pub -d -h localhost -p 1885 -u user -P x -t "/sensor/data" -m '{"serialNumber":"SOME DEVICE", "sensorType":"Farm", "sensorModel":"ciao", "temp":12,"hum":34}'`

zbeacon, 2020-04-06:
Change address in mqtt.json to 172.18.0.1 too. There is an error in connector.log file [Connection refused], it means - mqtt client cannot connect to the following address. This is a problem with routing.

tmontanaro, 2020-04-06:
But the error you are looking at is generated before the changes to the file, then it disappears.
Actually, the mqtt connector should be connected to the mosquitto Broker that is on the system, or not?
If so, 172.18.0.254 is the default gateway for the just created docker network and so it connects to the right gateway (running on local machine).
However, I tried to change IP address in mqtt.json to 172.18.0.1 and I obtain the expected error in the log:
```
''2020-04-06 09:41:26' - ERROR - mqtt_connector - 95 - [Errno 111] Connection refused'
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/mqtt/mqtt_connector.py", line 90, in run
    self.__broker.get('port', 1883))
  File "/usr/lib/python3/dist-packages/paho/mqtt/client.py", line 768, in connect
    return self.reconnect()
  File "/usr/lib/python3/dist-packages/paho/mqtt/client.py", line 895, in reconnect
    sock = socket.create_connection((self._host, self._port), source_address=(self._bind_address, 0))
  File "/usr/lib/python3.6/socket.py", line 724, in create_connection
    raise err
  File "/usr/lib/python3.6/socket.py", line 713, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 111] Connection refused
''2020-04-06 09:41:26' - DEBUG - tb_gateway_mqtt - 66 - Service subscription to topic v1/gateway/attributes - successfully completed.'
''2020-04-06 09:41:26' - DEBUG - tb_gateway_mqtt - 66 - Service subscription to topic v1/gateway/attributes/response - successfully completed.'
''2020-04-06 09:41:26' - DEBUG - tb_gateway_mqtt - 66 - Service subscription to topic v1/gateway/rpc - successfully completed.'
''2020-04-06 09:41:26' - DEBUG - tb_gateway_service - 178 - Received data: {}'
''2020-04-06 09:41:26' - DEBUG - tb_gateway_service - 179 - (None,)'
''2020-04-06 09:41:36' - ERROR - mqtt_connector - 95 - [Errno 111] Connection refused'
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/mqtt/mqtt_connector.py", line 90, in run
    self.__broker.get('port', 1883))
  File "/usr/lib/python3/dist-packages/paho/mqtt/client.py", line 768, in connect
    return self.reconnect()
  File "/usr/lib/python3/dist-packages/paho/mqtt/client.py", line 895, in reconnect
    sock = socket.create_connection((self._host, self._port), source_address=(self._bind_address, 0))
  File "/usr/lib/python3.6/socket.py", line 724, in create_connection
    raise err
  File "/usr/lib/python3.6/socket.py", line 713, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 111] Connection refused
''2020-04-06 09:41:46' - ERROR - mqtt_connector - 95 - [Errno 111] Connection refused'
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/mqtt/mqtt_connector.py", line 90, in run
    self.__broker.get('port', 1883))
  File "/usr/lib/python3/dist-packages/paho/mqtt/client.py", line 768, in connect
    return self.reconnect()
  File "/usr/lib/python3/dist-packages/paho/mqtt/client.py", line 895, in reconnect
    sock = socket.create_connection((self._host, self._port), source_address=(self._bind_address, 0))
  File "/usr/lib/python3.6/socket.py", line 724, in create_connection
    raise err
  File "/usr/lib/python3.6/socket.py", line 713, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 111] Connection refused
```

tmontanaro, 2020-04-06:
Finally, I found the problem: it was the Thingsboard platform itself!
I was assailed by a doubt ... what about checking the Thingsboard platform itself?

For other people
To understand if the problem was mine I tried to perform the following steps:
1. get the accessToken of the Tb_gateway device created at step 7 of [my previous comment](https://github.com/thingsboard/thingsboard-gateway/issues/274#issuecomment-609677258)
2. Try to push the following message directly on Thingsboard MQTT API (`91OM3oz3y0dG5AFoxwM9` is the accessToken):
   `-d -h localhost -p 1883 -u 91OM3oz3y0dG5AFoxwM9 -t "v1/gateway/telemetry" -m '{"DeviceA":[{"ts":1483228800000,"values":{"temperature":42,"humidity":80}},{"ts":1483228801000,"values":{"temperature":43,"humidity":82}}],"DeviceB":[{"ts":1483228800000,"values":{"temperature":42,"humidity":80}}]}'`
3. Open the Thingsboard web interface and check if the device was created and if it contains the passed attributes

There I found that the bug was in Thingsboard and not in the Thingsboard Gateway!
Thus, I completely removed all the docker container and the persistent data and I reinstalled everything! And now it works properly!!!

Thank you very much @zbeacon for your patience and sorry if I made you lose time!

closed 2020-04-06 by tmontanaro

zbeacon, 2020-04-06:
@tmontanaro 

Thank you for your interest in ThingsBoard and ThingsBBoard IoT gateway. Anyway we are improved the gateway, by updating the docker image and one more time check it work in some specific environment. 
I will pass information about this bug to ThingsBoard's developers and we will investigate this situation.

Thank you :)

sonvt8, 2021-08-03:
> I've also tried to change IP addresses but nothing changed!
> Let's summarize again all the procedure with new addresses.
> I performed the following actions:
> 
> 1. install mosquitto
>    `sudo apt-get install mosquitto`
> 2. Verify that mosquitto is working:
>    
>    * terminal 1:
>      `mosquitto_sub -h localhost -p 1883 -v -t "#"`
>    * terminal 2 (look in terminal 1 if you see the message):
>      `mosquitto_pub -d -h localhost -p 1883 -t "/topic1" -m 'hello'`
> 3. Change the mosquitto port:
>    
>    * `sudo nano /etc/mosquitto/mosquitto.conf`
>    * change port to 1885: `port 1885` in last line
> 4. Restart mosquitto:
>    
>    * `sudo systemctl start mosquitto.service`
> 5. Install the Thingsboard network in docker
>    `sudo docker network create -d bridge thingsboard-network --gateway 172.18.0.254 --ip-range 172.18.0.0/16 --subnet 172.18.0.0/16`
> 6. Install the Thingsboard container in docker:
>    `sudo docker run -d -it -p 9090:9090 -p 1883:1883 -p 5683:5683/udp -v ~/.mytb-data:/data -v ~/.mytb-logs:/var/log/thingsboard --name mytb --restart always --net thingsboard-network --ip 172.18.0.1 thingsboard/tb-postgres`
> 7. Create a "gateway" device on Thingsboard (check the "is_Gateway tip) and get the access token
> 8. Install the Thingsboard Gateway container in docker:
>    `sudo docker run -d -it -v ~/.tb-gateway/logs:/var/log/thingsboard-gateway -v ~/.tb-gateway/extensions:/var/lib/thingsboard_gateway/extensions -v ~/.tb-gateway/config:/etc/thingsboard-gateway/config --name tb-gateway --restart always --net thingsboard-network --ip 172.18.0.2 thingsboard/tb-gateway`
> 9. Stop the `tb-gateway` container
> 10. Change tb-gateway settings (the accessToken is the one copied at step 7)
>     `sudo nano .tb-gateway/config/tb_gateway.yaml`
> 
> ```
>  thingsboard:
>    host: 172.18.0.1
>    port: 1883
>    remoteConfiguration: false
>    security:
>      accessToken: Q65vseTJreuQ8RVtqTDK
>  storage:
>    type: memory
>    read_records_count: 100
>    max_records_count: 100000
>  connectors:
>    -
>      name: MQTT Broker Connector
>      type: mqtt
>      configuration: mqtt.json
> ```
> 
> 1. Change mqtt connector settings:
>    `sudo nano .tb-gateway/config/mqtt.json`
> 
> ```
>  {
>    "broker": {
>      "name":"Default Broker",
>      "host":"172.18.0.254",
>      "port":1885,
>      "security": {
>        "type": "anonymous"
>      }
>    },
>    "mapping": [
>      {
>        "topicFilter": "/sensor/data",
>        "converter": {
>          "type": "json",
>          "deviceNameJsonExpression": "${serialNumber}",
>          "deviceTypeJsonExpression": "${sensorType}",
>          "timeout": 60000,
>          "attributes": [
>            {
>              "type": "string",
>              "key": "model",
>              "value": "${sensorModel}"
>            }
>          ],
>          "timeseries": [
>            {
>              "type": "double",
>              "key": "temperature",
>              "value": "${temp}"
>            },
>            {
>              "type": "double",
>              "key": "humidity",
>              "value": "${hum}"
>            }
>          ]
>        }
>      },
>    ...
> ```
> 
> 1. Restart the `tb-gateway` container.
> 2. Push a data on the MQTT broker:
>    `mosquitto_pub -d -h localhost -p 1885 -u user -P x -t "/sensor/data" -m '{"serialNumber":"SOME DEVICE", "sensorType":"Farm", "sensorModel":"ciao", "temp":12,"hum":34}'`
> 
> In attachment you can find all the tb-gateway logs (converter.log and extensions.log are empty):
> 
> [connector.log](https://github.com/thingsboard/thingsboard-gateway/files/4437381/connector.log)
> [service.log](https://github.com/thingsboard/thingsboard-gateway/files/4437382/service.log)
> [storage.log](https://github.com/thingsboard/thingsboard-gateway/files/4437383/storage.log)
> [tb_connection.log](https://github.com/thingsboard/thingsboard-gateway/files/4437384/tb_connection.log)
> 
> (errors before 2020-04-06 09:05:06 should be caused by the fact that the gateway config file was not yet set for the new addresses)

I have difficulty to config gateway to have connection to MQTT Broker (HiveMQ). I am using  your instruction but nothing go well. Could you please give me your contact that you may help me in this case. Hope to see your feedback, thank you!

tmontanaro, 2021-08-03:
Hello, I'm sorry but I've never used this software since then. I could only suggest to change the MQTT broker with another one before giving up.

sonvt8, 2021-08-03:
> Hello, I'm sorry but I've never used this software since then. I could only suggest to change the MQTT broker with another one before giving up.

Thank you brother, What a pity to hear that!!

sonvt8, 2021-08-03:
> Change address in mqtt.json to 172.18.0.1 too. There is an error in connector.log file [Connection refused], it means - mqtt client cannot connect to the following address. This is a problem with routing.

I have a connection to HiveMQ like that:
`    "broker": {
      "name":"cde4d6ad15b0446b874d454412e61396",
      "host":"cde4d6ad15b0446b874d454412e61396.s1.eu.hivemq.cloud",
      "port":8883,
      "security": {
        "type": "basic",
        "username": "my_username",
        "password": "my_password"
      }
`
Here are some logs I collected in:
Connector.log
`""2021-08-03 12:09:38" - DEBUG - [mqtt_connector.py] - mqtt_connector - 308 - "MQTT Broker Connector" was disconnected. (<paho.mqtt.client.Client object at 0x7f4fad357400>, None, 1)"
""2021-08-03 12:11:39" - DEBUG - [mqtt_connector.py] - mqtt_connector - 308 - "MQTT Broker Connector" was disconnected. (<paho.mqtt.client.Client object at 0x7f4fad357400>, None, 1)"`

Service.log
`""2021-08-03 12:01:25" - DEBUG - [tb_utility.py] - tb_utility - 100 - ModuleSpec(name='MqttConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7f4faebeb610>, origin='/usr/local/lib/python3.9/site-packages/thingsboard_gateway-2.6-py3.9.egg/thingsboard_gateway/connectors/mqtt/mqtt_connector.py')"
""2021-08-03 12:01:25" - DEBUG - [tb_utility.py] - tb_utility - 107 - <module 'MqttConnector' from '/usr/local/lib/python3.9/site-packages/thingsboard_gateway-2.6-py3.9.egg/thingsboard_gateway/connectors/mqtt/mqtt_connector.py'>"
""2021-08-03 12:01:25" - INFO - [tb_utility.py] - tb_utility - 111 - Import MqttConnector from /usr/local/lib/python3.9/site-packages/thingsboard_gateway-2.6-py3.9.egg/thingsboard_gateway/connectors/mqtt."
""2021-08-03 12:01:25" - DEBUG - [tb_gateway_service.py] - tb_gateway_service - 664 - Loaded devices:
 {'Temp Sensor1': 'Modbus Default xracj', 'Temp Sensor2': 'Modbus Default xracj'}"
""2021-08-03 12:01:25" - DEBUG - [tb_gateway_service.py] - tb_gateway_service - 362 - Send data Thread has been started successfully."
""2021-08-03 12:01:25" - INFO - [tb_gateway_service.py] - tb_gateway_service - 135 - Gateway started."
""2021-08-03 12:01:25" - DEBUG - [tb_gateway_service.py] - tb_gateway_service - 234 - Received data: {}"`

Converter.log
`""2021-08-03 11:46:36" - DEBUG - [tb_logger.py] - tb_logger - 37 - Added remote handler to log converter"
""2021-08-03 11:53:35" - DEBUG - [tb_logger.py] - tb_logger - 37 - Added remote handler to log converter"
""2021-08-03 12:01:25" - DEBUG - [tb_logger.py] - tb_logger - 37 - Added remote handler to log converter"`

Any idea in this case???