thingsboard/thingsboard-gateway #277: [HELP]How to manipulate values before sending to thingsboard server?

pehedima, 2020-04-03:
I use a normal temperature sensor (MODBUS RTU) to send the temperature to thingsboard server.
The read value from the device is the temperature multiplied by 10. That means, that the value of the register is for example 194 -> 19.4 degrees Celsius. This value should be send to thingsboard server.

What is the easiest way to calculate that before sending?

Here the tb_gateway.yaml and the JSON script to communicate with the device.

[modbus-serial-temperature-config.zip](https://github.com/thingsboard/thingsboard-gateway/files/4425736/modbus-serial-temperature-config.zip)
[tb_gateway.zip](https://github.com/thingsboard/thingsboard-gateway/files/4425737/tb_gateway.zip)

Thanks in advance.

zbeacon, 2020-04-03:
Hi @pehedima,

I added parameters "divider" and "multiplier", specially for this, you can update your gateway to the version from master branch and check it.

To use it, you need just put `"divider": 10` into you config like:

```

{
  "server": {
    "name": "temerature serial",
    "type": "serial",
    "port": "/dev/ttyUSB0",
    "baudrate": 9600,
    "timeout": 35,
    "method": "rtu",
    "devices": [
      {
        "unitId": 1,
        "deviceName": "TEMP-Home-python-balena-serial",
        "attributesPollPeriod": 3600000,
        "timeseriesPollPeriod": 5000,
        "sendDataOnlyOnChange": false,
         "timeseries":[
            {
               "tag":"Temperatur_Buero",
               "type":"long",
               "functionCode":3,
               "address":0,
               "registerCount":1,
               "byteOrder":"BIG",
               "divider": 10
            }
         ]
      }
    ]
  }
}

```

pehedima, 2020-04-03:
Thanks you for your fast reply and the implementation of the two new parameters :)
It's working.

Olny the 'multipier'-Funktion should be changed from: 

```
decoded_data = decoded_data/configuration["multiplier"]
```
to:
```
decoded_data = decoded_data*configuration["multiplier"]
```
Then everthing should work correctly.

zbeacon, 2020-04-03:
Oh, yep thanks.

closed 2020-04-03 by zbeacon

SuperCodesMaker, 2020-12-24:
> 
> 
> Oh, yep thanks.

> Hi @pehedima,
>
> I added parameters "divider" and "multiplier", specially for this, you can update your gateway to the version from master branch and check it.
>
> To use it, you need just put [...]

Hello, I know   "devicer" and "multiplier" can config in *json file,but can you  add "offset" too?
for example ,Calibrate the temperature. source value is 10, offset = -2,end value is 8.