thingsboard/thingsboard-gateway #279: [HELP] Error installing tb-gateway raspberry pi Debian 10

JavierNR, 2020-04-06:
**Describe the issue**
Hello I can't install thingsboard gateway on raspberry pi with Debian 10

The issue raises installing the dependency orjson wich have dependencies:
maturin
 

**Error traceback (If it was raised):**
```
Failed building wheel for maturin
    Running setup.py clean for maturin
  Failed to build maturin
  Installing collected packages: toml, maturin
    Running setup.py install for maturin: started
      Running setup.py install for maturin: finished with status 'error'
      Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-snlptrsl/maturin/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-s96ykfx6/install-record.txt --single-version-externally-managed --prefix /tmp/pip-build-env-npvwyc5d --compile:
      running install
      Traceback (most recent call last):
        File "<string>", line 1, in <module>
        File "/tmp/pip-install-snlptrsl/maturin/setup.py", line 97, in <module>
          zip_safe=False,
        File "/tmp/pip-build-env-55hbfvm1/lib/python3.7/site-packages/setuptools/__init__.py", line 145, in setup
          return distutils.core.setup(**attrs)
        File "/usr/lib/python3.7/distutils/core.py", line 148, in setup
          dist.run_commands()
        File "/usr/lib/python3.7/distutils/dist.py", line 966, in run_commands
          self.run_command(cmd)
        File "/usr/lib/python3.7/distutils/dist.py", line 985, in run_command
          cmd_obj.run()
        File "/tmp/pip-install-snlptrsl/maturin/setup.py", line 54, in run
          "cargo not found in PATH. Please install rust "
      RuntimeError: cargo not found in PATH. Please install rust (https://www.rust-lang.org/tools/install) and try again
```
I have installed rust just in case is needed, but it's still failing. Tried to install from .deb file and from pip.

Some advice?

agtokty, 2020-04-07:
same issue +1
rust installed and `home/pi/.corgo/bin` added to PATH, but did'nt works

zbeacon, 2020-04-07:
Hi @JavierNR , @agtokty 

It is a problem, with orjson, unfortunately it cannot work on arm64. I will remove this package from project and release version without it.

zbeacon, 2020-04-07:
@JavierNR @agtokty 

Please try version 2.2.4.2 .

JavierNR, 2020-04-07:
Hello @zbeacon thanks for the update, but I have tried to install again and it's still trying to install orjson.
I've verified that the downloaded deb package is the right one, also doing dpkg remove + apt autoremove. I don't know if is because of some kind of cache...

zbeacon, 2020-04-07:
@JavierNR 

I regenerate deb package and replace the debian package in the release. Please try it.

JavierNR, 2020-04-07:
@zbeacon 
`Thingsboard Gateway has been installed. Have a nice day :)`
Problem solved! Now going test it, many thanks!

closed 2020-04-07 by JavierNR