thingsboard/thingsboard-gateway #280: [HELP]Why Modbus RPC only write value from config, not write the value from RPC params？

Armed79, 2020-04-07:
**Describe the issue**
Why Modbus RPC only write value from config, not write the value from RPC params？

**Configuration** (Attach your configuration file)
```json
 "rpc": {
          "turnLightOn": {
            "address": 4,
            "bit": 2,
            "value": true
          },
          "turnLightOff": {
            "address": 4,
            "bit": 2,
            "value": false
          },
          "setTestValue": {
            "tag": "Integer",
            "value": 42,
            "functionCode": 16,
            "address": 5,
            "unitId": 1,
            "byteOrder": "BIG",
            "registerCount": 1
          }
        }
```

**Connector name (If you need help with some connector/converter):**
Modbus Connector

**Error traceback (If it was raised):**
```
''2020-04-07 12:34:02' - DEBUG - tb_client - 63 - (<paho.mqtt.client.Client object at 0x76029b90>, None, 16, "Received PUBLISH (d0, q0, r0, m0), 'v1/gateway/rpc', ...  (80 bytes)")'
''2020-04-07 12:34:02' - DEBUG - modbus_connector - 239 - Modbus connector received rpc request for Modbus Connector with content: {'device': 'Flowmeter01', 'data': {'id': 173, 'method': 'setTestValue', 'params': '69'}}'
''2020-04-07 12:34:02' - ERROR - modbus_connector - 251 - 'deviceName''
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 249, in server_side_rpc_handler
    response = self.__function_to_device(rpc_command_config, rpc_command_config["unitId"])
  File "/usr/local/lib/python3.5/dist-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 233, in __function_to_device
    log.debug("To modbus device %s, \n%s", config["deviceName"], config)
KeyError: 'deviceName'
```

zbeacon, 2020-04-07:
Hi @Armed79 

Thank you for your interest in ThingsBoard IoT gateway. 

In current version - yes, but I will fix it in few days. Also I will here after fix.

Armed79, 2020-04-07:
Thanks for your quick update. But still have some issue.
I use knob control to send a rpc to the gateway. The knob control will send the params as a string. That make the downlink converter failed.

```
''2020-04-07 17:03:41' - DEBUG - modbus_connector - 241 - Modbus connector received rpc request for Flowmeter01 with content: {'device': 'Flowmeter01', 'data': {'id': 199, 'method': 'setTestValue', 'params': '59'}}'
''2020-04-07 17:03:41' - ERROR - tb_gateway_service - 429 - required argument is not an integer'
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/thingsboard_gateway/gateway/tb_gateway_service.py", line 397, in _rpc_request_handler
    connector_name.server_side_rpc_handler(content)
  File "/usr/local/lib/python3.5/dist-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 248, in server_side_rpc_handler
    rpc_command_config["payload"] = self.__devices[content["device"]]["downlink_converter"].convert(rpc_command_config, content)
  File "/usr/local/lib/python3.5/dist-packages/thingsboard_gateway/connectors/modbus/bytes_modbus_downlink_converter.py", line 55, in convert
    builder_functions[lower_type](value)
  File "/usr/local/lib/python3.5/dist-packages/pymodbus/payload.py", line 205, in add_8bit_int
    self._payload.append(pack(fstring, value))
struct.error: required argument is not an integer

···

zbeacon, 2020-04-07:
It is not a regular task, for converter. You can try to use "string" as a type parameter in configuration:
```
          "setTestValue": {
            "type": "string",
            "functionCode": 16,
            "address": 5,
            "unitId": 1,
            "byteOrder": "BIG",
            "registerCount": 1
          }
```
Maybe it will work, unfortunately, I don't have ability to check the same situation. If it won't work, you need to develop custom converter.

Armed79, 2020-04-08:
I make a custom widget to send integer params, and it work now. Thank you!

closed 2020-04-08 by Armed79