thingsboard/thingsboard-gateway #286: [HELP]

googlehim, 2020-04-13:
**Describe the issue**
Create description about your issue, and your actions to solve it.
If I use this modbus.json, connector.log shows ok:
[image]
[image]

If I use this modbus.json,   connector.log shows error:
[image]
[image]

**Configuration** (Attach your configuration file)
**Notate: Remove Access token from file if you wanna attach tb_gateway.yaml**

**Connector name (If you need help with some connector/converter):**
[e.g. MQTT Connector]
Modbus Connector

**Error traceback (If it was raised):**
```
'deviceName'
Traceback (most recent call last):
  File "<input>", line 2, in <module>
KeyError: 'deviceName'
```

**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 18.04]
 - Thingsboard IoT Gateway version [e.g. 2.2.4]
 - Python version[e.g. 3.7]
[image]
thingsboard v2.4.3

zbeacon, 2020-04-13:
Hi @googlehim 

It looks like you need to change count of registers (try to use registerCount 2 instead 1).
Also you can update your gateway to version from sources and use one of the following types:

string
bit
bits
8int
8uint
16int
16uint
16float
32int
32uint
32float
64int
64uint
64float

googlehim, 2020-04-13:
Thank you for your reply.
I also tried "type": "long", "registerCount": 1, "functionCode" : 4,  still error.
modbus.json
[image]
remote debug:
[image]

zbeacon, 2020-04-14:
@googlehim 

Ok, it looks like a problem with previous version, please try the following command to update the gateway:

`sudo pip3 install thingsboard-gateway --user thingsboard_gateway --upgrade`

googlehim, 2020-04-14:
@zbeacon 
 After update my gateway to version from sources, problem solved.
 Thank you so much.

closed 2020-04-14 by googlehim

bmesich, 2020-04-16:
@zbeacon - thanks for the fix.  I'm running into this same issue with the official docker image: docker.io/thingsboard/tb-gateway. I confirmed running the pip3 upgrade inside the container fixes the issue.  Rebuilding the container locally with the Dockerfile located in the repository does not pull your fix (and adding the pip install to the Dockerfile does not work either).  Any chance we could git the fix into the official docker image?

zbeacon, 2020-04-16:
@bmesich 

Thanks for your interest in ThingsBoard IoT gateway. Unfortunately, a new docker image will be available in next release. I'm going to publish on Monday.

SERGAMO, 2020-04-22:
The same error was happening to me, even with 2.2.5. After upgrading the pymodbus with `pip3 install pymodbus --upgrade` the decode_16bit_float is gone.