thingsboard/thingsboard-gateway #291: [HELP]Device attribute tag with no Unicode support

chad007, 2020-04-22:
**Describe the issue**
I added some chinese character to the tag of a device attribute in the `modbus.json`, and the gateway 2.2.4.1 throws an error and stops working, while 2.2.1rc actually supports Chinese character.

I can only use ascii character set in the tag of device attributes now, is there a way to get it support Unicode?

**Configuration** (Attach your configuration file)
`tb-gateway/config/modbus.json`
```
{
  "server": {
    "name": "Modbus Default Server",
    "type": "tcp",
    "host": "192.168.56.1",
    "port": 503,
    "timeout": 35,
    "rtuOverTcp": false,
    "devices": [
      {
        "unitId": 1,
        "deviceName": "Temp Sensor",
        "attributesPollPeriod": 5000,
        "timeseriesPollPeriod": 5000,
        "sendDataOnlyOnChange": false,
        "attributes": [
          {
            "byteOrder": "BIG",
            "tag": "test传感器1",
            "type": "long",
            "functionCode": 3,
            "registerCount": 1,
            "address": 0
          }
        ]
      }
    ]
  }
}
```

**Connector name (If you need help with some connector/converter):**
Modbus Connector

**Error traceback (If it was raised):**
```
tb-gateway-py_1    | ''2020-04-22 03:39:54' - ERROR - tb_gateway_service - 238 - 'ascii' codec can't decode byte 0xe4 in position 451: ordinal not in range(128)'
tb-gateway-py_1    | Traceback (most recent call last):
tb-gateway-py_1    |   File "/root/.local/lib/python3.6/site-packages/thingsboard_gateway/gateway/tb_gateway_service.py", line 231, in _load_connectors
tb-gateway-py_1    |     connector_conf = load(conf_file)
tb-gateway-py_1    |   File "/root/.local/lib/python3.6/site-packages/simplejson/__init__.py", line 467, in load
tb-gateway-py_1    |     return loads(fp.read(),
tb-gateway-py_1    |   File "/usr/lib/python3.6/encodings/ascii.py", line 26, in decode
tb-gateway-py_1    |     return codecs.ascii_decode(input, self.errors)[0]
tb-gateway-py_1    | UnicodeDecodeError: 'ascii' codec can't decode byte 0xe4 in position 451: ordinal not in range(128)
```

**Versions (please complete the following information):**
I use this docker image: `thingsboard/tb-gateway:2.2.4.1`

 - OS: Ubuntu 18.04
 - Thingsboard IoT Gateway version 2.2.4.1
 - Python version[e.g. 3.6.9

zbeacon, 2020-04-22:
Hi @chad007 

Please update your gateway to the version 2.2.5 and try again.

chad007, 2020-04-22:
@zbeacon I referred to #286 , use the command below updated the docker image `thingsboard/tb-gateway:2.2.4.1`

```
sudo pip3 install thingsboard-gateway --user thingsboard_gateway --upgrade
```

then I checked the upgraded version:
```
root@fcffdd3aaf3c:/# pip3 show thingsboard-gateway
WARNING: pip is being invoked by an old script wrapper. This will fail in a future version of pip.
Please see https://github.com/pypa/pip/issues/5599 for advice on fixing the underlying issue.
To avoid this problem you can invoke Python with '-m pip' instead of running pip directly.
Name: thingsboard-gateway
Version: 2.2.4.2
Summary: Thingsboard Gateway for IoT devices.
Home-page: https://github.com/thingsboard/thingsboard-gateway
Author: ThingsBoard
Author-email: info@thingsboard.io
License: Apache Software License (Apache Software License 2.0)
Location: /root/.local/lib/python3.6/site-packages
Requires: PyYAML, regex, pip, jsonschema, paho-mqtt, jsonpath-rw, simplejson, python-can, lxml, opcua, pymodbus, pyserial, pytz, pyrsistent, requests, cffi
Required-by: 
```

It was upgraded to 2.2.4.2. How can I upgrade to 2.2.5?

chad007, 2020-04-22:
Sorry, the upgrade I did was a few days a go. I just tried again the upgrade, the tb-gateway is upgraded to 2.2.5:
```
Building wheels for collected packages: thingsboard-gateway
  Building wheel for thingsboard-gateway (setup.py) ... done
  Created wheel for thingsboard-gateway: filename=thingsboard_gateway-2.2.5-py3-none-any.whl size=127032 sha256=24d9f8658f3241a9478b4fc789bb30bc95ac98a4f030a4f5b9de9c1dec3e77ff
  Stored in directory: /root/.cache/pip/wheels/83/95/c2/01347a4d967a820cd4c7eec696abc81d847ca3656a88d0264e
Successfully built thingsboard-gateway
Installing collected packages: bacpypes, thingsboard-gateway
  Attempting uninstall: thingsboard-gateway
    Found existing installation: thingsboard-gateway 2.2.4.2
    Uninstalling thingsboard-gateway-2.2.4.2:
      Successfully uninstalled thingsboard-gateway-2.2.4.2
  WARNING: The script thingsboard-gateway is installed in '/root/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
Successfully installed bacpypes-0.18.0 thingsboard-gateway-2.2.5
root@fcffdd3aaf3c:/# pip3 show thingsboard-gateway
WARNING: pip is being invoked by an old script wrapper. This will fail in a future version of pip.
Please see https://github.com/pypa/pip/issues/5599 for advice on fixing the underlying issue.
To avoid this problem you can invoke Python with '-m pip' instead of running pip directly.
Name: thingsboard-gateway
Version: 2.2.5
Summary: Thingsboard Gateway for IoT devices.
Home-page: https://github.com/thingsboard/thingsboard-gateway
Author: ThingsBoard
Author-email: info@thingsboard.io
License: Apache Software License (Apache Software License 2.0)
Location: /root/.local/lib/python3.6/site-packages
Requires: regex, PyYAML, pytz, opcua, pyrsistent, python-can, requests, pip, jsonschema, paho-mqtt, pymodbus, jsonpath-rw, bacpypes, pyserial, simplejson, cffi, lxml
Required-by:
```

chad007, 2020-04-22:
@zbeacon I tried with 2.2.5, the problem is still there:

```
tb-gateway-py_1    | ''2020-04-22 08:49:51' - ERROR - tb_gateway_service - 237 - Error on loading connector:'
tb-gateway-py_1    | ''2020-04-22 08:49:51' - ERROR - tb_gateway_service - 238 - 'ascii' codec can't decode byte 0xe4 in position 451: ordinal not in range(128)'
tb-gateway-py_1    | Traceback (most recent call last):
tb-gateway-py_1    |   File "/root/.local/lib/python3.6/site-packages/thingsboard_gateway/gateway/tb_gateway_service.py", line 231, in _load_connectors
tb-gateway-py_1    |     connector_conf = load(conf_file)
tb-gateway-py_1    |   File "/root/.local/lib/python3.6/site-packages/simplejson/__init__.py", line 467, in load
tb-gateway-py_1    |     return loads(fp.read(),
tb-gateway-py_1    |   File "/usr/lib/python3.6/encodings/ascii.py", line 26, in decode
tb-gateway-py_1    |     return codecs.ascii_decode(input, self.errors)[0]
tb-gateway-py_1    | UnicodeDecodeError: 'ascii' codec can't decode byte 0xe4 in position 451: ordinal not in range(128)
```

qwyinxiang, 2020-04-23:
@chad007, At tb_gateway_service.py,modifty the function _load_connectors()like below，that will be ok.
[image]

zbeacon, 2020-04-23:
@qwyinxiang 

Thank you for your investigations, I will add this fix into the gateway.
I used "UTF-8" encoding by default and could not to repeat the issue.

chad007, 2020-04-23:
@qwyinxiang Thank you for the temporary solution. It works! Device attributes can be reported from tb-gateway to tb after applying the modification you suggested. But there's another logging error occurred in tb-gateway:

```
tb-gateway-py_1    | ''2020-04-23 05:53:01' - DEBUG - modbus_connector - 233 - With result ReadRegisterResponse (1)'
tb-gateway-py_1    | ''2020-04-23 05:53:01' - DEBUG - modbus_connector - 233 - With result ReadRegisterResponse (1)'
tb-gateway-py_1    | ''2020-04-23 05:53:01' - DEBUG - modbus_connector - 233 - With result ReadRegisterResponse (1)'
tb-gateway-py_1    | ''2020-04-23 05:53:01' - DEBUG - modbus_connector - 233 - With result ReadRegisterResponse (1)'
tb-gateway-py_1    | ''2020-04-23 05:53:01' - DEBUG - modbus_connector - 129 - Checking attributes for device Temp Sensor'
tb-gateway-py_1    | ''2020-04-23 05:53:01' - DEBUG - bytes_modbus_uplink_converter - 53 - Tag: yearmonth Config: {'address': 386, 'bit': 0, 'byteOrder': 'BIG', 'functionCode': 3, 'pollPeriod': 1000, 'registerCount': 1, 'tag': 'yearmonth', 'type': 'long', 'deviceName': 'Temp Sensor'} registers: [5124]'
tb-gateway-py_1    | ''2020-04-23 05:53:01' - DEBUG - bytes_modbus_uplink_converter - 70 - datatype: attributes 	 key: yearmonth 	 value: 5124'
tb-gateway-py_1    | ''2020-04-23 05:53:01' - DEBUG - bytes_modbus_uplink_converter - 53 - Tag: dayhour Config: {'address': 387, 'bit': 0, 'byteOrder': 'BIG', 'functionCode': 3, 'pollPeriod': 1000, 'registerCount': 1, 'tag': 'dayhour', 'type': 'long', 'deviceName': 'Temp Sensor'} registers: [3848]'
tb-gateway-py_1    | ''2020-04-23 05:53:01' - DEBUG - bytes_modbus_uplink_converter - 70 - datatype: attributes 	 key: dayhour 	 value: 3848'
tb-gateway-py_1    | ''2020-04-23 05:53:01' - DEBUG - bytes_modbus_uplink_converter - 53 - Tag: minutesecond Config: {'address': 388, 'bit': 0, 'byteOrder': 'BIG', 'functionCode': 3, 'pollPeriod': 1000, 'registerCount': 1, 'tag': 'minutesecond', 'type': 'long', 'deviceName': 'Temp Sensor'} registers: [9474]'
tb-gateway-py_1    | ''2020-04-23 05:53:01' - DEBUG - bytes_modbus_uplink_converter - 70 - datatype: attributes 	 key: minutesecond 	 value: 9474'
tb-gateway-py_1    | Message: 'datatype: %s \t key: %s \t value: %s'
tb-gateway-py_1    | Arguments: ('attributes', 'test\u4f20\u611f\u56681', '2')
tb-gateway-py_1    | --- Logging error ---
tb-gateway-py_1    | Traceback (most recent call last):
tb-gateway-py_1    |   File "/usr/lib/python3.6/logging/__init__.py", line 996, in emit
tb-gateway-py_1    |     stream.write(msg)
tb-gateway-py_1    | UnicodeEncodeError: 'ascii' codec can't encode characters in position 163-165: ordinal not in range(128)
tb-gateway-py_1    | Call stack:
tb-gateway-py_1    |   File "/usr/lib/python3.6/threading.py", line 884, in _bootstrap
tb-gateway-py_1    |     self._bootstrap_inner()
tb-gateway-py_1    |   File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
tb-gateway-py_1    |     self.run()
tb-gateway-py_1    |   File "/root/.local/lib/python3.6/site-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 69, in run
tb-gateway-py_1    |     self.__process_devices()
tb-gateway-py_1    |   File "/root/.local/lib/python3.6/site-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 132, in __process_devices
tb-gateway-py_1    |     converted_data = self.__devices[device]["converter"].convert(config=None, data=device_responses)
tb-gateway-py_1    |   File "/root/.local/lib/python3.6/site-packages/thingsboard_gateway/connectors/modbus/bytes_modbus_uplink_converter.py", line 72, in convert
tb-gateway-py_1    |     log.debug(self.__result)
tb-gateway-py_1    | Message: {'deviceName': 'Temp Sensor', 'deviceType': 'default', 'telemetry': [], 'attributes': [{'test\u4f20\u611f\u56681': 2}, {'yearmonth': 5124}, {'dayhour': 3848}, {'minutesecond': 9474}]}
tb-gateway-py_1    | Arguments: ()
tb-gateway-py_1    | --- Logging error ---
tb-gateway-py_1    | Traceback (most recent call last):
tb-gateway-py_1    |   File "/usr/lib/python3.6/logging/__init__.py", line 996, in emit
tb-gateway-py_1    |     stream.write(msg)
tb-gateway-py_1    | UnicodeEncodeError: 'ascii' codec can't encode characters in position 163-165: ordinal not in range(128)
tb-gateway-py_1    | Call stack:
tb-gateway-py_1    |   File "/usr/lib/python3.6/threading.py", line 884, in _bootstrap
tb-gateway-py_1    |     self._bootstrap_inner()
tb-gateway-py_1    |   File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
tb-gateway-py_1    |     self.run()
tb-gateway-py_1    |   File "/root/.local/lib/python3.6/site-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 69, in run
tb-gateway-py_1    |     self.__process_devices()
tb-gateway-py_1    |   File "/root/.local/lib/python3.6/site-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 132, in __process_devices
tb-gateway-py_1    |     converted_data = self.__devices[device]["converter"].convert(config=None, data=device_responses)
tb-gateway-py_1    |   File "/root/.local/lib/python3.6/site-packages/thingsboard_gateway/connectors/modbus/bytes_modbus_uplink_converter.py", line 72, in convert
tb-gateway-py_1    |     log.debug(self.__result)
tb-gateway-py_1    | Message: {'deviceName': 'Temp Sensor', 'deviceType': 'default', 'telemetry': [], 'attributes': [{'test\u4f20\u611f\u56681': 2}, {'yearmonth': 5124}, {'dayhour': 3848}, {'minutesecond': 9474}]}
[... cut ...]
tb-gateway-py_1    |   File "/usr/lib/python3.6/logging/__init__.py", line 996, in emit
tb-gateway-py_1    |     stream.write(msg)
tb-gateway-py_1    | UnicodeEncodeError: 'ascii' codec can't encode characters in position 79-81: ordinal not in range(128)
tb-gateway-py_1    | Call stack:
tb-gateway-py_1    |   File "/usr/lib/python3.6/threading.py", line 884, in _bootstrap
tb-gateway-py_1    |     self._bootstrap_inner()
```

It looks like the another ascii encoding problem, with the tb-gateway logging.

zbeacon, 2020-04-23:
This is the same issue but with logging. To solve it you should add string `# -*- coding: utf-8 -*-` into the every .py file at first line. I will also add it into the repository. In you case - you need add it in every file in thingsboard_gateway/connectors/modbus folder and thingsboard_gateway/gateway folder.

zbeacon, 2020-04-23:
@chad007 

Or try to replace strings in the logs.conf file from `args=("./logs/connector.log", "d", 1, 7, )` to `args=("./logs/connector.log", "d", 1, 7, "UTF-8", )` (You need place encoding to the same strings in file)

chad007, 2020-04-23:
@zbeacon Thank you!

I tried the first solution by adding `# -*- coding: utf-8 -*-` into every .py file in thingsboard_gateway/connectors/modbus folder and thingsboard_gateway/gateway folder, at the first line, with no luck, the problem is still there.

And I tried editing logs.conf as you suggested, it throws another exception looks quite the same:

```
tb-gateway-py_1    | ''2020-04-23 06:48:49' - DEBUG - modbus_connector - 233 - With result ReadRegisterResponse (1)'
tb-gateway-py_1    | ''2020-04-23 06:48:49' - DEBUG - modbus_connector - 233 - With result ReadRegisterResponse (1)'
tb-gateway-py_1    | ''2020-04-23 06:48:49' - DEBUG - modbus_connector - 233 - With result ReadRegisterResponse (1)'
tb-gateway-py_1    | ''2020-04-23 06:48:49' - DEBUG - modbus_connector - 233 - With result ReadRegisterResponse (1)'
tb-gateway-py_1    | ''2020-04-23 06:48:49' - DEBUG - modbus_connector - 129 - Checking attributes for device Temp Sensor'
tb-gateway-py_1    | ''2020-04-23 06:48:49' - DEBUG - bytes_modbus_uplink_converter - 53 - Tag: yearmonth Config: {'address': 386, 'bit': 0, 'byteOrder': 'BIG', 'functionCode': 3, 'pollPeriod': 1000, 'registerCount': 1, 'tag': 'yearmonth', 'type': 'long', 'deviceName': 'Temp Sensor'} registers: [5124]'
tb-gateway-py_1    | ''2020-04-23 06:48:49' - DEBUG - bytes_modbus_uplink_converter - 70 - datatype: attributes 	 key: yearmonth 	 value: 5124'
tb-gateway-py_1    | ''2020-04-23 06:48:49' - DEBUG - bytes_modbus_uplink_converter - 53 - Tag: dayhour Config: {'address': 387, 'bit': 0, 'byteOrder': 'BIG', 'functionCode': 3, 'pollPeriod': 1000, 'registerCount': 1, 'tag': 'dayhour', 'type': 'long', 'deviceName': 'Temp Sensor'} registers: [3848]'
tb-gateway-py_1    | ''2020-04-23 06:48:49' - DEBUG - bytes_modbus_uplink_converter - 70 - datatype: attributes 	 key: dayhour 	 value: 3848'
tb-gateway-py_1    | ''2020-04-23 06:48:49' - DEBUG - bytes_modbus_uplink_converter - 53 - Tag: minutesecond Config: {'address': 388, 'bit': 0, 'byteOrder': 'BIG', 'functionCode': 3, 'pollPeriod': 1000, 'registerCount': 1, 'tag': 'minutesecond', 'type': 'long', 'deviceName': 'Temp Sensor'} registers: [9474]'
tb-gateway-py_1    | ''2020-04-23 06:48:49' - DEBUG - bytes_modbus_uplink_converter - 70 - datatype: attributes 	 key: minutesecond 	 value: 9474'
tb-gateway-py_1    | ''2020-04-23 06:48:54' - DEBUG - modbus_connector - 233 - With result ReadRegisterResponse (1)'
tb-gateway-py_1    | ''2020-04-23 06:48:54' - DEBUG - modbus_connector - 233 - With result ReadRegisterResponse (1)'
tb-gateway-py_1    |   File "/root/.local/lib/python3.6/site-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 69, in run
tb-gateway-py_1    |     self.__process_devices()
tb-gateway-py_1    |   File "/root/.local/lib/python3.6/site-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 132, in __process_devices
tb-gateway-py_1    |     converted_data = self.__devices[device]["converter"].convert(config=None, data=device_responses)
tb-gateway-py_1    |   File "/root/.local/lib/python3.6/site-packages/thingsboard_gateway/connectors/modbus/bytes_modbus_uplink_converter.py", line 72, in convert
tb-gateway-py_1    |     log.debug(self.__result)
tb-gateway-py_1    | Message: {'deviceName': 'Temp Sensor', 'deviceType': 'default', 'telemetry': [], 'attributes': [{'test\u4f20\u611f\u56681': 2}, {'yearmonth': 5124}, {'dayhour': 3848}, {'minutesecond': 9474}]}
tb-gateway-py_1    | Arguments: ()
tb-gateway-py_1    | --- Logging error ---
tb-gateway-py_1    | Traceback (most recent call last):
tb-gateway-py_1    |   File "/usr/lib/python3.6/logging/__init__.py", line 996, in emit
tb-gateway-py_1    |     stream.write(msg)
tb-gateway-py_1    | UnicodeEncodeError: 'ascii' codec can't encode characters in position 97-99: ordinal not in range(128)
tb-gateway-py_1    | Call stack:
tb-gateway-py_1    |   File "/usr/lib/python3.6/threading.py", line 884, in _bootstrap
tb-gateway-py_1    |     self._bootstrap_inner()
tb-gateway-py_1    |   File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
tb-gateway-py_1    |     self.run()
tb-gateway-py_1    |   File "/root/.local/lib/python3.6/site-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 69, in run
tb-gateway-py_1    |     self.__process_devices()
tb-gateway-py_1    |   File "/root/.local/lib/python3.6/site-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 131, in __process_devices
tb-gateway-py_1    |     log.debug(device_responses)
tb-gateway-py_1    | Message: {'timeseries': {}, 'attributes': {'test\u4f20\u611f\u56681': {'data_sent': {'byteOrder': 'BIG', 'tag': 'test\u4f20\u611f\u56681', 'type': 'long', 'functionCode': 3, 'registerCount': 1, 'address': 0, 'deviceName': 'Temp Sensor'}, 'input_data': <pymodbus.register_read_message.ReadHoldingRegistersResponse object at 0x7f97a933db70>}, 'yearmonth': {'data_sent': {'address': 386, 'bit': 0, 'byteOrder': 'BIG', 'functionCode': 3, 'pollPeriod': 1000, 'registerCount': 1, 'tag': 'yearmonth', 'type': 'long', 'deviceName': 'Temp Sensor'}, 'input_data': <pymodbus.register_read_message.ReadHoldingRegistersResponse object at 0x7f97a933da58>}, 'dayhour': {'data_sent': {'address': 387, 'bit': 0, 'byteOrder': 'BIG', 'functionCode': 3, 'pollPeriod': 1000, 'registerCount': 1, 'tag': 'dayhour', 'type': 'long', 'deviceName': 'Temp Sensor'}, 'input_data': <pymodbus.register_read_message.ReadHoldingRegistersResponse object at 0x7f97a933dac8>}, 'minut [... cut ...]
tb-gateway-py_1    | ''2020-04-23 06:48:54' - DEBUG - bytes_modbus_uplink_converter - 70 - datatype: attributes 	 key: dayhour 	 value: 3848'
tb-gateway-py_1    | ''2020-04-23 06:48:54' - DEBUG - bytes_modbus_uplink_converter - 53 - Tag: minutesecond Config: {'address': 388, 'bit': 0, 'byteOrder': 'BIG', 'functionCode': 3, 'pollPeriod': 1000, 'registerCount': 1, 'tag': 'minutesecond', 'type': 'long', 'deviceName': 'Temp Sensor'} registers: [9474]'
tb-gateway-py_1    | ''2020-04-23 06:48:54' - DEBUG - bytes_modbus_uplink_converter - 70 - datatype: attributes 	 key: minutesecond 	 value: 9474'
```

qwyinxiang, 2020-04-23:
@chad007 
I used both the mqtt connector and modbus connector. And the ascii code are all displayed well.
"温湿度传感器" are created by the mqtt connector and "audiblealarm001" are created by the modbus connector.
[image]
[image]
[image]

zbeacon, 2020-04-23:
@chad007 

Unfortunately, I cannot repeat the issue you described. But to solve this problem you can try to use docker installation or create a virtual machine and set up there UTF-8 as a default encoding. Also I found another possible solution [here](https://github.com/DocNow/diffengine/issues/24#issuecomment-276110566). Please try and let me know about result.

chad007, 2020-04-23:
@qwyinxiang Yes, device attributes can be reported from tb-gateway to tb after applying the modification you suggested. Thank you!

[image]

But the gateway logs still have some problem with the ascii encoding problem. I will try with the solutions @zbeacon provided later, it's not urgent as long as the device data can be reported from tb-gateway to tb for the moment.

closed 2020-05-07 by zbeacon