thingsboard/thingsboard-gateway #292: Modbus RPC Bug using v2.2.5

qwyinxiang, 2020-04-23:
On the thingsboard, I make a dashboard use the Switch Control widget,
[image]
On the thingsboard-gateway, I configed the modbus.json like below:
[image]

And I can turn on the switch, but can not turn off the switch.
Whether I set the "registerCount" to 1 or 2, the result is same.
The "address" and "functionCode" params are ok.
Is there any problems on the Modbus RPC function?

qwyinxiang, 2020-04-23:
I think that Modbus RPC control only need the "address" and the "functionCode" params, other params are not needed.
When I use the Modbus Poll tools. when I want to write single coil, the needed params are only "unitId","address" and "value"
[image]

qwyinxiang, 2020-04-23:
On the thingsboard, when I clicked the on button,
witer_coil(self, address, value,**kwargs)'s value is a list and value[07] is true, and I can turn on the switch;
[image]
when I clicked the off button,
witer_coil(self, address, value,**kwargs)'s value is a list and value[07] is false, but I cannot turn off the switch.;
[image]

zbeacon, 2020-04-23:
Hi @qwyinxiang 

Thank you for your investigations, it looks like a problem in write function (it used to write coil function 15 instead of 5). I changed it and push the commit into develop/2.4-python branch. 
Please try a version from develop branch and let me know about result.

Thank you.

qwyinxiang, 2020-04-23:
I used the witer_coil function 5, I have fixed this bug before I use the RPC function.
[image]

zbeacon, 2020-04-23:
Ok, got it. Please try develop version, with commit  be797ae

qwyinxiang, 2020-04-23:
I tried the develop version at develop/2.4-python.
The modbus rpc result is still as before. I can only turn on the switch, but I can't turn off the switch.
And there is no error on the terminal.

zbeacon, 2020-04-23:
@qwyinxiang 

Please try a version from develop branch, It has to send the response correctly.

qwyinxiang, 2020-04-24:
I have just tried the develop version  develop/2.4-python, and the result is still as before. I can only turn on the switch, but can't turn off the switch.
Is there any thing wrong with the modbus.conf file. I don't sure whether the "type" and "registerCount" is ok. The value from thingsboard is "True" and "False".
      {
        "unitId": 1,
        "deviceName": "switch222",
        "deviceType": "switch",
        "attributesPollPeriod": 5000,
        "timeseriesPollPeriod": 5000,
        "sendDataOnlyOnChange": true,
        "rpc": [
           {
             "tag": "setValue",
             "address": 1,
             "type": "int",
             "registerCount": 2,
             "functionCode": 5
          }
        ]
      }

zbeacon, 2020-04-24:
@qwyinxiang 

Ok, try to add some configuration to section attributes or timeseries in the configuration file. The gateway sends RPC only if at least one data object was received from a device and sent to the ThingsBoard.

zbeacon, 2020-04-24:
Also please try to configure the widget advanced settings like on screenshot below.
[image]

qwyinxiang, 2020-04-24:
It's not the problem of thingsboard. 
I want to control a relay from thingsboard dashboard using the switch widget.
I think there is no relation to the timeserials or the attributes.
My advanced settings is like this:
[image]
And my modbus.json is like this:
[image]

When I clicked ON on the dashboard, the switch widget turns orange and when I clicked OFF on the dashboard, the switch widget turns gray. 

And when I use the modbus poll tool to turn on off turn off the switch, every things went well.
But when I control the relay form thingsboard, I can only turn on the relay but can't turn off the relay.
When I clicked OFF on thingsboard dashboard, the gateway's write_coil() 's value is all False like below:
<class 'list'>: [False, False, False, False, False, False, False, False, False, False, False, False, False, False, False, False]
[image]

When I clicked ON on thingsboard dashboard, the gateway's write_coil() 's value is 
<class 'list'>: [False, False, False, False, False, False, False, True, False, False, False, False, False, False, False, False]

[image]

zbeacon, 2020-04-24:
Ok, please update to the version from the develop branch and try the following configuration file:

```
{
  "server": {
    "host": "127.0.0.1",
    "method": "socket",
    "port": 5020,
    "timeout": 35,
    "type": "tcp",
    "devices": [
      {
        "attributes": [
          {
            "address": 4,
            "functionCode": 1,
            "registersCount": 1,
            "tag": "state",
            "type": "bit"
          }
        ],
        "attributesPollPeriod": 5000,
        "deviceName": "Temp Sensor",
        "rpc": [
          {
            "address": 4,
            "functionCode": 5,
            "registersCount": 1,
            "tag": "setValue",
            "type": "bit"
          }
        ],
        "sendDataOnlyOnChange": false,
        "timeseriesPollPeriod": 5000,
        "unitId": 1
      }
    ]
  }
}

```

qwyinxiang, 2020-04-26:
I have just tried the develop version, The result is still as before, I started to wonder if there is a problem with my device. But when I use the modbus pool tool, I can easily turn on or turn off my switch.
I retrived the value from "subscribe from attributes" mothod. And the attribute value "state" is always true.
[image]
My config file is like below:
{
  "server": {
    "name": "Modbus Default Server",
    "type": "tcp",
    "host": "192.168.200.172",
    "port": 23,
    "timeout": 35,
    "method": "socket",
    "devices": [
      {
        "unitId": 1,
        "deviceName": "switch222",
        "deviceType": "switch",
        "sendDataOnlyOnChange": false,
        "timeseriesPollPeriod": 5000,
        "attributesPollPeriod": 5000,
        "attributes": [
          {
            "address": 0,
            "functionCode": 1,
            "registersCount": 1,
            "tag": "state",
            "type": "bit"
          }
        ],
        "rpc": [
          {
            "address": 0,
            "functionCode": 5,
            "registersCount": 1,
            "tag": "setValue",
            "type": "bit"
          }
        ]
      }
    ]
  }
}

zbeacon, 2020-04-27:
@qwyinxiang 

Hi, ok. Please try a new version from the develop branch. Also please use the following config to check it:

```
{
  "server": {
    "host": "127.0.0.1",
    "method": "socket",
    "port": 5020,
    "timeout": 35,
    "type": "tcp",
    "devices": [
      {
        "attributes": [
          {
            "address": 4,
            "functionCode": 1,
            "registersCount": 1,
            "tag": "state",
            "type": "bit"
          }
        ],
        "attributesPollPeriod": 500000,
        "deviceName": "Temp Sensor",
        "rpc": [
          {
            "address": 4,
            "functionCode": 5,
            "registersCount": 1,
            "tag": "setValue",
            "type": "bit"
          },
          {
            "address": 4,
            "functionCode": 1,
            "registersCount": 1,
            "tag": "getValue",
            "type": "bit"
          }
        ],
        "sendDataOnlyOnChange": false,
        "timeseriesPollPeriod": 500000,
        "unitId": 1
      }
    ]
  }
}

```

Widget advanced configuration should be default (Status it will check by another RPC call "getValue").

qwyinxiang, 2020-04-28:
I tried the new version of develop/2.4-python and the result is OK.
I clicked the ON, my relay is turned ON, I clicked the OFF , my relay is turned OFF.
Thank you very much.

zbeacon, 2020-04-28:
@qwyinxiang 

Thank you for your interest in ThingsBoard IoT gateway and patience. Please, let me know if you will have any questions.

commit d6dedde2 (2020-04-27): Fix #292 "state" parameter always True
closed 2020-04-28 by zbeacon