thingsboard/thingsboard-gateway #293: [BUG]Building from source failed,because distribution name is not specified in setup.py

qwyinxiang, 2020-04-23:
**Describe the bug**
Clear and concise description of what the bug is.
When I build the source of v2.2.5, I got a error in tb_gateway_service.py at  line self.version = get_distribution('thingsboard_gateway').version
 I found there is no distribution name in setup.py
when I add the name="thingsboard_gateway" at the setup.py file. It works well.
[image]

zbeacon, 2020-04-23:
Hi @qwyinxiang 

The version of the gateway in the file "setup.cfg". It should works properly. But maybe it is caused by wrong encoding... I will investigate this bug. 

Thank you

qwyinxiang, 2020-04-23:
Not at all. I build it on Windows, maybe it works well on Linux.
And I will try on the develop/2.4-python on windows to see if the bug is still exist.

zbeacon, 2020-04-23:
@qwyinxiang  

It is possible. You could noticed we didn't describe the installation in the Windows OS (only docker) in our site. There are a lot of bugs can appears on the Windows. 
I will move all configuration from setup.cfg to setup.py to increase compatibility, but I don't know how many bugs will be appeared on Windows. I will be thankful, if you test the gateway on Windows)

closed 2020-04-24 by zbeacon