thingsboard/thingsboard-gateway #295: [HELP]

ramoun, 2020-05-06:
I've loaded TB + TB IOT GW + configured mqtt per web site example etc. (file copies below)
IOT GW appears to connect to TB, as does mqtt to the external mosquitto broker (logs below).
Using the TB  hello world demo, I can curl data to TB. Using MQTT_Spy & Mosquitto I can publish data to TB directly.
	Topic: v1/devices/me/telemetry
	Data: {"temperature":60, "humidity":60.0, "active": false}

When I send the same data / topic via MQTT_Spy - mosquitto broker - TB GW I cant see the data in the TB demo device/dashboard.  If I look at the TB GW under devices / telemetry data for the gateway Eventsproduced & mqttbrokerconnectorEventsProduced display a value that’s matches the number of mqtt messages sent from the mqtt client.

It feels like it is a config between the TB IOT GW and TB.

Any help or pointers would be greatly appreciated.

- RPI3x
- Linux ThingsBoardDEMO 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linux
- TB V243 
- TB IOT GW mqtt

converter.log & extension.log have zero size

[mqtt.txt](https://github.com/thingsboard/thingsboard-gateway/files/4585112/mqtt.txt)
[tb_gateway.txt](https://github.com/thingsboard/thingsboard-gateway/files/4585113/tb_gateway.txt)
[connector.log.txt](https://github.com/thingsboard/thingsboard-gateway/files/4585115/connector.log.txt)
[service.log.txt](https://github.com/thingsboard/thingsboard-gateway/files/4585117/service.log.txt)
[storage.log.txt](https://github.com/thingsboard/thingsboard-gateway/files/4585118/storage.log.txt)
[tb_connection.log.txt](https://github.com/thingsboard/thingsboard-gateway/files/4585128/tb_connection.log.txt)

zbeacon, 2020-05-06:
Hi @ramoun,

Thank you for your interest in the ThingsBoard IoT gateway.
Could you try to send the message like `{"temp":60, "hum":60.0}` to topic `/sensor/TEST_DEVICE/data` ? - It will create the device with name "TEST_DEVICE" and telemetry keys "temperature" and "humidity" on your ThingsBoard instance. Please read [this article](https://thingsboard.io/docs/iot-gateway/config/mqtt/). It will help you to configure the MQTT connector.

ramoun, 2020-05-06:
Hi,
Thanks for your quick response.   I tried your suggestion direct to TB  GW but it did not create a device.   I did read in one of the tutorials a device should be auto created.
Should this example work with the base TB and TB GW install.

ramoun, 2020-05-06:
I built an additional TB server and loaded TB GW following the all the steps on the [web site](https://thingsboard.io/docs/iot-gateway/install/deb-installation/). 
It did display and error when installing the TB GW package 
`sudo apt install ./python3-thingsboard-gateway.deb -y`

"pi@ThingsBoardIOTGW:~ $ sudo apt install ./python3-thingsboard-gateway.deb -y
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'python3-thingsboard-gateway' instead of './python3-thingsboard-gateway.deb'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 python3-thingsboard-gateway : Depends: python3-pyrsistent but it is not installable
E: Unable to correct problems, you have held broken packages.
pi@ThingsBoardIOTGW:~ $"

zbeacon, 2020-05-06:
This issue caused by new the new ubuntu distributive. Probably, this package was removed from distributor site. Please try to install this package using `sudo pip3 install pyrsistent` and run the gateway installation again. Also please let me know about result.

ramoun, 2020-05-06:
Commands runs and installs ok, but looks like there must be some other dependencies buried in there as I get the same error.

pi@ThingsBoardIOTGW:~ $ sudo apt install ./python3-thingsboard-gateway.deb -y
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'python3-thingsboard-gateway' instead of './python3-thingsboard-gateway.deb'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 python3-thingsboard-gateway : Depends: python3-pyrsistent but it is not installable
E: Unable to correct problems, you have held broken packages.

zbeacon, 2020-05-06:
Ok, got it, I released version 2.3.0.1 - and move this dependency, please try to install it.

ramoun, 2020-05-06:
Great thanks, built a new PI, loaded and configured TB and GW from scratch. Successfully sent  message {"temp":60, "hum":60.0} to topic /sensor/TEST_DEVICE/data, to external broker and a new device was created :-)
Thanks for your help and example / test, I can build of this, hopefully someone else will find this thread useful.

Search words: mqtt, external mqtt broker, thingsboard IOT gateway, mqtt client, MQTT_Spy, Mosquitto

closed 2020-05-06 by ramoun

ramoun, 2020-05-06:
How would you send data to the TB hello world example, from an external mqtt broker. Assuming TB IOT GW is configured and working. Are there an examples I've missed

zbeacon, 2020-05-07:
Hi @ramoun 

There is a topic/message description for default settings in the mapping section of  [configuration guide](https://thingsboard.io/docs/iot-gateway/config/mqtt/#section-mapping). 

Topic: 
`/sensor/data`

Payload:
`{“serialNumber”: “SN-001”, “sensorType”: “Thermometer”, “sensorModel”: “T1000”, “temp”:  42, “hum”: 58}`