thingsboard/thingsboard-gateway #297: [BUG] unable to install or build docker

atolia, 2020-05-06:
**Describe the bug**
git clone https://github.com/thingsboard/thingsboard-gateway.git
cd thingsboard-gateway/docker
docker build -t gw .

**Error traceback (If available):**
```
Step 6/17 : RUN sudo apt-get install -y /python3-thingsboard-gateway.deb
 ---> Running in 613ee1ce1345
Reading package lists...
Building dependency tree...
Reading state information...
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 python3-thingsboard-gateway : Depends: python3-pyrsistent but it is not installable
E: Unable to correct problems, you have held broken packages.
```

**Versions (please complete the following information):**
 - OS: MacOs Catalina
 - Thingsboard IoT Gateway version: 3
 - Python version: 3.7

m-krasikov, 2020-05-06:
The cause is in Base 18.04 version. I've updated to 20.04 in the Dockerfile. Please, pull and try again

atolia, 2020-05-06:
thanks. docker works now. 
do you have ideas how to fix that on raspberry?

zbeacon, 2020-05-06:
@atolia 

Try to use deb package from version 2.3.0.1. We moved the dependency to another installation point. It should work.

atolia, 2020-05-06:
thanks. now it stuck on next step :)
```
12506 pts/2    Ss+    0:00  |                           \_ /usr/bin/dpkg --status-fd 21 --configure --pending
12507 pts/2    S+     0:00  |                               \_ sh -c (test -x /usr/lib/needrestart/dpkg-status && /usr/lib/needrestart/dpkg-status || cat > /dev/null)
12508 pts/2    S+     0:00  |                               |   \_ sh -c (test -x /usr/lib/needrestart/dpkg-status && /usr/lib/needrestart/dpkg-status || cat > /dev/null)
12509 pts/2    S+     0:00  |                               |       \_ /bin/sh /usr/lib/needrestart/dpkg-status
12783 pts/2    S+     0:00  |                               \_ /bin/sh /var/lib/dpkg/info/python3-thingsboard-gateway.postinst configure
12979 pts/2    S+     0:00  |                                   \_ sudo systemctl start thingsboard-gateway
12984 pts/2    S+     0:00  |                                       \_ systemctl start thingsboard-gateway
12985 pts/2    Z+     0:00  |                                           \_ [systemd-tty-ask] <defunct>
```

atolia, 2020-05-06:
actually even 'systemctl start thingsboard-gateway' hangs
```
\_ systemctl start thingsboard-gateway
           \_ /bin/systemd-tty-ask-password-agent --watch
```
and doesn't respond on any keys except CTRL+C

zbeacon, 2020-05-07:
@atolia 

It tries to start the service, could you attach the gateway logs, or looking for any errors in the  journalctl?

viktorkho, 2020-05-07:
@atolia,

IoT Gateway from dockerhub works fine, https://hub.docker.com/r/thingsboard/tb-gateway

May be this will helpfull for you

closed 2020-05-15 by zbeacon