thingsboard/thingsboard-gateway #298: [HELP]

Adrianoghi, 2020-05-06:
**Describe the issue**
I'm using a device connected to the tb-gateway via ModBus and when I run the gateway it works fine until it doesn't recieve a response from the device.
When it happens the gateway continues to retrieve data from device and after the device starts to response again it doesn't upload it to the tb server.
I've noticed on the server that the gateway sends an inactivity event when it doesn't recieve a response from the device, but when it restarts to recieve it doesn't send a activity event and restart upload the data.

Forgive any mistake, my english isn't so good and i'm new in this area.

**Connector name (If you need help with some connector/converter):**
ModBus Connector

**Versions (please complete the following information):**
 - Thingsboard IoT Gateway version 2.2.5
 - Python version 3.7.3

Config and log files:
[modbus_serial.txt](https://github.com/thingsboard/thingsboard-gateway/files/4588974/modbus_serial.txt)
[tb_gateway.txt](https://github.com/thingsboard/thingsboard-gateway/files/4588975/tb_gateway.txt)
[connector.txt](https://github.com/thingsboard/thingsboard-gateway/files/4588986/connector.txt)
[service.txt](https://github.com/thingsboard/thingsboard-gateway/files/4588987/service.txt)
[storage.txt](https://github.com/thingsboard/thingsboard-gateway/files/4588988/storage.txt)
[tb_connection.txt](https://github.com/thingsboard/thingsboard-gateway/files/4588989/tb_connection.txt)

zbeacon, 2020-05-07:
Hi @Adrianoghi,

Thank you for your interest in ThingsBoard IoT gateway. 
Please tell the time, when the latest data was arrived on ThingsBoard.

Adrianoghi, 2020-05-07:
Hi,

These are the lastest telemetrys sent

[image]

zbeacon, 2020-05-15:
Hi @Adrianoghi 

It looks like the gateway lost connection to the Modbus server. Usually, it happens with serial connection, if the bad USB-TTL converter has been used, please try another converter and let me know about results.

closed 2020-05-20 by zbeacon