thingsboard/thingsboard-gateway #299: [BUG] ThingsBoard IoT Gateway stalled on "nan" messages in any mqtt topic

viktorkho, 2020-05-07:
**Describe the bug**
ThingsBoard IoT Gateway with custom converter (I've [described here](https://stackoverflow.com/q/61634163/5239616)) stalled if I publish "nan" message in any topic.

**Connector name (If bug in the some connector):**
MQTT Connector

**Error traceback (If available):**
Nothing in docker container's output after message published to the topic (to which it is subscribed)

**Versions (please complete the following information):**
 - docker-compose version 1.17.1
 - image: thingsboard/tb-gateway:latest
 - Python 3.6.9

zbeacon, 2020-05-07:
Hi @viktorkho 

It is an issue with mosquitto_pub, could you try to use 
`mosquitto_pub -t /devices/wb-adc/controls/test -m "NaN"`
instead of 
`mosquitto_pub -t /devices/wb-adc/controls/test -m nan`

I used the custom converter from your question on StackOverflow.

Please try and let me know, about results.

viktorkho, 2020-05-07:
Hi, @zbeacon 

In this case, the issue is NOT ONLY with mosquitto_pub. My `mosquitto_pub ...` case is just reproduction example.

Originally I faced this problem connecting IoT GW to [Wirenboard](wirenboard.com) 's mosquitto broker. I'll try to cpecify librs they are use. (I don't exclude that it is Paho, just like mosquitto_pub do.)

Is it possible to solve this kind of issues with "nan"? (it will take me much more time to understand your code..)

zbeacon, 2020-05-07:
@viktorkho 

Thank you for your cooperation, 
I found the issue, it was with message decoding. I fixed it in the develop branch, but the ThingsBoard won't save "null" data.

zbeacon, 2020-05-07:
@viktorkho 

If you need to save this data, you can check in converter - is it None and save string "None" instead of null value.

viktorkho, 2020-05-07:
Thank you, @zbeacon !

I found your commit [8a7cd76ac617583cca2ad02ca12ada17546712ea](https://github.com/thingsboard/thingsboard-gateway/commit/8a7cd76ac617583cca2ad02ca12ada17546712ea#diff-7279cbd6be0e8c75dc9f94713681e73b) and tested it, all works fine.

closed 2020-05-07 by viktorkho