thingsboard/thingsboard-gateway #300: modbus rtu-buffered function in thingsboard gateway

pehedima, 2020-05-07:
Hello,
we use a netPI with internal RS485 interface (miniUART, ttyS0) as TB-gateway and send modbus serial rtu values to TB-server.  

I checked the communication between tb-gateway and interface and found a problem, that not every responce from the interface is valid. 
Errors are: not enough bytes were received, checksum error and so on.

I used the following python script (check_client_tmp_ttyS0.py) to check that in detail:

```
import logging
from pymodbus.client.sync import ModbusSerialClient as ModbusClient #initialize a serial RTU client instance

client= ModbusClient(method = "RTU", port = "/dev/ttyS0", stopbits = 1, bytesize = 8, parity = 'N', baudrate = 9600, timeout=0.1)
connection = client.connect()

FORMAT = ('%(asctime)-15s %(threadName)-15s'
' %(levelname)-8s %(module)-15s:%(lineno)-8s %(message)s')
logging.basicConfig(format=FORMAT)
log = logging.getLogger()
log.setLevel(logging.DEBUG)

def processResponse(result):
    log.debug(result)

rr = client.read_holding_registers(0, 1, unit=0x01)
assert(not rr.isError())     # test that we are not an error
log.debug(rr)

client.close()
```

This are the results (8 attempts)

```
root@77621fc:/etc/thingsboard-gateway/config# python3 check_client_tmp_ttyS0.py
2020-04-27 14:17:43,418 MainThread      DEBUG    transaction    :115      Current transaction state - IDLE
2020-04-27 14:17:43,419 MainThread      DEBUG    transaction    :120      Running transaction 1
2020-04-27 14:17:43,421 MainThread      DEBUG    transaction    :219      SEND: 0x1 0x3 0x0 0x0 0x0 0x1 0x84 0xa
2020-04-27 14:17:43,422 MainThread      DEBUG    sync           :75       New Transaction state 'SENDING'
2020-04-27 14:17:43,423 MainThread      DEBUG    transaction    :228      Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2020-04-27 14:17:43,535 MainThread      DEBUG    transaction    :302      Incomplete message received, Expected 7 bytes Recieved 6 bytes !!!!
2020-04-27 14:17:43,537 MainThread      DEBUG    transaction    :304      Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2020-04-27 14:17:43,540 MainThread      DEBUG    transaction    :233      RECV: 0x3 0x2 0x0 0xd7 0xf8 0x1a
2020-04-27 14:17:43,542 MainThread      DEBUG    rtu_framer     :97       CRC invalid, discarding header!!
2020-04-27 14:17:43,544 MainThread      DEBUG    rtu_framer     :128      Resetting frame - Current Frame in buffer - 0x3 0x2 0x0 0xd7 0xf8 0x1a
2020-04-27 14:17:43,545 MainThread      DEBUG    rtu_framer     :232      Frame check failed, ignoring!!
2020-04-27 14:17:43,546 MainThread      DEBUG    rtu_framer     :128      Resetting frame - Current Frame in buffer -
2020-04-27 14:17:43,547 MainThread      DEBUG    transaction    :394      Getting transaction 1
2020-04-27 14:17:43,548 MainThread      DEBUG    transaction    :193      Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
Traceback (most recent call last):
  File "check_client_tmp_ttyS0.py", line 17, in <module>
    assert(not rr.isError())     # test that we are not an error
AssertionError
root@77621fc:/etc/thingsboard-gateway/config# python3 check_client_tmp_ttyS0.py
2020-04-27 14:17:45,770 MainThread      DEBUG    transaction    :115      Current transaction state - IDLE
2020-04-27 14:17:45,771 MainThread      DEBUG    transaction    :120      Running transaction 1
2020-04-27 14:17:45,771 MainThread      DEBUG    transaction    :219      SEND: 0x1 0x3 0x0 0x0 0x0 0x1 0x84 0xa
2020-04-27 14:17:45,771 MainThread      DEBUG    sync           :75       New Transaction state 'SENDING'
2020-04-27 14:17:45,772 MainThread      DEBUG    transaction    :228      Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2020-04-27 14:17:45,788 MainThread      DEBUG    transaction    :304      Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2020-04-27 14:17:45,789 MainThread      DEBUG    transaction    :233      RECV: 0x1 0x3 0x2 0x0 0xd5 0x79 0xdb
2020-04-27 14:17:45,790 MainThread      DEBUG    rtu_framer     :180      Getting Frame - 0x3 0x2 0x0 0xd5
2020-04-27 14:17:45,791 MainThread      DEBUG    factory        :266      Factory Response[ReadHoldingRegistersResponse: 3]
2020-04-27 14:17:45,792 MainThread      DEBUG    rtu_framer     :115      Frame advanced, resetting header!!
2020-04-27 14:17:45,793 MainThread      DEBUG    transaction    :383      Adding transaction 1
2020-04-27 14:17:45,793 MainThread      DEBUG    transaction    :394      Getting transaction 1
2020-04-27 14:17:45,794 MainThread      DEBUG    transaction    :193      Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2020-04-27 14:17:45,795 MainThread      DEBUG    check_client_tmp_ttyS0:18       ReadRegisterResponse (1)
root@77621fc:/etc/thingsboard-gateway/config# python3 check_client_tmp_ttyS0.py
2020-04-27 14:17:47,677 MainThread      DEBUG    transaction    :115      Current transaction state - IDLE
[... 93 lines cut ...]
2020-04-27 14:17:56,275 MainThread      DEBUG    transaction    :304      Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2020-04-27 14:17:56,277 MainThread      DEBUG    transaction    :233      RECV: 0x3 0x2 0x0 0xd5 0x79 0xdb
2020-04-27 14:17:56,278 MainThread      DEBUG    rtu_framer     :97       CRC invalid, discarding header!!
2020-04-27 14:17:56,278 MainThread      DEBUG    rtu_framer     :128      Resetting frame - Current Frame in buffer - 0x3 0x2 0x0 0xd5 0x79 0xdb
2020-04-27 14:17:56,279 MainThread      DEBUG    rtu_framer     :232      Frame check failed, ignoring!!
2020-04-27 14:17:56,280 MainThread      DEBUG    rtu_framer     :128      Resetting frame - Current Frame in buffer -
2020-04-27 14:17:56,280 MainThread      DEBUG    transaction    :394      Getting transaction 1
2020-04-27 14:17:56,281 MainThread      DEBUG    transaction    :193      Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
Traceback (most recent call last):
  File "check_client_tmp_ttyS0.py", line 17, in <module>
    assert(not rr.isError())     # test that we are not an error
AssertionError
```

Then I checked the 'rtu-buffered' function which is used in 'node-red'. This function is especially wrote for such cases and is working without any problems.

definition:
```
RTU Buffered
... your device is splitting response messages sometimes or every time - you got CRC errors or expected errors
```

[Here](https://www.npmjs.com/package/modbus-serial) are some additional information.

It is possible to implement such 'rtu-buffered' function in thingsboard gateway, so that only valid values are send to the server?

zbeacon, 2020-05-07:
Hi @pehedima 

Unfortunately, I have no ability to check this issue at the moment, but could you uncomment the following (Selected, on screenshot below) lines in the modbus_connector.py and check the result.

[image]

pehedima, 2020-05-07:
Thanks for your fast response :)

This is the result:
[connector.log](https://github.com/thingsboard/thingsboard-gateway/files/4591396/connector.log)

Sometimes the correct temperature values ( e.g. : 18.2) are send to TB-Server.

I uncommed the lines and used the following script:
[modbus_connector.zip](https://github.com/thingsboard/thingsboard-gateway/files/4591411/modbus_connector.zip)

My basic package is (2.2.5-1) and pymodbus version 2.3.0 ist installed.

Thanks for your cooperation

zbeacon, 2020-05-07:
@pehedima 

Please update your gateway to the latest version, and try again. Also please read [this article](https://thingsboard.io/docs/iot-gateway/install/upgrade-instructions/), I rewrote the update instructions.

pehedima, 2020-05-08:
I installed the latest package python3-thingsboard-gateway (2.3.0-1) and uncommented the three lines in modbus_connector.py.

Here the result:

```
''2020-05-08 07:39:24' - DEBUG - modbus_connector - 138 - Checking timeseries for device TEMP-Home-python-netpi-balena-serial-ttyS0-1'
''2020-05-08 07:39:24' - DEBUG - modbus_connector - 140 - {'timeseries': {}, 'attributes': {}}'
''2020-05-08 07:39:24' - DEBUG - bytes_modbus_uplink_converter - 80 - {'deviceName': 'TEMP-Home-python-netpi-balena-serial-ttyS0-1', 'deviceType': 'default', 'telemetry': [], 'attributes': []}'
''2020-05-08 07:39:24' - DEBUG - modbus_connector - 260 - With result Modbus Error: [Input/Output] read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)'
''2020-05-08 07:39:24' - ERROR - modbus_connector - 133 - Modbus Error: [Input/Output] read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)'
NoneType: None
''2020-05-08 07:39:24' - DEBUG - modbus_connector - 138 - Checking timeseries for device TEMP-Home-python-netpi-balena-serial-ttyS0-1'
''2020-05-08 07:39:24' - DEBUG - modbus_connector - 140 - {'timeseries': {}, 'attributes': {}}'
''2020-05-08 07:39:24' - DEBUG - bytes_modbus_uplink_converter - 80 - {'deviceName': 'TEMP-Home-python-netpi-balena-serial-ttyS0-1', 'deviceType': 'default', 'telemetry': [], 'attributes': []}'
''2020-05-08 07:39:29' - DEBUG - modbus_connector - 260 - With result Modbus Error: [Input/Output] read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)'
''2020-05-08 07:39:29' - ERROR - modbus_connector - 133 - Modbus Error: [Input/Output] read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)'
NoneType: None
''2020-05-08 07:39:29' - DEBUG - modbus_connector - 138 - Checking timeseries for device TEMP-Home-python-netpi-balena-serial-ttyS0-1'
''2020-05-08 07:39:29' - DEBUG - modbus_connector - 140 - {'timeseries': {}, 'attributes': {}}'
''2020-05-08 07:39:29' - DEBUG - bytes_modbus_uplink_converter - 80 - {'deviceName': 'TEMP-Home-python-netpi-balena-serial-ttyS0-1', 'deviceType': 'default', 'telemetry': [], 'attributes': []}'
''2020-05-08 07:39:34' - DEBUG - modbus_connector - 260 - With result ReadRegisterResponse (1)'
''2020-05-08 07:39:34' - DEBUG - modbus_connector - 138 - Checking timeseries for device TEMP-Home-python-netpi-balena-serial-ttyS0-1'
''2020-05-08 07:39:34' - DEBUG - modbus_connector - 140 - {'timeseries': {'Temperatur_Buero': {'data_sent': {'tag': 'Temperatur_Buero', 'type': 'long', 'functionCode': 3, 'address': 0, 'registerCount': 1, 'byteOrder': 'BIG', 'divider': 10, 'deviceName': 'TEMP-Home-python-netpi-balena-serial-ttyS0-1'}, 'input_data': <pymodbus.register_read_message.ReadHoldingRegistersResponse object at 0x74de0970>}}, 'attributes': {}}'
''2020-05-08 07:39:34' - DEBUG - bytes_modbus_uplink_converter - 59 - Tag: Temperatur_Buero Config: {'tag': 'Temperatur_Buero', 'type': 'long', 'functionCode': 3, 'address': 0, 'registerCount': 1, 'byteOrder': 'BIG', 'divider': 10, 'deviceName': 'TEMP-Home-python-netpi-balena-serial-ttyS0-1'} registers: [187]'
''2020-05-08 07:39:34' - DEBUG - bytes_modbus_uplink_converter - 76 - datatype: telemetry        key: Temperatur_Buero   value: 18.7'
''2020-05-08 07:39:34' - DEBUG - bytes_modbus_uplink_converter - 80 - {'deviceName': 'TEMP-Home-python-netpi-balena-serial-ttyS0-1', 'deviceType': 'default', 'telemetry': [{'Temperatur_Buero': 18.7}], 'attributes': []}'
''2020-05-08 07:39:39' - DEBUG - modbus_connector - 260 - With result Modbus Error: [Input/Output] [Errno 5] Input/output error'
''2020-05-08 07:39:39' - ERROR - modbus_connector - 133 - Modbus Error: [Input/Output] [Errno 5] Input/output error'
NoneType: None
''2020-05-08 07:39:39' - DEBUG - modbus_connector - 138 - Checking timeseries for device TEMP-Home-python-netpi-balena-serial-ttyS0-1'
''2020-05-08 07:39:39' - DEBUG - modbus_connector - 140 - {'timeseries': {}, 'attributes': {}}'
''2020-05-08 07:39:39' - DEBUG - bytes_modbus_uplink_converter - 80 - {'deviceName': 'TEMP-Home-python-netpi-balena-serial-ttyS0-1', 'deviceType': 'default', 'telemetry': [], 'attributes': []}'
''2020-05-08 07:39:54' - DEBUG - modbus_connector - 260 - With result Modbus Error: [Input/Output] read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)'
''2020-05-08 07:39:54' - ERROR - modbus_connector - 133 - Modbus Error: [Input/Output] read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)'
NoneType: None
''2020-05-08 07:39:54' - DEBUG - modbus_connector - 138 - Checking timeseries for device TEMP-Home-python-netpi-balena-serial-ttyS0-1'
''2020-05-08 07:39:54' - DEBUG - modbus_connector - 140 - {'timeseries': {}, 'attributes': {}}'
''2020-05-08 07:39:54' - DEBUG - bytes_modbus_uplink_converter - 80 - {'deviceName': 'TEMP-Home-python-netpi-balena-serial-ttyS0-1', 'deviceType': 'default', 'telemetry': [], 'attributes': []}'
''2020-05-08 07:39:55' - DEBUG - modbus_connector - 260 - With result ReadRegisterResponse (1)'
''2020-05-08 07:39:55' - DEBUG - modbus_connector - 138 - Checking timeseries for device TEMP-Home-python-netpi-balena-serial-ttyS0-1'
[... cut ...]
''2020-05-08 07:39:55' - DEBUG - bytes_modbus_uplink_converter - 59 - Tag: Temperatur_Buero Config: {'tag': 'Temperatur_Buero', 'type': 'long', 'functionCode': 3, 'address': 0, 'registerCount': 1, 'byteOrder': 'BIG', 'divider': 10, 'deviceName': 'TEMP-Home-python-netpi-balena-serial-ttyS0-1'} registers: [187]'
''2020-05-08 07:39:55' - DEBUG - bytes_modbus_uplink_converter - 76 - datatype: telemetry        key: Temperatur_Buero   value: 18.7'
''2020-05-08 07:39:55' - DEBUG - bytes_modbus_uplink_converter - 80 - {'deviceName': 'TEMP-Home-python-netpi-balena-serial-ttyS0-1', 'deviceType': 'default', 'telemetry': [{'Temperatur_Buero': 18.7}], 'attributes': []}'

```

zbeacon, 2020-05-08:
@pehedima ,

As we can see, sometimes the gateway receives errors while reading data (`Modbus Error: [Input/Output] read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)`). 
But  also it reads data - `{'deviceName': 'TEMP-Home-python-netpi-balena-serial-ttyS0-1', 'deviceType': 'default', 'telemetry': [{'Temperatur_Buero': 18.7}], 'attributes': []}`.

Errors are ignored and processing continues.
Was that your goal? Or am I misunderstood something?

pehedima, 2020-05-08:
Thanks for your answer :)

I use the following JSON file to read out the temperature every 5 Sec (testing mode):

```
{
  "server": {
    "name": "temperature serial",
    "type": "serial",
    "port": "/dev/ttyS0",
    "baudrate": 19200,
    "timeout": 35,
    "method": "rtu",
    "devices": [
      {
        "unitId": 1,
        "deviceName": "TEMP-Home-python-netpi-balena-serial-ttyS0-1",
        "attributesPollPeriod": 3600000,
        "timeseriesPollPeriod": 5000,
        "sendDataOnlyOnChange": false,
         "timeseries":[
            {
               "tag":"Temperatur_Buero",
               "type":"long",
               "functionCode":3,
               "address":0,
               "registerCount":1,
               "byteOrder":"BIG",
               "divider":10
            }
         ]
      }
    ]
  }
}
```

The problem is, that not every 5sec a value is send to TB Server.
The logfile below shows, that in a case of an error the value is missing and sometimes after 15-40 sec. later the next value is read.

```
''2020-05-08 11:00:25' - DEBUG - modbus_connector - 140 - {'timeseries': {'Temperatur_Buero': {'data_sent': {'tag': 'Temperatur_Buero', 'type': 'long', 'functionCode': 3, 'address': 0, 'registerCount': 1, 'byteOrder': 'BIG', 'divider': 10, 'deviceName': 'TEMP-Home-python-netpi-balena-serial-ttyS0-1'}, 'input_data': <pymodbus.register_read_message.ReadHoldingRegistersResponse object at 0x74d40c70>}}, 'attributes': {}}'
''2020-05-08 11:00:25' - DEBUG - bytes_modbus_uplink_converter - 59 - Tag: Temperatur_Buero Config: {'tag': 'Temperatur_Buero', 'type': 'long', 'functionCode': 3, 'address': 0, 'registerCount': 1, 'byteOrder': 'BIG', 'divider': 10, 'deviceName': 'TEMP-Home-python-netpi-balena-serial-ttyS0-1'} registers: [197]'
''2020-05-08 11:00:25' - DEBUG - bytes_modbus_uplink_converter - 76 - datatype: telemetry        key: Temperatur_Buero   value: 19.7'
''2020-05-08 11:00:25' - DEBUG - bytes_modbus_uplink_converter - 80 - {'deviceName': 'TEMP-Home-python-netpi-balena-serial-ttyS0-1', 'deviceType': 'default', 'telemetry': [{'Temperatur_Buero': 19.7}], 'attributes': []}'
''2020-05-08 11:00:30' - DEBUG - modbus_connector - 260 - With result ReadRegisterResponse (1)'
''2020-05-08 11:00:30' - DEBUG - modbus_connector - 138 - Checking timeseries for device TEMP-Home-python-netpi-balena-serial-ttyS0-1'
''2020-05-08 11:00:30' - DEBUG - modbus_connector - 140 - {'timeseries': {'Temperatur_Buero': {'data_sent': {'tag': 'Temperatur_Buero', 'type': 'long', 'functionCode': 3, 'address': 0, 'registerCount': 1, 'byteOrder': 'BIG', 'divider': 10, 'deviceName': 'TEMP-Home-python-netpi-balena-serial-ttyS0-1'}, 'input_data': <pymodbus.register_read_message.ReadHoldingRegistersResponse object at 0x74d40cf0>}}, 'attributes': {}}'
''2020-05-08 11:00:30' - DEBUG - bytes_modbus_uplink_converter - 59 - Tag: Temperatur_Buero Config: {'tag': 'Temperatur_Buero', 'type': 'long', 'functionCode': 3, 'address': 0, 'registerCount': 1, 'byteOrder': 'BIG', 'divider': 10, 'deviceName': 'TEMP-Home-python-netpi-balena-serial-ttyS0-1'} registers: [194]'
''2020-05-08 11:00:30' - DEBUG - bytes_modbus_uplink_converter - 76 - datatype: telemetry        key: Temperatur_Buero   value: 19.4'
''2020-05-08 11:00:30' - DEBUG - bytes_modbus_uplink_converter - 80 - {'deviceName': 'TEMP-Home-python-netpi-balena-serial-ttyS0-1', 'deviceType': 'default', 'telemetry': [{'Temperatur_Buero': 19.4}], 'attributes': []}'
''2020-05-08 11:00:35' - DEBUG - modbus_connector - 260 - With result Modbus Error: [Input/Output] [Errno 5] Input/output error'
''2020-05-08 11:00:35' - ERROR - modbus_connector - 133 - Modbus Error: [Input/Output] [Errno 5] Input/output error'
NoneType: None
''2020-05-08 11:00:35' - DEBUG - modbus_connector - 138 - Checking timeseries for device TEMP-Home-python-netpi-balena-serial-ttyS0-1'
''2020-05-08 11:00:35' - DEBUG - modbus_connector - 140 - {'timeseries': {}, 'attributes': {}}'
''2020-05-08 11:00:35' - DEBUG - bytes_modbus_uplink_converter - 80 - {'deviceName': 'TEMP-Home-python-netpi-balena-serial-ttyS0-1', 'deviceType': 'default', 'telemetry': [], 'attributes': []}'
''2020-05-08 11:00:50' - DEBUG - modbus_connector - 260 - With result Modbus Error: [Input/Output] read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)'
''2020-05-08 11:00:50' - ERROR - modbus_connector - 133 - Modbus Error: [Input/Output] read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)'
NoneType: None
''2020-05-08 11:00:50' - DEBUG - modbus_connector - 138 - Checking timeseries for device TEMP-Home-python-netpi-balena-serial-ttyS0-1'
''2020-05-08 11:00:50' - DEBUG - modbus_connector - 140 - {'timeseries': {}, 'attributes': {}}'
''2020-05-08 11:00:50' - DEBUG - bytes_modbus_uplink_converter - 80 - {'deviceName': 'TEMP-Home-python-netpi-balena-serial-ttyS0-1', 'deviceType': 'default', 'telemetry': [], 'attributes': []}'
''2020-05-08 11:00:50' - DEBUG - modbus_connector - 260 - With result ReadRegisterResponse (1)'
''2020-05-08 11:00:50' - DEBUG - modbus_connector - 138 - Checking timeseries for device TEMP-Home-python-netpi-balena-serial-ttyS0-1'
''2020-05-08 11:00:50' - DEBUG - modbus_connector - 140 - {'timeseries': {'Temperatur_Buero': {'data_sent': {'tag': 'Temperatur_Buero', 'type': 'long', 'functionCode': 3, 'address': 0, 'registerCount': 1, 'byteOrder': 'BIG', 'divider': 10, 'deviceName': 'TEMP-Home-python-netpi-ba [... cut ...]
If a normal USB adapter for Modbus serial is used the temperature values are send every 5 sec. without any problems. This adapter delivers correct values every 5 sec. instead of the internal S0 interface. That is the reason why I mentioned the rtu-buffered function of node red. This function is able to collect data and deliver only complete and valid values. So as I tested. 

Logfile using USB adapter
[modbus_serial_USB0_5sec.txt](https://github.com/thingsboard/thingsboard-gateway/files/4598291/modbus_serial_USB0_5sec.txt)

Well, I search for a solution for the internal S0 interface of my IoT device that the TB gateway sends a valid value every 5 sec. (e.g.). 

Thanks for your help :)

closed 2020-05-20 by zbeacon