thingsboard/thingsboard-gateway #302: [BUG] ThingsBoard IoT Gateway stalled on messages with spaces

viktorkho, 2020-05-07:
**Describe the bug**
ThingsBoard IoT Gateway with custom converter (the same as in #299) stalled if I publish messages containing spaces, e.g.:
```
mosquitto_pub -t "/devices/test/controls/spaces" -m "foo"     # INFO Successfully converted
mosquitto_pub -t "/devices/test/controls/spaces" -m "foo bar" # Nothing in log
mosquitto_pub -t "/devices/test/controls/spaces" -m "bar"     # Nothing in log
```
**Connector name (If bug in the some connector):**
MQTT Connector

**Error traceback (If available):**
Nothing in docker container's output after message with spaces published to the topic (to which it is subscribed)

**Versions (please complete the following information):**
 - docker-compose version 1.17.1
 - image: thingsboard/tb-gateway:latest
 - Python 3.6.9

zbeacon, 2020-05-07:
Hi @viktorkho 

Ok, I refactored it. Please update your version to the version from develop/2.4-python branch and check it.

zbeacon, 2020-05-07:
Commit 78deef16c81ef3a2eca9fbfda0fa78c06f46421e

viktorkho, 2020-05-07:
@zbeacon , thank you again!

Now it passes even hex:
```
mosquitto_pub -t /devices/test/controls/hex -m "$(echo -en '\x0d\x0e\x0a\x0d\x0b\x0e\x0e\x0f')"
# DEBUG ... 'telemetry': [{'test/hex': '\r\x0e\n\r\x0b\x0e\x0e\x0f'}]}'
# INFO ... Successfully converted message from topic /devices/test/controls/hex'
```

closed 2020-05-07 by viktorkho