thingsboard/thingsboard-gateway #303: [HELP]

hamed-boumaiza-protectline, 2020-05-11:
Hi,
After installing TB IOT Gateway and configuring it locally, i can't understand what are these telemetries i am getting in the screenshot below (how to disable them?). And how can i add custom telemetries sent by mosquitto.

[image]

[image]

`    thingsboard:
      host: 10.10.0.1
      port: 1883
      remoteConfiguration: false
      security:
        accessToken: accessToken
    storage:
      type: memory
      read_records_count: 100
      max_records_count: 100000
    connectors:
      -
    name: MQTT Broker Connector
    type: mqtt
    configuration: mqtt.json`

[image]
`{
  "broker": {
    "name":"Default",
    "host":"127.0.0.1",
    "port":1883,
    "security": {
      "type": "anonymous",
    }
  },
  "mapping": [
    {
      "topicFilter": "/sensor/data",
      "converter": {
        "type": "json",
        "deviceNameJsonExpression": "${serialNumber}",
        "deviceTypeJsonExpression": "${sensorType}",
        "timeout": 60000,
        "attributes": [
          {
            "type": "string",
            "key": "model",
            "value": "${sensorModel}"
          }
        ],
        "timeseries": [
          {
            "type": "double",
            "key": "temperature",
            "value": "${temp}"
          },
          {
            "type": "double",
            "key": "humidity",
            "value": "${hum}"
          }
        ]
      }
    },
    {
      "topicFilter": "/sensor/+/data",
      "converter": {
        "type": "json",
        "deviceNameTopicExpression": "(?<=sensor\/)(.*?)(?=\/data)",
        "deviceTypeTopicExpression": "Thermometer",
        "timeout": 60000,
        "attributes": [
          {
            "type": "string",
            "key": "model",
            "value": "${sensorModel}"
          }
        ],
        "timeseries": [
          {
            "type": "double",
            "key": "temperature",
            "value": "${temp}"
          },
          {
            "type": "double",
            "key": "humidity",
            "value": "${hum}"
          }
        ]
      }
    },
    {
      "topicFilter": "/custom/sensors/+",
      "converter": {
        "type": "custom",
        "extension": "CustomMqttUplinkConverter",
        "extension-config": {
            "temperatureBytes" : 2,
            "humidityBytes" :  2,
            "batteryLevelBytes" : 1
        }
      }
    }
  ],
  "connectRequests": [
    {
      "topicFilter": "sensor/connect",
      "deviceNameJsonExpression": "${SerialNumber}"
    },
    {
      "topicFilter": "sensor/+/connect",
      "deviceNameTopicExpression": "(?<=sensor\/)(.*?)(?=\/connect)"
    }
  ],
  "disconnectRequests": [
    {
      "topicFilter": "sensor/disconnect",
      "deviceNameJsonExpression": "${SerialNumber}"
    },
    {
      "topicFilter": "sensor/+/disconnect",
      "deviceNameTopicExpression": "(?<=sensor\/)(.*?)(?=\/disconnect)"
    }
  ],
  "attributeUpdates": [
    {
      "deviceNameFilter": "SmartMeter.*",
      "attributeFilter": "uploadFrequency",
      "topicExpression": "sensor/${deviceName}/${attributeKey}",
      "valueExpression": "{\"${attributeKey}\":\"${attributeValue}\"}"
    }
  ],
  "serverSideRpc": [
    {
      "deviceNameFilter": ".*",
      "methodFilter": "echo",
      "requestTopicExpression": "sensor/${deviceName}/request/${methodName}/${requestId}",
      "responseTopicExpression": "sensor/${deviceName}/response/${methodName}/${requestId}",
      "responseTimeout": 10000,
      "valueExpression": "${params}"
    },
    {
      "deviceNameFilter": ".*",
      "methodFilter": "no-reply",
      "requestTopicExpression": "sensor/${deviceName}/request/${methodName}/${requestId}",
      "valueExpression": "${params}"
    }
  ]
}
`

 - OS: Windows 10
 - Thingsboard IoT Gateway version : Lastest
 - Python version : 3.8

zbeacon, 2020-05-12:
Hi @hamed-boumaiza-protectline,

Thank you for your interest in ThingsBoard IoT gateway.
The screenshot contains the default gateway statistics, which makes it possible to analyze the number of messages for each connector (These values are updated every 5 seconds, you can create timeseries dashboard to check the statistics). To get information from an external broker, you need to create another device, for example, “Service device” (By editing MQTT configuration file), subscribe to some topics, send data, it will be just a regular device, but you can get any information you want.

Or you can create a new connector or change current to get required information.

hamed-boumaiza-protectline, 2020-05-12:
Hi @zbeacon,
Thanks for your reply, can you show me any example with some details of custom configuration that can work.

closed 2020-05-12 by hamed-boumaiza-protectline