thingsboard/thingsboard-gateway #304: where find  privateKey: /etc/thingsboard-gateway/privateKey.pem     caCert: /etc/thingsboard-gateway/ca.pem     cert: /etc/thingsboard-gateway/certificate.pem

waytai, 2020-05-14:
**Describe the issue**
Create description about your issue, and your actions to solve it.
In thingsboard there just generate mqttclient.pem mqttclient.pub.pem, mqttserver.pub.pem,mqttclient.nopass.pem. and which file is privateKey: /etc/thingsboard-gateway/privateKey.pem in tb_gateway config, which file is  caCert: /etc/thingsboard-gateway/ca.pem in tb gateway config , and caCert?

**Configuration** (Attach your configuration file)
**Notate: Remove Access token from file if you wanna attach tb_gateway.yaml**

**Connector name (If you need help with some connector/converter):**
[e.g. MQTT Connector]
gateway mqtt tls connect to thingsboard

zbeacon, 2020-05-15:
Hi @waytai 

Thank your for your interest in the ThingsBoard IoT gateway.
Try to use `mqttserver.pub.pem` as **caCert**, `mqttclient.nopass.pem` as **cert** and remove privateKey. Please let me know about results.

closed 2020-05-20 by zbeacon

zbeacon, 2020-05-20:
@waytai 

Thank you for your interest in the ThingsBoard IoT gateway. Please let me know, if you will have any questions.