thingsboard/thingsboard-gateway #306: [HELP]How does the RPC getValue method works?

qwyinxiang, 2020-05-15:
**Describe the issue**
I want to know how does the RPC getValue method works?
I created a dashboard using the switch control widget and connect my real device a relay.
when I clicked "ON" on the dashboard, my relay turns on, and when I clicked "OFF" on the dashboard, my relay turns off.
Now I want to retrive the on/off value using the "Call RPC get value method".
I viewed the tb-gateway source code, but don't know how the gateway process the RPC getValue method. and also don't know in which case, the getValue method can be called.
[image]

When I clicked the "ON/OFF" button on the dashboard, the gateway's server_side_rpc_handler(self, content)' content is like below:
`content: {'device': '继电器001', 'data': {'id': 445, 'method': 'setValue', 'params': False}}
content: {'device': '继电器001', 'data': {'id': 446, 'method': 'setValue', 'params': True}}
content: {'device': '继电器001', 'data': {'id': 447, 'method': 'setValue', 'params': False}}
content: {'device': '继电器001', 'data': {'id': 448, 'method': 'setValue', 'params': True}}`
In which case, the method 'getValue' can be called?

zbeacon, 2020-05-15:
Hi @qwyinxiang 

To configure the RPC calls, you need to change your RPC section in the Modbus configuration file. "getValue" RPC call can read value (True/False) from some coil. The default configuration for this method is following:
`
          {
            "tag": "getValue",
            "type": "bits",
            "functionCode": 1,
            "objectsCount": 1,
            "address": 31
          },
`

In the example, you sent, it uses to get current value from configured coil. To write the data to the some coil, using example above, you need to configure the section for "setValue" method. The default configuration for this method is:
`
          {
            "tag": "setValue",
            "type": "bits",
            "functionCode": 5,
            "objectsCount": 1,
            "address": 31
          },
`

If you can't configure RPC methods, please share your sections from Modbus configuration file, I will try to help you.

qwyinxiang, 2020-05-15:
@zbeacon Thank you very much. I configed the getValue in my modbus.json file.
`{
  "server": {
    "type": "tcp",
    "host": "192.168.200.173",
    "port": 32,
    "timeout": 35,
    "method": "rtu",
    "byteOrder": "BIG",
    "devices": [
      {
        "unitId": 1,
        "deviceName": "继电器001",
        "deviceType": "继电器",
        "attributesPollPeriod": 5000,
        "timeseriesPollPeriod": 5000,
        "sendDataOnlyOnChange": true,
        "rpc": [
          {
            "tag": "setValue",
            "type": "bits",
            "functionCode": 5,
            "objectsCount": 1,
            "address": 0
          },
          {
            "tag": "getValue",
            "type": "bits",
            "functionCode": 1,
            "objectsCount": 1,
            "address": 0
          }
        ]
      }
    ]
  }
}`
Where can I see the getValue method's return value?

zbeacon, 2020-05-15:
The widget state (ON/OFF) depends on result of this method. 
Also you can use it as a regular RPC method - create a widget "RPC Debug Terminal" set the device as an entity and send `getValue` as a RPC method. [RPC Debug Terminal description](https://thingsboard.io/docs/iot-gateway/guides/how-to-use-gateway-rpc-methods/)

qwyinxiang, 2020-05-18:
Thank you very much for your reply.
My modbus.json file contains the setValue() and getValue() method.
    "devices": [
      {
        "unitId": 1,
        "deviceName": "继电器001",
        "deviceType": "继电器",
        "attributesPollPeriod": 5000,
        "timeseriesPollPeriod": 5000,
        "sendDataOnlyOnChange": true,
        "rpc": [
          {
            "tag": "setValue",
            "type": "bits",
            "functionCode": 5,
            "objectsCount": 1,
            "address": 0
          },
          {
            "tag": "getValue",
            "type": "bits",
            "functionCode": 1,
            "objectsCount": 1,
            "address": 0
          }
     ]

I don’t quite understand what's the meaning of "Retrieve on/off value using method
"——“call RPC get value method”. Is this mean I can obtain the 'on/off' value by the getValue() method? If this is the case, where can I see the 'on/off' value?
[image]

If I choose the  "Retrieve on/off value using method"——“subscribe for attribute" or "subscribe for timeseries", Is this mean I can see the 'on/off' value on the device's attribute or timeseries page?
But I can see nothing on the attribute page and the timeseries page.

zbeacon, 2020-05-18:
>> Is this mean I can obtain the 'on/off' value by the getValue() method? 

Yes, the method returns value of one coil (True or False) and it will be interpreted as a current state.

>> Is this mean I can see the 'on/off' value on the device's attribute or timeseries page?

No, the widget subscribes to the some attribute/telemetry key and uses it value as a current state.

qwyinxiang, 2020-05-19:
OK， I understand, thank you very much for your patience.

closed 2020-05-22 by zbeacon

santobedi, 2021-12-03:
> Hi @qwyinxiang
>
> To configure the RPC calls, you need to change your RPC section in the Modbus configuration file. "getValue" RPC call can read value (True/False) from some coil. The default [...]

Hello @zbeacon ,

My scenario is similar to @qwyinxiang 's case except I'm using LoRaWAN (Chirpstack server) to give commands to the Modbus relay device (TB gateway is not involved, plz excuse me if I'm commenting on the wrong thread).  I'm using Switch Control (Control widgets) to send commands via external `rest API call `. I want to do it using the `integration downlink` (I've not able to do it yet and posted a question [here](https://github.com/thingsboard/thingsboard/issues/5658) for any help). Either way (external `rest API call` or `integration downlink`), how can I use RPC getValue method?