thingsboard/thingsboard-gateway #309: [BUG]make_packages.sh error

qwyinxiang, 2020-05-18:
**Describe the bug**
I clone the latest release of thingsboard-gateway 2.3.0.1 on my centos7.7. and make some changes to the source code. Now I want to generate an rpm package, but when I execute the make_packages.sh file. some errors occurs.

**Error traceback (If available):**
```
warning: install_lib: byte-compiling is disabled, skipping.

+ /usr/lib/rpm/find-debuginfo.sh --strict-build-id -m --run-dwz --dwz-low-mem-die-limit 10000000 --dwz-max-die-limit 110000000 /opt/tb-gateway/build/bdist.linux-x86_64/rpm/BUILD/thingsboard-gateway-2.3.0.1
find: 'debug': No such file or directory
+ /usr/lib/rpm/check-buildroot
+ /usr/lib/rpm/redhat/brp-compress
+ /usr/lib/rpm/redhat/brp-strip-static-archive /usr/bin/strip
+ /usr/lib/rpm/brp-python-bytecompile python3 1
+ /usr/lib/rpm/redhat/brp-python-hardlink
+ /usr/lib/rpm/redhat/brp-java-repack-jars
+ umask 022
+ cd /opt/tb-gateway/build/bdist.linux-x86_64/rpm/BUILD
+ cd thingsboard-gateway-2.3.0.1
+ rm -rf /opt/tb-gateway/build/bdist.linux-x86_64/rpm/BUILDROOT/thingsboard-gateway-2.3.0.1-1.x86_64
+ exit 0
+ umask 022
+ cd /opt/tb-gateway/build/bdist.linux-x86_64/rpm/BUILD
+ rm -rf thingsboard-gateway-2.3.0.1
+ exit 0
+ umask 022
+ cd /root/rpmbuild/BUILD
+ sudo mkdir -p /root/rpmbuild/BUILDROOT/thingsboard-gateway-2.3.0-1.x86_64/etc/thingsboard-gateway
+ sudo mkdir -p /root/rpmbuild/BUILDROOT/thingsboard-gateway-2.3.0-1.x86_64/var/lib/thingsboard_gateway
+ sudo mkdir -p /root/rpmbuild/BUILDROOT/thingsboard-gateway-2.3.0-1.x86_64/var/lib/thingsboard_gateway/extensions
+ sudo mkdir -p /root/rpmbuild/BUILDROOT/thingsboard-gateway-2.3.0-1.x86_64/var/log/thingsboard-gateway
+ sudo chown -R thingsboard_gateway:thingsboard_gateway /root/rpmbuild/BUILDROOT/thingsboard-gateway-2.3.0-1.x86_64/var/log/thingsboard-gateway
chown: invalid user: 'thingsboard_gateway:thingsboard_gateway'
错误：/var/tmp/rpm-tmp.bbzhb7 (%build) 退出状态不好
    /var/tmp/rpm-tmp.bbzhb7 (%build) 退出状态不好
cp: 无法获取"/home/root/rpmbuild/RPMS/noarch/*.rpm" 的文件状态(stat): 没有那个文件或目录
mv: 无法获取"thingsboard-gateway-2.3.0.1-1.noarch.rpm" 的文件状态(stat): 没有那个文件或目录
chown: 无法访问"*.rpm": 没有那个文件或目录
[root@wlw153 tb-gateway]# clear
[root@wlw153 tb-gateway]# ./make_packages.sh
Building DEB package
usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
   or: setup.py --help [cmd1 cmd2 ...]
   or: setup.py --help-commands
   or: setup.py cmd --help

error: invalid command 'bdist_deb'
cp: 无法创建目录"deb_dist/thingsboard-gateway-2.3.0.1/debian/python3-thingsboard-gateway": 没有那个文件或目录
cp: 无法创建目录"deb_dist/thingsboard-gateway-2.3.0.1/debian/python3-thingsboard-gateway": 没有那个文件或目录
cp: 无法创建目录"deb_dist/thingsboard-gateway-2.3.0.1/debian/python3-thingsboard-gateway": 没有那个文件或目录
chown: 无法访问"deb_dist/thingsboard-gateway-2.3.0.1/debian/python3-thingsboard-gateway/": 没有那个文件或目录
chown: 无法访问"deb_dist/thingsboard-gateway-2.3.0.1/debian/python3-thingsboard-gateway/var/": 没有那个文件或目录
chmod: 无法访问"deb_dist/thingsboard-gateway-2.3.0.1/debian/python3-thingsboard-gateway/DEBIAN/preinst": 没有那个文件或目录
chown: 无法访问"deb_dist/thingsboard-gateway-2.3.0.1/debian/python3-thingsboard-gateway/DEBIAN/preinst": 没有那个文件或目录
./make_packages.sh:行56: dpkg-deb: 未找到命令
cp: 无法获取"deb_dist/thingsboard-gateway-2.3.0.1/debian/python3-thingsboard-gateway.deb" 的文件状态(stat): 没有那个文件或目录
cp: 无法获取"python3-thingsboard-gateway.deb" 的文件状态(stat): 没有那个文件或目录
Building RPM package
running bdist_rpm
running egg_info
writing thingsboard_gateway.egg-info/PKG-INFO
writing dependency_links to thingsboard_gateway.egg-info/dependency_links.txt
writing entry points to thingsboard_gateway.egg-info/entry_points.txt
writing requirements to thingsboard_gateway.egg-info/requires.txt
writing top-level names to thingsboard_gateway.egg-info/top_level.txt
reading manifest file 'thingsboard_gateway.egg-info/SOURCES.txt'
writing manifest file 'thingsboard_gateway.egg-info/SOURCES.txt'
writing 'build/bdist.linux-x86_64/rpm/SPECS/thingsboard-gateway.spec'
running sdist
running check
creating thingsboard-gateway-2.3.0.1
[... 585 lines cut ...]
extensions/tcpserver/tcpserver_connector.py
extensions/tcpserver/tcpserver_converter.py
执行(%build): /bin/sh -e /var/tmp/rpm-tmp.bwMznU
+ umask 022
+ cd /root/rpmbuild/BUILD
+ sudo mkdir -p /root/rpmbuild/BUILDROOT/thingsboard-gateway-2.3.0-1.x86_64/etc/thingsboard-gateway
+ sudo mkdir -p /root/rpmbuild/BUILDROOT/thingsboard-gateway-2.3.0-1.x86_64/var/lib/thingsboard_gateway
+ sudo mkdir -p /root/rpmbuild/BUILDROOT/thingsboard-gateway-2.3.0-1.x86_64/var/lib/thingsboard_gateway/extensions
+ sudo mkdir -p /root/rpmbuild/BUILDROOT/thingsboard-gateway-2.3.0-1.x86_64/var/log/thingsboard-gateway
+ sudo chown -R thingsboard_gateway:thingsboard_gateway /root/rpmbuild/BUILDROOT/thingsboard-gateway-2.3.0-1.x86_64/var/log/thingsboard-gateway
chown: invalid user: 'thingsboard_gateway:thingsboard_gateway'
错误：/var/tmp/rpm-tmp.bwMznU (%build) 退出状态不好

RPM 构建错误：
    /var/tmp/rpm-tmp.bwMznU (%build) 退出状态不好
cp: 无法获取"/home/root/rpmbuild/RPMS/noarch/*.rpm" 的文件状态(stat): 没有那个文件或目录
mv: 无法获取"thingsboard-gateway-2.3.0.1-1.noarch.rpm" 的文件状态(stat): 没有那个文件或目录
chown: 无法访问"*.rpm": 没有那个文件或目录

```

**Versions (please complete the following information):**
 - OS: [CentOS Linux release 7.7.1908 (Core)]

 - Thingsboard IoT Gateway version [e2.3.0.1]

zbeacon, 2020-05-18:
Hi @qwyinxiang 

Thank you, for your feedback. This issue caused by not existing user (thingsboard_gateway), to create it, you can use following commands:

`sudo mkdir -p $RPM_BUILD_ROOT/var/lib/thingsboard_gateway || echo "ThingsBoard user home directory already exists"`  

`/usr/bin/getent passwd thingsboard_gateway || /usr/sbin/useradd -c "ThingsBoard-Gateway Service" -r -U -d /var/lib/thingsboard_gateway thingsboard_gateway && passwd -d thingsboard_gateway
`

`/usr/bin/chown thingsboard_gateway:thingsboard_gateway $RPM_BUILD_ROOT/var/lib/thingsboard_gateway -R
`

oliver225, 2020-05-20:
the   RPM_BUILD_ROOT  is what path

zbeacon, 2020-05-20:
@oliver225 

Usually, it is a rpmbuild folder in user home directory.

oliver225, 2020-05-20:
create  thingsboard_gateway user ,
then on thingsboard_gateway's home directory  run this  script?

zbeacon, 2020-05-20:
@oliver225 

No, you have to run this script, only if the thingsboard_gateway user is existed. The simplest way to run it correctly - install  the version from deb/rpm package and create your own package after installation. Installation from deb/rpm will create required user and folders.

qwyinxiang, 2020-05-22:
@zbeacon ,The following is the commands I executed :
I logged in my CentOS system as ROOT user, so I make a directory in /home directory use 
mkdir -p /home/root/rpmbuild
Then I executed the following commands:
` mkdir -p /home/root/rpmbuild/var/lib/thingsboard_gateway || echo "thingsboard user home directory already exists"
/usr/bin/getent passwd thingsboard_gateway || /usr/sbin/useradd -c "ThingsBoard-Gateway Service" -r -U -d /var/lib/thingsboard_gateway thingsboard_gateway && passwd -d thingsboard_gateway
/usr/bin/chown thingsboard_gateway:thingsboard_gateway /home/root/rpmbuild/var/lib/thingsboard_gateway -R`

Then, I execute the make_packages.sh. The following errors occurs.

running install_egg_info
Copying thingsboard_gateway.egg-info to /opt/tb-gateway/build/bdist.linux-x86_64/rpm/BUILDROOT/thingsboard-gateway-2.3.0.1-1.x86_64/usr/lib/python3.6/site-packages/thingsboard_gateway-2.3.0.1-py3.6.egg-info
running install_scripts
Installing thingsboard-gateway script to /opt/tb-gateway/build/bdist.linux-x86_64/rpm/BUILDROOT/thingsboard-gateway-2.3.0.1-1.x86_64/usr/bin
writing list of installed files to 'INSTALLED_FILES'
+ /usr/lib/rpm/find-debuginfo.sh --strict-build-id -m --run-dwz --dwz-low-mem-die-limit 10000000 --dwz-max-die-limit 110000000 /opt/tb-gateway/build/bdist.linux-x86_64/rpm/BUILD/thingsboard-gateway-2.3.0.1
/usr/lib/rpm/sepdebugcrcfix: Updated 0 CRC32s, 0 CRC32s did match.
find: 'debug': No such file or directory
+ /usr/lib/rpm/check-buildroot
+ /usr/lib/rpm/redhat/brp-compress
+ /usr/lib/rpm/redhat/brp-strip-static-archive /usr/bin/strip
+ /usr/lib/rpm/brp-python-bytecompile python3 1
Bytecompiling .py files below /opt/tb-gateway/build/bdist.linux-x86_64/rpm/BUILDROOT/thingsboard-gateway-2.3.0.1-1.x86_64/usr/lib/python3.6 using /usr/bin/python3.6
+ /usr/lib/rpm/redhat/brp-python-hardlink
+ /usr/lib/rpm/redhat/brp-java-repack-jars
处理文件：thingsboard-gateway-2.3.0.1-1.noarch
Provides: thingsboard-gateway = 2.3.0.1-1
Requires(rpmlib): rpmlib(CompressedFileNames) <= 3.0.4-1 rpmlib(FileDigests) <= 4.6.0-1 rpmlib(PayloadFilesHavePrefix) <= 4.0-1
Requires: /usr/bin/python3 python(abi) = 3.6
检查未打包文件：/usr/lib/rpm/check-files /opt/tb-gateway/build/bdist.linux-x86_64/rpm/BUILDROOT/thingsboard-gateway-2.3.0.1-1.x86_64
写道:/opt/tb-gateway/build/bdist.linux-x86_64/rpm/SRPMS/thingsboard-gateway-2.3.0.1-1.src.rpm
写道:/opt/tb-gateway/build/bdist.linux-x86_64/rpm/RPMS/noarch/thingsboard-gateway-2.3.0.1-1.noarch.rpm
执行(%clean): /bin/sh -e /var/tmp/rpm-tmp.g2lJZd
+ umask 022
+ cd /opt/tb-gateway/build/bdist.linux-x86_64/rpm/BUILD
+ cd thingsboard-gateway-2.3.0.1
+ rm -rf /opt/tb-gateway/build/bdist.linux-x86_64/rpm/BUILDROOT/thingsboard-gateway-2.3.0.1-1.x86_64
+ exit 0
执行(--clean): /bin/sh -e /var/tmp/rpm-tmp.njoyiO
+ umask 022
+ cd /opt/tb-gateway/build/bdist.linux-x86_64/rpm/BUILD
+ rm -rf thingsboard-gateway-2.3.0.1
+ exit 0
moving build/bdist.linux-x86_64/rpm/SRPMS/thingsboard-gateway-2.3.0.1-1.src.rpm -> dist
moving build/bdist.linux-x86_64/rpm/RPMS/noarch/thingsboard-gateway-2.3.0.1-1.noarch.rpm -> dist
config/ble.json
config/can.json
config/custom_serial.json
config/logs.conf
config/modbus.json
config/modbus_serial.json
config/mqtt.json
config/odbc.json
config/opcua.json
config/request.json
config/tb_gateway.yaml
extensions/bacnet/
extensions/bacnet/__init__.py
extensions/ble/
extensions/ble/__init__.py
extensions/can/
extensions/can/__init__.py
extensions/__init__.py
extensions/modbus/
extensions/modbus/__init__.py
extensions/mqtt/
extensions/mqtt/__init__.py
extensions/mqtt/custom_mqtt_uplink_converter.py
extensions/odbc/
extensions/odbc/__init__.py
extensions/opcua/
extensions/opcua/__init__.py
extensions/request/
extensions/request/__init__.py
extensions/request/custom_request_uplink_converter.py
extensions/serial/
extensions/serial/__init__.py
extensions/serial/custom_serial_connector.py
extensions/serial/custom_serial_converter.py
extensions/tcp/
extensions/tcp/__init__.py
extensions/tcp/audible_converter.py
extensions/tcp/audible_rpc_invoke.py
extensions/tcp/custom_tcp_connector.py
extensions/tcp/custom_tcp_converter.py
extensions/tcp/rpc_tcp_factory.py
extensions/tcp/rpcinvoke.py
extensions/tcpserver/
extensions/tcpserver/__init__.py
extensions/tcpserver/tcpserver_connector.py
extensions/tcpserver/tcpserver_converter.py
执行(%build): /bin/sh -e /var/tmp/rpm-tmp.vfz8Bw
+ umask 022
+ cd /root/rpmbuild/BUILD
+ sudo mkdir -p /root/rpmbuild/BUILDROOT/thingsboard-gateway-2.3.0-1.x86_64/etc/thingsboard-gateway
+ sudo mkdir -p /root/rpmbuild/BUILDROOT/thingsboard-gateway-2.3.0-1.x86_64/var/lib/thingsboard_gateway
+ sudo mkdir -p /root/rpmbuild/BUILDROOT/thingsboard-gateway-2.3.0-1.x86_64/var/lib/thingsboard_gateway/extensions
+ sudo mkdir -p /root/rpmbuild/BUILDROOT/thingsboard-gateway-2.3.0-1.x86_64/var/log/thingsboard-gateway
+ sudo chown -R thingsboard_gateway:thingsboard_gateway /root/rpmbuild/BUILDROOT/thingsboard-gateway-2.3.0-1.x86_64/var/log/thingsboard-gateway
+ sudo install -p -D -m 644 /root/rpmbuild/SOURCES/thingsboard-gateway.service /root/rpmbuild/BUILDROOT/thingsboard-gateway-2.3.0-1.x86_64/etc/systemd/system/thingsboard-gateway.service
install: cannot stat '/root/rpmbuild/SOURCES/thingsboard-gateway.service': No such file or directory
错误：/var/tmp/rpm-tmp.vfz8Bw (%build) 退出状态不好

RPM 构建错误：
    /var/tmp/rpm-tmp.vfz8Bw (%build) 退出状态不好
cp: 无法获取"/home/root/rpmbuild/RPMS/noarch/*.rpm" 的文件状态(stat): 没有那个文件或目录
mv: 无法获取"thingsboard-gateway-2.3.0.1-1.noarch.rpm" 的文件状态(stat): 没有那个文件或目录
chown: 无法访问"*.rpm": 没有那个文件或目录

zbeacon, 2020-05-22:
@qwyinxiang 

It looks like by some reasons the script didn't execute command `cp for_build/etc/systemd/system/thingsboard-gateway.service /home/root/rpmbuild/SOURCES/` correctly. Please check - is file thingsboard-gateway.service exists in folder /home/root/rpmbuild/SOURCES/ ?

qwyinxiang, 2020-05-22:
yes, The thingsboard-gateway.service exist.
[image]

zbeacon, 2020-05-22:
The issue in the script - If you use root user to build a package, you need to replace "/home/$CURRENT_USER/..." to "/root/...".

qwyinxiang, 2020-05-22:
I replaced "/home/$CURRENT_USER/" to "/root/", and my make_packages.sh file looks like below:
# Bulding Deb package
#  sudo sed -i 's/Build-Depends.*/Build-Depends: python3-setuptools, python3-all, debhelper (>= 7.4.3), libffi-dev, libglib2.0-dev, libxml2-dev, libxslt-dev, libssl-dev, zlib1g-dev, ${python3: Depends}/g' deb_dist/thingsboard-gateway-$CURRENT_VERSION/debian/python3-thingsboard-gateway/DEBIAN/control >> deb_dist/thingsboard-gateway-$CURRENT_VERSION/debian/python3-thingsboard-gateway/DEBIAN/control
  dpkg-deb -b deb_dist/thingsboard-gateway-$CURRENT_VERSION/debian/python3-thingsboard-gateway/
  cp deb_dist/thingsboard-gateway-$CURRENT_VERSION/debian/python3-thingsboard-gateway.deb .
  cp python3-thingsboard-gateway.deb docker/
  # Create sources for RPM Package
  echo 'Building RPM package'
  sudo find thingsboard_gateway/ -name "*.pyc" -exec rm -f {} \;
  python3 setup.py bdist_rpm
  sudo find thingsboard_gateway/ -name "*.pyc" -exec rm -f {} \;
  cp build/bdist.linux-x86_64/rpm/* **/root/rpmbuild/** -r
  # Adding the file, scripts and permissions
  cp for_build/etc/systemd/system/thingsboard-gateway.service **/root/rpmbuild**/SOURCES/
#  cd for_build/etc/thingsboard-gateway/
  sudo find thingsboard_gateway/ -name "*.pyc" -exec rm -f {} \;
  cp -r thingsboard_gateway/extensions for_build/etc/thingsboard-gateway/
  cd for_build/etc/thingsboard-gateway || echo 0 > /dev/null
  tar -zcvf configs.tar.gz config/*
#  cp -r extensions **/root/rpmbuild**/SOURCES/
  tar -zcvf extensions.tar.gz extensions/*
  mv configs.tar.gz ../../../
#  mv extensions.tar.gz **/root/rpmbuild**/SOURCES/extensions.tar.gz
  cd ../../../
  rm **/root/rpmbuild/**SOURCES/configs.tar.gz
  cp configs.tar.gz **/root/rpmbuild/**SOURCES/
  # Bulding RPM Package
  cp thingsboard-gateway.spec **/root/rpmbuild/**SPECS/
  rpmbuild -ba thingsboard-gateway.spec
  sudo cp **/root/rpmbuild/**RPMS/noarch/*.rpm .
  sudo mv thingsboard-gateway-$CURRENT_VERSION-1.noarch.rpm python3-thingsboard-gateway.rpm
  sudo chown $CURRENT_USER. *.rpm
#  sudo apt install ./deb_dist/thingsboard-gateway-$CURRENT_VERSION/debian/python3-thingsboard-gateway.deb -y
#  echo "Building Docker image and container"
#  cp -r for_build/etc/thingsboard-gateway/config docker/
#  cp -r for_build/etc/thingsboard-gateway/extensions docker/
#  cd docker || exit
#  sudo docker build -t thingsboard_gateway .
#  sudo docker run -d -it --mount type=bind,source="$(pwd)""/logs",target=/var/log/thingsboard-gateway -v config:/etc/thingsboard-gateway/config -v extensions:/var/lib/thingsboard_gateway/extensions --name tb_gateway thingsboard_gateway
#  sudo docker ps -a | grep tb_gateway
fi

Then, I execute the following commands:
`sudo mkdir -p /root/rpmbuild/var/lib/thingsboard_gateway || echo "ThingsBoard user home directory already exists"`
`/usr/bin/getent passwd thingsboard_gateway || /usr/sbin/useradd -c "ThingsBoard-Gateway Service" -r -U -d /var/lib/thingsboard_gateway thingsboard_gateway && passwd -d thingsboard_gateway`
`/usr/bin/chown thingsboard_gateway:thingsboard_gateway /root/rpmbuild/var/lib/thingsboard_gateway -R`
Then I execute the make_packages.sh. The following error occurs:
+ exit 0
执行(%install): /bin/sh -e /var/tmp/rpm-tmp.AncoUF
+ umask 022
+ cd /root/rpmbuild/BUILD
+ '[' /root/rpmbuild/BUILDROOT/thingsboard-gateway-2.3.0-1.x86_64 '!=' / ']'
+ rm -rf /root/rpmbuild/BUILDROOT/thingsboard-gateway-2.3.0-1.x86_64
++ dirname /root/rpmbuild/BUILDROOT/thingsboard-gateway-2.3.0-1.x86_64
+ mkdir -p /root/rpmbuild/BUILDROOT
+ mkdir /root/rpmbuild/BUILDROOT/thingsboard-gateway-2.3.0-1.x86_64
+ find /root/rpmbuild/BUILDROOT/thingsboard-gateway-2.3.0-1.x86_64 -name .pyc -delete
+ /usr/lib/rpm/check-buildroot
+ /usr/lib/rpm/redhat/brp-compress
+ /usr/lib/rpm/redhat/brp-strip /usr/bin/strip
+ /usr/lib/rpm/redhat/brp-strip-comment-note /usr/bin/strip /usr/bin/objdump
+ /usr/lib/rpm/redhat/brp-strip-static-archive /usr/bin/strip
+ /usr/lib/rpm/brp-python-bytecompile /usr/bin/python 1
+ /usr/lib/rpm/redhat/brp-python-hardlink
+ /usr/lib/rpm/redhat/brp-java-repack-jars
处理文件：thingsboard-gateway-2.3.0-1.noarch
错误：没有找到文件：/root/rpmbuild/BUILDROOT/thingsboard-gateway-2.3.0-1.x86_64/etc/systemd/system/thingsboard-gateway.service
错误：没有找到目录：/root/rpmbuild/BUILDROOT/thingsboard-gateway-2.3.0-1.x86_64/etc/thingsboard-gateway
错误：没有找到目录：/root/rpmbuild/BUILDROOT/thingsboard-gateway-2.3.0-1.x86_64/var/log/thingsboard-gateway
错误：没有找到目录：/root/rpmbuild/BUILDROOT/thingsboard-gateway-2.3.0-1.x86_64/var/lib/thingsboard_gateway
警告：没有找到文件：/root/rpmbuild/BUILDROOT/thingsboard-gateway-2.3.0-1.x86_64/usr/local/lib/python3.8
警告：没有找到文件：/root/rpmbuild/BUILDROOT/thingsboard-gateway-2.3.0-1.x86_64/usr/local/bin/thingsboard-gateway

RPM 构建错误：
    没有找到文件：/root/rpmbuild/BUILDROOT/thingsboard-gateway-2.3.0-1.x86_64/etc/systemd/system/thingsboard-gateway.service
    没有找到目录：/root/rpmbuild/BUILDROOT/thingsboard-gateway-2.3.0-1.x86_64/etc/thingsboard-gateway
    没有找到目录：/root/rpmbuild/BUILDROOT/thingsboard-gateway-2.3.0-1.x86_64/var/log/thingsboard-gateway
    没有找到目录：/root/rpmbuild/BUILDROOT/thingsboard-gateway-2.3.0-1.x86_64/var/lib/thingsboard_gateway
    没有找到文件：/root/rpmbuild/BUILDROOT/thingsboard-gateway-2.3.0-1.x86_64/usr/local/lib/python3.8
    没有找到文件：/root/rpmbuild/BUILDROOT/thingsboard-gateway-2.3.0-1.x86_64/usr/local/bin/thingsboard-gateway
cp: 无法获取"/root/rpmbuild/RPMS/noarch/*.rpm" 的文件状态(stat): 没有那个文件或目录
mv: 无法获取"thingsboard-gateway-2.3.0.1-1.noarch.rpm" 的文件状态(stat): 没有那个文件或目录
chown: 无法访问"*.rpm": 没有那个文件或目录
[root@wlw153 tb-gateway]# cd /root/rpmbuild/BUILDROOT/thingsboard-gateway-2.3.0-1.x86_64/
[root@wlw153 thingsboard-gateway-2.3.0-1.x86_64]# ls
[root@wlw153 thingsboard-gateway-2.3.0-1.x86_64]# 
[root@wlw153 thingsboard-gateway-2.3.0-1.x86_64]# 
[root@wlw153 thingsboard-gateway-2.3.0-1.x86_64]# 
[root@wlw153 thingsboard-gateway-2.3.0-1.x86_64]# 
[root@wlw153 thingsboard-gateway-2.3.0-1.x86_64]#

zbeacon, 2020-09-09:
Hi everybody,

To solve this issue, before generation the package you have to has installed all required dependencies on your system, even the yours version of the gateway. Only after all dependencies will be installed in the system, you will be able to generate rpm package.

closed 2020-09-09 by zbeacon