thingsboard/thingsboard-gateway #311: How to generate DEB package by myself

oliver225, 2020-05-20:
I execute the make_packages.sh\ generate_deb_packages.sh.   but all fail

zbeacon, 2020-05-20:
Hi @oliver225 

Thank you for your interest in ThingsBoard IoT gateway.
Please paste there output from your console to investigate the issue.

oliver225, 2020-05-20:
## execute generate_deb_packages.sh ##
root@dell:/thingsboard-gateway# ./generate_deb_package.sh
grep: setup.cfg: No such file or directory
Installing libraries for building deb package.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
fakeroot is already the newest version (1.20.2-1ubuntu1).
python3-stdeb is already the newest version (0.8.5-1).
python-all is already the newest version (2.7.12-1~16.04).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
3 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] Y
Setting up grub-efi-amd64 (2.02~beta2-36ubuntu3.23) ...
dpkg: error processing package grub-efi-amd64 (--configure):
 subprocess installed post-installation script returned error exit status 10
Setting up rsyslog (8.16.0-1ubuntu3.1) ...
dpkg: error processing package rsyslog (--configure):
 subprocess installed post-installation script returned error exit status 10
dpkg: dependency problems prevent configuration of grub-efi-amd64-signed:
 grub-efi-amd64-signed depends on grub-efi-amd64 (= 2.02~beta2-36ubuntu3.23); however:
  Package grub-efi-amd64 is not configured yet.

dpkg: error processing package grub-efi-amd64-signed (--configure):
 dependency problems - leaving unconfigured
No apport report written because the error message indicates its a followup error from a previous failure.
                                                                                                          Errors were encountered while processing:
 grub-efi-amd64
 rsyslog
 grub-efi-amd64-signed
E: Sub-process /usr/bin/dpkg returned an error code (1)
Building DEB package
Creating sources for DEB package...
/usr/lib/python3.5/distutils/dist.py:261: UserWarning: Unknown distribution option: 'long_description_content_type'
  warnings.warn(msg)
/usr/lib/python3.5/distutils/dist.py:261: UserWarning: Unknown distribution option: 'python_requires'
  warnings.warn(msg)
running bdist_deb
running sdist_dsc
running egg_info
writing dependency_links to thingsboard_gateway.egg-info/dependency_links.txt
writing thingsboard_gateway.egg-info/PKG-INFO
writing top-level names to thingsboard_gateway.egg-info/top_level.txt
writing requirements to thingsboard_gateway.egg-info/requires.txt
writing entry points to thingsboard_gateway.egg-info/entry_points.txt
reading manifest file 'thingsboard_gateway.egg-info/SOURCES.txt'
writing manifest file 'thingsboard_gateway.egg-info/SOURCES.txt'
running sdist
warning: sdist: standard file not found: should have one of README, README.rst, README.txt

running check
creating thingsboard-gateway-2.3.1
creating thingsboard-gateway-2.3.1/thingsboard_gateway
creating thingsboard-gateway-2.3.1/thingsboard_gateway.egg-info
creating thingsboard-gateway-2.3.1/thingsboard_gateway/connectors
creating thingsboard-gateway-2.3.1/thingsboard_gateway/connectors/bacnet
creating thingsboard-gateway-2.3.1/thingsboard_gateway/connectors/bacnet/bacnet_utilities
creating thingsboard-gateway-2.3.1/thingsboard_gateway/connectors/ble
creating thingsboard-gateway-2.3.1/thingsboard_gateway/connectors/can
creating thingsboard-gateway-2.3.1/thingsboard_gateway/connectors/modbus
creating thingsboard-gateway-2.3.1/thingsboard_gateway/connectors/mqtt
creating thingsboard-gateway-2.3.1/thingsboard_gateway/connectors/odbc
creating thingsboard-gateway-2.3.1/thingsboard_gateway/connectors/opcua
creating thingsboard-gateway-2.3.1/thingsboard_gateway/connectors/request
creating thingsboard-gateway-2.3.1/thingsboard_gateway/extensions
creating thingsboard-gateway-2.3.1/thingsboard_gateway/extensions/bacnet
creating thingsboard-gateway-2.3.1/thingsboard_gateway/extensions/ble
creating thingsboard-gateway-2.3.1/thingsboard_gateway/extensions/can
creating thingsboard-gateway-2.3.1/thingsboard_gateway/extensions/modbus
[... 136 lines cut ...]
    self.distribution.run_command(command)
  File "/usr/lib/python3.5/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/usr/lib/python3/dist-packages/stdeb/command/sdist_dsc.py", line 140, in run
    remove_expanded_source_dir=self.remove_expanded_source_dir,
  File "/usr/lib/python3/dist-packages/stdeb/util.py", line 1397, in build_dsc
    dpkg_genchanges(cwd=fullpath_repackaged_dirname)
  File "/usr/lib/python3/dist-packages/stdeb/util.py", line 538, in dpkg_genchanges
    process_command(args,cwd=cwd)
  File "/usr/lib/python3/dist-packages/stdeb/util.py", line 183, in process_command
    check_call(args, cwd=cwd)
  File "/usr/lib/python3/dist-packages/stdeb/util.py", line 46, in check_call
    raise CalledProcessError(retcode)
stdeb.util.CalledProcessError: 2
Adding the files, scripts and permissions in the package
cp: cannot create directory 'deb_dist/thingsboard-gateway-/debian/python3-thingsboard-gateway': No such file or directory
cp: cannot create directory 'deb_dist/thingsboard-gateway-/debian/python3-thingsboard-gateway': No such file or directory
cp: cannot create directory 'deb_dist/thingsboard-gateway-/debian/python3-thingsboard-gateway': No such file or directory
chown: cannot access 'deb_dist/thingsboard-gateway-/debian/python3-thingsboard-gateway/': No such file or directory
chown: cannot access 'deb_dist/thingsboard-gateway-/debian/python3-thingsboard-gateway/var/': No such file or directory
chmod: cannot access 'deb_dist/thingsboard-gateway-/debian/python3-thingsboard-gateway/DEBIAN/preinst': No such file or directory
[... cut ...] /python3-thingsboard-gateway/DEBIAN/preinst': No such file or directory
./generate_deb_package.sh: line 40: deb_dist/thingsboard-gateway-/debian/python3-thingsboard-gateway/DEBIAN/control: No such file or directory
dpkg-deb: error: failed to open package info file 'deb_dist/thingsboard-gateway-/debian/python3-thingsboard-gateway//DEBIAN/control' for reading: No such file or directory
cp: cannot stat 'deb_dist/thingsboard-gateway-/debian/python3-thingsboard-gateway.deb': No such file or directory

zbeacon, 2020-05-20:
Try to run the bash script using sudo command instead of root console.

oliver225, 2020-05-20:
but the Setup.cfg file not found

oliver225, 2020-05-20:
create thingsboard_gateway user
then on thingsboard_gateway's  home directory run this script

zbeacon, 2020-05-20:
@oliver225 
Oh, my bad, you are right. I commited the new version of the script in the master branch.
Please try it and let me know about results.

oliver225, 2020-05-21:
oliver@iZbp17xad8flexazzc21vjZ:/home/oliver/thingsboard-gateway$ ./generate_deb_package.sh 
Installing libraries for building deb package.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
fakeroot is already the newest version (1.20.2-1ubuntu1).
python3-stdeb is already the newest version (0.8.5-1).
python-all is already the newest version (2.7.12-1~16.04).
0 upgraded, 0 newly installed, 0 to remove and 97 not upgraded.
Building DEB package
Creating sources for DEB package...
/usr/lib/python3.5/distutils/dist.py:261: UserWarning: Unknown distribution option: 'python_requires'
  warnings.warn(msg)
/usr/lib/python3.5/distutils/dist.py:261: UserWarning: Unknown distribution option: 'long_description_content_type'
  warnings.warn(msg)
running bdist_deb
running sdist_dsc
running egg_info
writing entry points to thingsboard_gateway.egg-info/entry_points.txt
writing thingsboard_gateway.egg-info/PKG-INFO
writing requirements to thingsboard_gateway.egg-info/requires.txt
writing top-level names to thingsboard_gateway.egg-info/top_level.txt
writing dependency_links to thingsboard_gateway.egg-info/dependency_links.txt
reading manifest file 'thingsboard_gateway.egg-info/SOURCES.txt'
writing manifest file 'thingsboard_gateway.egg-info/SOURCES.txt'
running sdist
warning: sdist: standard file not found: should have one of README, README.rst, README.txt

running check
creating thingsboard-gateway-2.3.1
creating thingsboard-gateway-2.3.1/thingsboard_gateway
creating thingsboard-gateway-2.3.1/thingsboard_gateway.egg-info
creating thingsboard-gateway-2.3.1/thingsboard_gateway/connectors
creating thingsboard-gateway-2.3.1/thingsboard_gateway/connectors/bacnet
creating thingsboard-gateway-2.3.1/thingsboard_gateway/connectors/bacnet/bacnet_utilities
creating thingsboard-gateway-2.3.1/thingsboard_gateway/connectors/ble
creating thingsboard-gateway-2.3.1/thingsboard_gateway/connectors/can
creating thingsboard-gateway-2.3.1/thingsboard_gateway/connectors/modbus
creating thingsboard-gateway-2.3.1/thingsboard_gateway/connectors/mqtt
creating thingsboard-gateway-2.3.1/thingsboard_gateway/connectors/odbc
creating thingsboard-gateway-2.3.1/thingsboard_gateway/connectors/opcua
creating thingsboard-gateway-2.3.1/thingsboard_gateway/connectors/request
creating thingsboard-gateway-2.3.1/thingsboard_gateway/extensions
creating thingsboard-gateway-2.3.1/thingsboard_gateway/extensions/bacnet
creating thingsboard-gateway-2.3.1/thingsboard_gateway/extensions/ble
creating thingsboard-gateway-2.3.1/thingsboard_gateway/extensions/can
creating thingsboard-gateway-2.3.1/thingsboard_gateway/extensions/modbus
creating thingsboard-gateway-2.3.1/thingsboard_gateway/extensions/mqtt
creating thingsboard-gateway-2.3.1/thingsboard_gateway/extensions/odbc
creating thingsboard-gateway-2.3.1/thingsboard_gateway/extensions/opcua
creating thingsboard-gateway-2.3.1/thingsboard_gateway/extensions/request
creating thingsboard-gateway-2.3.1/thingsboard_gateway/extensions/serial
creating thingsboard-gateway-2.3.1/thingsboard_gateway/gateway
creating thingsboard-gateway-2.3.1/thingsboard_gateway/storage
creating thingsboard-gateway-2.3.1/thingsboard_gateway/tb_client
creating thingsboard-gateway-2.3.1/thingsboard_gateway/tb_utility
making hard links in thingsboard-gateway-2.3.1...
hard linking setup.py -> thingsboard-gateway-2.3.1
hard linking thingsboard_gateway/__init__.py -> thingsboard-gateway-2.3.1/thingsboard_gateway
hard linking thingsboard_gateway/tb_gateway.py -> thingsboard-gateway-2.3.1/thingsboard_gateway
hard linking thingsboard_gateway.egg-info/PKG-INFO -> thingsboard-gateway-2.3.1/thingsboard_gateway.egg-info
hard linking thingsboard_gateway.egg-info/SOURCES.txt -> thingsboard-gateway-2.3.1/thingsboard_gateway.egg-info
hard linking thingsboard_gateway.egg-info/dependency_links.txt -> thingsboard-gateway-2.3.1/thingsboard_gateway.egg-info
hard linking thingsboard_gateway.egg-info/entry_points.txt -> thingsboard-gateway-2.3.1/thingsboard_gateway.egg-info
hard linking thingsboard_gateway.egg-info/requires.txt -> thingsboard-gateway-2.3.1/thingsboard_gateway.egg-info
hard linking thingsboard_gateway.egg-info/top_level.txt -> thingsboard-gateway-2.3.1/thingsboard_gateway.egg-info
hard linking thingsboard_gateway/connectors/__init__.py -> thingsboard-gateway-2.3.1/thingsboard_gateway/connectors
hard linking thingsboard_gateway/connectors/connector.py -> thingsboard-gateway-2.3.1/thingsboard_gateway/connectors
hard linking thingsboard_gateway/connectors/converter.py -> thingsboard-gateway-2.3.1/thingsboard_gateway/connectors
hard linking thingsboard_gateway/connectors/bacnet/__init__.py -> thingsboard-gateway-2.3.1/thingsboard_gateway/connectors/bacnet
[... 56 lines cut ...]
hard linking thingsboard_gateway/gateway/tb_gateway_service.py -> thingsboard-gateway-2.3.1/thingsboard_gateway/gateway
hard linking thingsboard_gateway/gateway/tb_logger.py -> thingsboard-gateway-2.3.1/thingsboard_gateway/gateway
hard linking thingsboard_gateway/gateway/tb_updater.py -> thingsboard-gateway-2.3.1/thingsboard_gateway/gateway
hard linking thingsboard_gateway/storage/__init__.py -> thingsboard-gateway-2.3.1/thingsboard_gateway/storage
hard linking thingsboard_gateway/storage/event_storage.py -> thingsboard-gateway-2.3.1/thingsboard_gateway/storage
hard linking thingsboard_gateway/storage/event_storage_files.py -> thingsboard-gateway-2.3.1/thingsboard_gateway/storage
[... cut ...]
removing 'thingsboard-gateway-2.3.1' (and everything under it)
error: [Errno 13] Permission denied: 'PKG-INFO'
Adding the files, scripts and permissions in the package
./generate_deb_package.sh: line 40: deb_dist/thingsboard-gateway-2.3.1/debian/python3-thingsboard-gateway/DEBIAN/control: Permission denied
dpkg-deb: error: no package information in 'deb_dist/thingsboard-gateway-2.3.1/debian/python3-thingsboard-gateway//DEBIAN/control'
cp: cannot stat 'deb_dist/thingsboard-gateway-2.3.1/debian/python3-thingsboard-gateway.deb': No such file or directory

oliver225, 2020-05-21:
we urgently need a deb package on our equipment  support KEPServer EX6
please could you send me deb package on master version?
thank you very much

zbeacon, 2020-05-21:
Hi @oliver225 

Thank you for your feedback, I will investigate this issue deeply. I published pre-release 2.3.1.1, with your improvements, you can use deb or rpm package from it.

oliver225, 2020-05-21:
thank you very much

qwyinxiang, 2020-05-22:
@oliver225 , I generate DEB package successful.
You should add execute permission to the file /for_build/DEBIAN/postinst.

oliver225, 2020-05-24:
@qwyinxiang 
chmod 777 postinet?

qwyinxiang, 2020-05-25:
chmod +x postinst

closed 2020-07-03 by zbeacon