thingsboard/thingsboard-gateway #313: [HELP]Thingsboard Server doesn't get updated with Thingsboard gateway MQTT telemetry Data

pamudithaA, 2020-05-25:
I have configured IOT gateway and Thingsboard server. 
IOT gateway is brokering Mosquitto MQTT client. 
Now from IOT gateway has successfully subscribed to topics and keep sending mqtt messages. 
I have verified it from logs as well as from a pcap in the thingsboard server side. 
But device telemetry doesn't get updated at all. 
Image attach from the pcap. 

[image]

Sample from connector.log of iot gateway is below. 

''2020-05-25 14:18:28' - DEBUG - mqtt_connector - 342 - Successfully converted message from topic AJXWI7JX/devices/wb-map12h_30/controls/Urms L3'
''2020-05-25 14:18:29' - DEBUG - custom_mqtt_uplink_converter - 27 - New data received: AJXWI7JX/devices/wb-adc/controls/R1: nan'
''2020-05-25 14:18:29' - DEBUG - custom_mqtt_uplink_converter - 37 - Result: {'deviceName': 'wb-adc', 'deviceType': 'buzzer', 'telemetry': [{'R1': 'nan'}]}'
''2020-05-25 14:18:29' - DEBUG - mqtt_connector - 342 - Successfully converted message from topic AJXWI7JX/devices/wb-adc/controls/R1'
''2020-05-25 14:18:29' - DEBUG - custom_mqtt_uplink_converter - 27 - New data received: AJXWI7JX/devices/wb-adc/controls/Vin: 23.86'
''2020-05-25 14:18:29' - DEBUG - custom_mqtt_uplink_converter - 37 - Result: {'deviceName': 'wb-adc', 'deviceType': 'buzzer', 'telemetry': [{'Vin': 23.86}]}'
''2020-05-25 14:18:29' - DEBUG - mqtt_connector - 342 - Successfully converted message from topic AJXWI7JX/devices/wb-adc/controls/Vin'

Any help would be highly appreciated.

zbeacon, 2020-05-25:
Hi @pamudithaA 

Thank you for your interest in ThingsBoard IoT gateway. Please attach connector.log and service.log file for investigations.

pamudithaA, 2020-05-25:
Okay. I will attach. Btw, should Access token be sent on every mqtt publish message?

zbeacon, 2020-05-26:
@pamudithaA 

Yes, it is the same like you use mosquitto_pub and give the -u (user) -P (password) parameters.

closed 2020-06-16 by zbeacon