thingsboard/thingsboard-gateway #314: [HELP]Dynamic reload custom connector configuration

bloooood, 2020-06-04:
**Describe the issue**
I am using custom connector feature, the feature is fine.  Just want to check if it is possible to reload the configuration json file or convertor without restart gateway.  The idea is to add devices with their convertors seamlessly.  Thank you.

zbeacon, 2020-06-04:
Hi @jcao000 

Thank you for your interest in the ThingsBoard IoT gateway.
We have not released this feature yet, but you can reload the gateway remotely, using the RPC command `gateway_restart` to the gateway device from the ThingsBoard. [Read more](https://thingsboard.io/docs/iot-gateway/guides/how-to-use-gateway-rpc-methods/).

If you want - you can add some function to the tb_gateway_service.py file and create a Pull Request to add this functional into the gateway.

jcao000, 2020-06-04:
Thanks a lot @zbeacon . will try the gateway RPC

closed 2020-06-04 by jcao000

subtux, 2022-05-05:
Hello @imbeacon 

Is it possible to use the gateway_restart command to use in the RPC Button widget, when i press the button the gateway restarts?