thingsboard/thingsboard-gateway #315: [HELP] Dictionary changed size during iteration when using RPC

sobehard, 2020-06-10:
**Describe the issue**
There is some issue for the gateway when testing the RPC button at thingsboard dasboard. When the button is pressed for few times, the mqtt broker connector of the thingsboard gateway will closed connection and the gateway stop pushing data to the thingsboard server. And the thingsboard gateway does not connect back mqtt broker connecto, so I will have to restart the thingsboard gateway to let it restart. Can I know how to solve this?

Below is the button type and configuration at thingsboard dashboard.
[image]

[image]

**Error traceback (If it was raised):**
```
''2020-06-03 21:31:28' - ERROR - tb_gateway_service - 184 - dictionary changed size during iteration'
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/thingsboard_gateway/gateway/tb_gateway_service.py", line 159, in __init__
    new_rpc_request_in_progress = {key: value for key, value in self.__rpc_requests_in_progress.items() if value != 'del'}
  File "/usr/local/lib/python3.6/dist-packages/thingsboard_gateway/gateway/tb_gateway_service.py", line 159, in <dictcomp>
    new_rpc_request_in_progress = {key: value for key, value in self.__rpc_requests_in_progress.items() if value != 'del'}
RuntimeError: dictionary changed size during iteration
```

**Versions:**
 - OS: Ubuntu 18.04
 - Thingsboard IoT Gateway version 2.3.1.1
 - Python version 3.6

zbeacon, 2020-06-16:
Hi @sobehard 

I'm sorry for long term answer, it is a bug with RPC processing. I've fixed it in the develop branch.
Please use the version from the develop branch or wait for the next release.

sobehard, 2020-06-17:
Alright.

closed 2020-07-01 by zbeacon