thingsboard/thingsboard-gateway #316: [Help] installation fails in Raspberry pi python3-paho-mqtt

hasikp1, 2020-06-12:
I am trying to install the thingsboard gateway in Raspberry pi using following guide.
[installing gateway as package ](https://thingsboard.io/docs/iot-gateway/install/deb-installation/)
I am receiving following error.

>sudo apt install ./python3-thingsboard-gateway.deb -y
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'python3-thingsboard-gateway' instead of './python3-thingsboard-gateway.deb'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:
The following packages have unmet dependencies:
 python3-thingsboard-gateway : Depends: python3-paho-mqtt but it is not installable
E: Unable to correct problems, you have held broken packages.

Os information is following

> NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"

hasikp1, 2020-06-12:
Fixed after manual installation of python3-paho-mqtt

closed 2020-06-12 by hasikp1

KarimSamehSadek, 2021-12-09:
Same issue persists for me even after manual installation of python3-paho-mqtt