thingsboard/thingsboard-gateway #317: [HELP]

giuseppe-rana-83, 2020-06-12:
Hi TB's guys,
 
I am experiencing some troubles running an RTU serial modbus connection after having installed the TB gateway on a Raspberry pi 3. In particular, I tried to communicate with a modbus serial device configured as follows: 

{
  "server": {
    "name": "Modbus Default Server",
    "type": "serial",
    "method": "rtu",
    "port": "/dev/ttyUSB0",
    "baudrate": 9600,
    "stopbits": 1,
    "bytesize": 8,
    "parity": "N",
    "strict": "true",
    "timeout": 35,
    "byteOrder": "BIG",
    "devices": [
      {
        "unitId": 1,
        "deviceName": "Smart Meter",
        "attributesPollPeriod": 5000,
        "timeseriesPollPeriod": 5000,
        "sendDataOnlyOnChange": true,
        "attributes": [
          {
            "tag": "Line Frequency",
            "type": "bytes",
            "functionCode": 4,
            "objectsCount": 1,
            "address": 4134 
          }
      ],
      "timeseries": {
            "tag": "Line 1 Voltage",
            "type": "16uint",
            "functionCode": 3,
            "objectsCount": 1,
            "address": 4096
          }
         ],
        "attributeUpdates": [
          {
            "tag": "shared_attribute_write",
            "type": "bytes",
            "functionCode": 6,
            "objectsCount": 2,
            "address": 29
          }
        ],
        "rpc": [
          {
            "tag": "setValue",
            "type": "bits",
            "functionCode": 5,
            "objectsCount": 1,
            "address": 31
         }
        ]
      } 
    ]
  }
}

The following error log is shown on the shell:

Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/thingsboard_gateway-2.3.1.1-py3.7.egg/thingsboard_gateway/connectors/modbus/bytes_modbus_uplink_converter.py", line 58, in convert
    registers = response.registers
AttributeError: 'str' object has no attribute 'registers'
""2020-06-12 16:19:58" - ERROR - [bytes_modbus_uplink_converter.py] - bytes_modbus_uplink_converter - 79 - 'str' object has no attribute 'registers'"
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/thingsboard_gateway-2.3.1.1-py3.7.egg/thingsboard_gateway/connectors/modbus/bytes_modbus_uplink_converter.py", line 58, in convert
    registers = response.registers
AttributeError: 'str' object has no attribute 'registers'
""2020-06-12 16:20:03" - ERROR - [modbus_connector.py] - modbus_connector - 262 - Exception Response(132, 4, IllegalFunction)"
NoneType: None
""2020-06-12 16:20:03" - ERROR - [bytes_modbus_uplink_converter.py] - bytes_modbus_uplink_converter - 79 - 'str' object has no attribute 'registers'"

I'm working with the following OS and software versions:  

 - OS: Raspbian version 10 (buster)
 - Thingsboard IoT Gateway version 2.3.1.1
 - Python version 3.7.3

zbeacon, 2020-06-16:
Hi @giuseppe-rana-83 

Thank you for your interest in ThingsBoard IoT gateway.

It is an issue with the Modbus server, by some reasons it sent error **IllegalFunction** - It can be caused by wrong number of the register or some different problems. Please try to read some else register. Also please give more information about server, you used. I will try to repeat the issue in my environment.

closed 2020-07-01 by zbeacon

shengpj, 2020-12-08:
[image]

shengpj, 2020-12-08:
> Hi TB's guys,
>
> I am experiencing some troubles running an RTU serial modbus connection after having installed the TB gateway on a Raspberry pi 3. In particular, I tried to communicate with a modbus [...]
I also met this problem. How did you solve it

shengpj, 2020-12-08:
I also met this problem. How did you solve it

divyasakre, 2023-08-28:
facing same issue. Post the solution to it.