thingsboard/thingsboard-gateway #318: [HELP]

zdenoL, 2020-06-16:
**Connection to external MQTT with SSL/TLS without need to updload certificate**
Is there a chance to use gateway and mace secured connection to mqtt broker via SSL, port 8883
and without need to upload certificate?
as certificate on MQTT is lets enctypt, expiration period is 3 months and refreshing it willl be overkill, so I am curious if there is some chance to put something like

**"ssl":true** 

**Configuration** 
{
  "broker": {
    "name": "iot_MQTT",
    "host": "mqtt.xxxxxx.sk",
    "port": 8883,
    "security": {
      "type": "basic",
      "username": "abcabc",
      "password": "abcabc"
    }
  },

**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 20.04] - dockerized
-version of gatweay: most recent

Thanks a lot

zbeacon, 2020-06-17:
Hi @zdenoL 

Thank you for your interest in the ThingsBoard IoT gateway.
To use just SSL host certificate validation without certificates, you can change the type to the "tls" and try to run. The gateway will try to use the SSL context with TLS version 1.2.

zdenoL, 2020-06-17:
If I get you right, something like this:

Configuration
{
"broker": {
"name": "iot_MQTT",
"host": "mqtt.xxxxxx.sk",
"port": 8883,
"security": {
**"type": "tls",**
"username": "abcabc",
"password": "abcabc"
}
}

Anyway, great work!!! far better than former java version.

zbeacon, 2020-06-17:
Yep, please try and let me know about the result.

Thank you :smile:

zdenoL, 2020-06-17:
hey, half good news :)
Connestion established!!!!
anyway, with following setup I gave got strange TB behaviour:

{
  "broker": {
    "name": "iot_MQTT",
    "host": "mqtt.*******.sk",
    "port": 8883,
    "security": {
      "type": "tls",
      "username": "user",
      "password": "password"
    }
  },
  "mapping": [
    {
      "topicFilter": "gastanks/+/telemetry",
      "converter": {
        "type": "json",
        "deviceNameTopicExpression": "(?<=gastanks\/)(.*?)(?=\/telemetry)",
        "deviceTypeTopicExpression": "gastank",
        "timeout": 60000,
        "attributes": [
          {
            "type": "string",
            "key": "model",
            "value": "${sensorModel}"
          }
        ],
        "timeseries": [
          {
            "type": "double",
            "key": "temperature",
            "value": "${temperature}"
          },
          {
            "type": "double",
            "key": "humidity",
            "value": "${humidity}"
          }
        ]
      }
    }
  ],
  "connectRequests": [
    {
      "topicFilter": "gastanks/+/connect",
      "deviceNameTopicExpression": "(?<=gastanks\/)(.*?)(?=\/connect)"
    }
  ]
}

Device is however, created with type DEFAULT, even when I set that it should be gastank.
And back to telemetries:
I see in TB, that they are saved with Key=25.5 and valiue = 25.5 and second key=36.1 and value=36.1

am I missing something important?  the same setup of converter  when sent via basic security and 1883, worked well, but I can check that once again if needed

zbeacon, 2020-06-17:
deviceTypeTopicExpression - uses to get the device type from the topic(It use only regex to find the type). Replace this parameter to deviceTypeJsonExpression, remove the device from the ThingsBoard and try again to setup the correct type. It is a bug, I will fix it.

It caused because the key is the same like in the incoming message, to solve it now, just set the key name not the same like in the incoming message. I also will fix this.

zdenoL, 2020-06-17:
OK, renaming the property name in telemetry helped, please make note for future.
Device type is after change from deviceTypeTopicExpression  to deviceTypeJsonExpression and placing value "gastank" still set as default. BUt I was able to change it in TB. Anyway, would be nice to fetch it from configuration file as constant parameter.
thanks a lot for support.. :)

closed 2020-06-17 by zdenoL