thingsboard/thingsboard-gateway #320: [BUG] ThingsBoard IoT Gateway stalled on empty messages

viktorkho, 2020-06-24:
**Describe the bug**
ThingsBoard IoT Gateway with custom converter (the same as in #299) stalled if I publish messages is empty.

How to reproduce:
```
$ mosquitto_pub -t AJXWI7JX/devices/wb-adc/controls/5Vout -m "5"  # INFO Successfully converted
$ mosquitto_pub -t AJXWI7JX/devices/wb-adc/controls/5Vout -m ""   # Nothing in log
$ mosquitto_pub -t AJXWI7JX/devices/wb-adc/controls/5Vout -m "5"  # Nothing in log
```

**Connector name (If bug in the some connector):**
MQTT Connector

Error traceback (If available):
Nothing in docker container's output after message with spaces published to the topic (to which it is subscribed)

Versions (please complete the following information):

docker-compose version 1.17.1
image: thingsboard/tb-gateway:latest
Python 3.6.9

zbeacon, 2020-06-24:
Hi @viktorkho 

I have changed the tb_utility to solve this issue, please try to use it.

closed 2020-06-24 by zbeacon

zbeacon, 2020-06-24:
if you will have any problem with this issue, just reopen it.

viktorkho, 2020-06-24:
Perfect! Thank you, @zbeacon!

commit 8d82b9f1 (2020-06-24): Fix #320