thingsboard/thingsboard-gateway #322: MQTT config [HELP]

Patriciawa, 2020-06-24:
Hi, 

I'm having some configuration issues of my MQTT.json which I can't solve on my own, I am probably doing something wrong here,
Here's my config:

`
{
  "broker": {
    "name":"Default Local Broker",
    "host":"127.0.0.1",
    "port":1883,
    "security": {
      "type": "anonymous"
    }
  },
  "mapping": [
    {
      "topicFilter": "homie/patrick/+/onoff",
      "converter": {
        "type": "json",
        "deviceNameJsonExpression": "(?<=homie\/)(.*?)(?=\/onoff)",
        "deviceTypeJsonExpression": "(?<=/\/)(.*?)(?=\/onoff)",
        "timeout": 60000,
        "attributes": [
          {
            "type": "string",
            "key": "model",
            "value": "${sensorModel}"
          }
        ],
        "timeseries": [
          {
            "type": "double",
            "key": "temperature",
            "value": "${temp}"
          },
          {
            "type": "double",
            "key": "humidity",
            "value": "${hum}"
          }
        ],
  "connectRequests": [
    {
      "topicFilter": "homie/patrick/+/onoff",
      "deviceNameJsonExpression": "(?<=homie\/)(.*?)(?=\/onoff)"
    }
  ]
      }
    }
  ]
}
`

When I turn a light on or off (onoff) I can see the telemetery values raise (the eventsProduced & eventsSend), but the MQTT devices won't show up in the devices' section of the ThingsBoard Web UI, any thoughts?

 - OS:  Ubuntu 18.04 LTS
 - Thingsboard IoT Gateway version  2.4.0
 - Python version 3.7.3

zbeacon, 2020-06-24:
Hi @Patriciawa  
Thank you for your interest in the ThingsBoard IoT gateway. 
It looks like you have some issue with deviceTypeTopicExpression. Please try to change it to "default" and restart the gateway.

zbeacon, 2020-06-24:
Hi @Patriciawa 

Thank you for your interest in the ThingsBoard IoT gateway.

The issue is caused by wrong parameter names - deviceNameJsonExpression and deviceTypeJsonExpression are used to parse data from the incoming messages. To solve the problem just replace these parameters with following deviceNameTopicExpression and deviceTypeTopicExpression.

zbeacon, 2020-06-24:
Also I recommend to replace the expression for device type, it looks wrong.

Patriciawa, 2020-06-24:
Hi @zbeacon 
Thanks for your quick response! I changed both expressions to `deviceNameTopicExpression` and `deviceTypeTopicExpression`, still no luck. Indeed the expression didn't look good, I've changed it back to `(?<=homie\/)(.*?)(?=\/onoff)`. But also did not work.

zbeacon, 2020-06-24:
Okay, try to use "deviceTypeNameExpression": "Test device" and "deviceTypeJsonExpression": "default" for test purposes.
If it will create a device and send some telemetry, the issue in the regular expressions. Anyway please also look for any exceptions in the logs. They are in the folder /var/log/thingsboard_gateway/ , if you installed the gateway as a daemon. In other case they should be in the folder, where you started the gateway.

Patriciawa, 2020-06-24:
Ok, so the connector log says i'm succesfully connected. Converter says this:
`""2020-06-24 13:14:07" - ERROR - [json_mqtt_uplink_converter.py] - json_mqtt_uplink_converter - 43 - The expression for looking "deviceName" not found in config {"type": "json", "deviceTypeNameExpression": "Test device", "deviceTypeJsonExpression": "de$`
This is the service log: `""2020-06-24 13:14:07" - ERROR - [tb_utility.py] - tb_utility - 62 - deviceName is empty in data: {"deviceName": null, "deviceType": "default", "attributes": [], "telemetry": []}"
""2020-06-24 13:14:07" - ERROR - [tb_gateway_service.py] - tb_gateway_service - 304 - Data from MQTT Broker Connector connector is invalid."
`

zbeacon, 2020-06-25:
Hi @Patriciawa 

Please try the following configuration:

```

{
  "broker": {
    "name": "Default Local Broker",
    "host": "127.0.0.1",
    "port": 1883,
    "security": {
      "type": "anonymous"
    }
  },
  "mapping": [
    {
      "topicFilter": "homie/patrick/+/onoff",
      "converter": {
        "type": "json",
        "deviceNameTopicExpression": "(?<=homie/patrick/)(.*?)(?=/onoff)",
        "deviceTypeTopicExpression": "(?<=homie/patrick/)(.*?)(?=/onoff)",
        "timeout": 60000,
        "attributes": [
          {
            "type": "string",
            "key": "model",
            "value": "${sensorModel}"
          }
        ],
        "timeseries": [
          {
            "type": "double",
            "key": "temperature",
            "value": "${temp}"
          },
          {
            "type": "double",
            "key": "humidity",
            "value": "${hum}"
          }
        ],
        "connectRequests": [
          {
            "topicFilter": "homie/patrick/+/onoff",
            "deviceNameJsonExpression": "(?<=homie/)(.*?)(?=/onoff)"
          }
        ]
      }
    }
  ]
}

```

Patriciawa, 2020-06-25:
Hi @zbeacon Thanks, it's almost working, they won't show up but in the service.log I can see the following:
`""2020-06-25 12:35:26" - ERROR - [tb_utility.py] - tb_utility - 62 - No telemetry and attributes in data: {"deviceName": "iks05", "deviceType": "iks05", "attributes": [], "telemetry": []}"`
This is just one device, but the error appears for all devices.

Patriciawa, 2020-06-25:
Let me just say to read the status it's simply /homie/patrick/devicename/onoff, which returns a value of true or false.
When you want to change the value (e.g. turn the light off),  the topic becomes /homie/patrick/devicename/onoff/set and in the payload you give it a false to turn it off or a true to turn it on.

zbeacon, 2020-06-25:
If the topic /homie/patrick/devicename/onoff sends only true or false, the default converter cannot convert it and you have an error. To parse this, you should write a custom converter, which will read the data from the message and interpret it as you want.

/homie/patrick/devicename/onoff/set - This is more simple to realise - you should create an RPC switch widget and try to use the following configuration for RPC section:

 ```

  "serverSideRpc": [
    {
      "deviceNameFilter": ".*",
      "methodFilter": "getValue",
      "requestTopicExpression": "/homie/patrick/devicename/onoff/",
      "valueExpression": "${params}"
    },
    {
      "deviceNameFilter": ".*",
      "methodFilter": "setValue",
      "requestTopicExpression": "/homie/patrick/devicename/onoff/set",
      "responseTopicExpression": "/homie/patrick/devicename/onoff/set",
      "responseTimeout": 10000,
      "valueExpression": "${params}"
    }
  ]

```

Patriciawa, 2020-06-25:
Okay thanks! that'll do!

closed 2020-06-25 by Patriciawa