thingsboard/thingsboard-gateway #324: [HELP] How to insert more than on device in MODBUS-TCP.JSON

pehedima, 2020-06-25:
Hi,
in the past we only inserts one electric meter device in one JSON file and inserts that to tb_gateway.yaml.
Now we want to read out 40 electric meters. 

Witch syntax is the correct one to insert more than one meter?

Here an example for one meter :

```
{
  "server": {
    "name": "electricity meters",
    "type": "tcp",
    "host": "192.168.178.19",
    "port": 502,
    "timeout": 35,
    "method": "socket",
    "devices": [
      {
        "unitId": 255,
        "deviceName": "P-Home-revpi-python-balena",
        "attributesPollPeriod": 3600000,
        "timeseriesPollPeriod": 5000,
        "sendDataOnlyOnChange": false,
        "attributes": [
            {
               "tag": "SerialNumber",
               "type": "string",
               "functionCode": 3,
               "address": 1280,
               "registerCount": 6,
               "byteOrder": "BIG"
            }
        ],
        "timeseries":[
            {
               "tag":"U_L1_V",
               "type":"double",
               "functionCode":3,
               "address":4096,
               "registerCount":2,
               "byteOrder":"BIG"
            }
        ]
      }
    ]
  }
}
```
Thanks for your help

zbeacon, 2020-06-25:
Hi @pehedima,

You can add one more device object into "devices" array. E.g.

```
{
  "server": {
    "name": "electricity meters",
    "type": "tcp",
    "host": "192.168.178.19",
    "port": 502,
    "timeout": 35,
    "method": "socket",
    "devices": [
      {
        "unitId": 255,
        "deviceName": "P-Home-revpi-python-balena",
        "attributesPollPeriod": 3600000,
        "timeseriesPollPeriod": 5000,
        "sendDataOnlyOnChange": false,
        "attributes": [
            {
               "tag": "SerialNumber",
               "type": "string",
               "functionCode": 3,
               "address": 1280,
               "registerCount": 6,
               "byteOrder": "BIG"
            }
        ],
        "timeseries":[
            {
               "tag":"U_L1_V",
               "type":"double",
               "functionCode":3,
               "address":4096,
               "registerCount":2,
               "byteOrder":"BIG"
            }
        ]
      },
      {
        "unitId": 254,
        "deviceName": "P-Home-revpi-python-balena-2",
        "attributesPollPeriod": 3600000,
        "timeseriesPollPeriod": 5000,
        "sendDataOnlyOnChange": false,
        "attributes": [
            {
               "tag": "SerialNumber",
               "type": "string",
               "functionCode": 3,
               "address": 1280,
               "registerCount": 6,
               "byteOrder": "BIG"
            }
        ],
        "timeseries":[
            {
               "tag":"U_L1_V",
               "type":"double",
               "functionCode":3,
               "address":4096,
               "registerCount":2,
               "byteOrder":"BIG"
            }
        ]
      }
    ]
  }
}

```

pehedima, 2020-06-25:
Thanks for your answer.

The second device has IP: 192.168.178.20 and the unitId is 255 (fix).
How is the syntax in this case?

zbeacon, 2020-06-25:
@pehedima  

In this case you should create another one modbus connector, for the second device with a different modbus.json configuration file and connect it in the tb_gateway.yaml file.

pehedima, 2020-06-25:
Thanks.
That means, I have to insert 40 files in one tb_gateway.yaml?
I' ll test that.

We have the 'old' Java version running on other 'older' devices. There it is possible to insert more than one MODBUS-TCP device in one file. 

Do you think, this will be possible for the python solution also in the future?

Thank you for your help.

zbeacon, 2020-06-25:
I agree with you, I didn't think about the devices with different ip addresses for one connector. I think we can solve this situation by adding the optional parameters to the devices "host" and "port". I will write here when they will be added.

zbeacon, 2020-09-11:
Hi @pehedima ,

Finally this functionality was added and it is possible to use more then one device connection in Modbus connector. 
But it works in one thread so every device will be processed one by one.

Example of configuration:

```
{
  "server": {
    "type": "tcp",
    "host": "127.0.0.1",
    "port": 5020,
    "timeout": 35,
    "method": "socket",
    "byteOrder": "BIG",
    "devices": [
      {
        "unitId": 1,
        "deviceName": "Temp Sensor",
        "host": "127.0.0.1",
        "port": 5021,
        "timeout": 35,
        "method": "socket",
        "byteOrder": "BIG",
        "connectAttemptCount": 5,
        "connectAttemptTimeMs": 500,
        "attributesPollPeriod": 5000,
        "timeseriesPollPeriod": 5000,
        "sendDataOnlyOnChange": true,
        "attributes": [
          {
            "tag": "string_read",
            "type": "string",
            "functionCode": 4,
            "objectsCount": 4,
            "address": 1
          }
      ],
      "timeseries": [
          {
            "tag": "8uint_read",
            "type": "8uint",
            "functionCode": 4,
            "objectsCount": 1,
            "address": 17
          }
        ],
        "attributeUpdates": [
          {
            "tag": "shared_attribute_write",
            "type": "32int",
            "functionCode": 6,
            "objectsCount": 2,
            "address": 29
          }
        ],
        "rpc": [
          {
            "tag": "setValue",
            "type": "bits",
            "functionCode": 5,
            "objectsCount": 1,
            "address": 31
          }
        ]
      },
      {
        "unitId": 1,
        "deviceName": "Temp Sensor with default server",
        "attributesPollPeriod": 5000,
        "timeseriesPollPeriod": 5000,
        "sendDataOnlyOnChange": true,
        "attributes": [
          {
            "tag": "string_read",
            "type": "string",
            "functionCode": 4,
            "objectsCount": 4,
            "address": 1
          }
      ],
      "timeseries": [
          {
            "tag": "8uint_read",
            "type": "8uint",
            "functionCode": 4,
            "objectsCount": 1,
            "address": 17
          }
        ],
        "attributeUpdates": [
          {
            "tag": "shared_attribute_write",
            "type": "32int",
            "functionCode": 6,
            "objectsCount": 2,
            "address": 29
          }
        ],
        "rpc": [
          {
            "tag": "setValue",
            "type": "bits",
            "functionCode": 5,
            "objectsCount": 1,
            "address": 31
          }
        ]
      }
    ]
  }
}

```

pehedima, 2020-09-14:
Thanks for the implementation. 
If I want to use two different Modbus TCP devices how they have to inserted in the configuration file? E. g. ohne device with IP 192.168.178.19 and the other with 192.168.178.20

zbeacon, 2020-09-15:
Hi @pehedima ,

Please check the configuration, I have attached earlier. There is device with name Temp Sensor and it has overwritten configuration.

pehedima, 2020-10-01:
I checked the new function using the following JSON file (cutted to two meters, normally there are four):

```
{
  "server": {
    "name": "GW01-01",
      "type":"tcp",
      "host":"192.168.100.10",
      "port":502,
      "timeout":3,
	  "method": "socket",     
	  "byteOrder": "BIG",
	"devices":[
      {
         "unitId":255,
         "deviceName":"GW01-02",
		 "host": "192.168.100.11",
		 "port": 502,
		 "timeout": 3,
		 "method": "socket",
		 "byteOrder": "BIG",
         "connectAttemptCount": 5,
         "connectAttemptTimeMs": 500,
         "attributesPollPeriod":60000,
         "timeseriesPollPeriod":5000,
         "sendDataOnlyOnChange": false,     
         "attributes":[
            {
               "tag":"SerialNumber",
               "type":"string",
               "functionCode":3,
               "address":1280,
               "registerCount":6,
               "byteOrder":"BIG"
            }
         ],
         "timeseries":[ ... ]
      },
      {
         "unitId":255,
         "deviceName":"GW01-01",
         "attributesPollPeriod":60000,
         "timeseriesPollPeriod":5000,
         "sendDataOnlyOnChange": false,     
         "attributes":[
            {
               "tag":"SerialNumber",
               "type":"string",
               "functionCode":3,
               "address":1280,
               "registerCount":6,
               "byteOrder":"BIG"
            }
         ],
         "timeseries":[ ... ]
      }
   ]
 }
}

```

That is working fine :)

I've an additional MODBUS RTU sensor and this is not working neither in the same file as the other meters nor in a seperate file 

contents of the seperate file:

```
{
  "server": {
    "name": "temperature serial",
    "type": "serial",
    "port": "/dev/ttyRS485",
    "baudrate": 9600,
    "timeout": 35,
    "method": "rtu",
    "devices": [
      {
        "unitId": 1,
        "deviceName": "TEMP-Home-python-revpi-balena-serial-ttyRS485-1",
        "attributesPollPeriod": 3600000,
        "timeseriesPollPeriod": 5000,
        "sendDataOnlyOnChange": false,
         "timeseries":[
            {
               "tag":"Temperatur_Buero",
               "type":"long",
               "functionCode":3,
               "address":0,
               "registerCount":1,
               "byteOrder":"BIG",
               "divider":10
            }
         ]
      }
    ]
  }
}
```
For the test in one file I extended the device section of the temp sensor with the other meter devices like in the example above.
The result was, that only the MODBUS TCP Meter GW01-01 sends data to the server. 

If I use a seperate file for the temp sensor and adapt the yaml file like:

```
...
connectors:
  -
    name: Modbus Temperature Serial RS485 Connector
    type: modbus
    configuration: modbus-serial-temperature-config-ttyRS485-1.json
  -
    name: GW01-01-04
    type: modbus
    configuration: GW01-01-04.json
...
```

no data will be send from the temp sensor to the server. Only the values from the two (four) TCP meters.

Has somebody an idea what is going wrong here?

In the old version (2.5.0) using for all meters seperate files all data are send (incl. Temp sensor values) 

By the way I want to mention that it would great if there were an additional parameter 'polltime' for each telemetrie tag.
Is function is also implemented in the java version of this gateway. We need not all necessary data in the same time intervall (e.g. frequency, we need this value once an hour and other values every minutes. 

Let me know if it possible to implement such function and/or if I have to create an additional feature request in case:)

pehedima, 2020-10-26:
Hello togehter,
I'd like to refresh my last questions above and hope for an answer :)

Summary:
1. How it is possible to used more than one JSON File in case of more than one device is used one file?
2. It is possible to insert the parameter 'pulltime' (like in JAVA) for each telemetry value?

For more information see the points I mentioned above.

Thanks for your help.

pehedima, 2020-11-20:
Seems so, that noboby want to solve that problem or want to write something about that issues.

What is the reason?

zbeacon, 2021-03-02:
Hi @pehedima ,

Ability, described in this issue, available in the gateway. Please open the new issue, if you have any questions about the gateway.

closed 2021-03-02 by zbeacon