thingsboard/thingsboard-gateway #325: MODBUS serial permission issue in raspberry pi

hasikp1, 2020-06-25:
Hi I am getting the following error

Jun 25 11:18:42 raspberrypi python3[604]: ""2020-06-25 11:18:42" - WARNING - [modbus_connector.py] - modbus_connector - 72 - Modbus trying reconnect to Modbus Default Server"
Jun 25 11:18:42 raspberrypi python3[604]: ""2020-06-25 11:18:42" - ERROR - [sync.py] - sync - 603 - [Errno 13] could not open port /dev/ttyUSB0: [Errno 13] Permission denied: '/dev/ttyUSB0'"

How can I fix this?

zbeacon, 2020-06-25:
Hi @hasikp1,

Thank you for your interest in the ThingsBoard IoT gateway. 

To solve this issue, you should add thingsboard_gateway user to dialout group.
To do this, please use the following command:

`sudo adduser thingsboard_gateway dialout`

hasikp1, 2020-06-25:
Thank you. working

closed 2020-06-25 by hasikp1