thingsboard/thingsboard-gateway #326: Remote configuration failure

hasikp1, 2020-06-25:
I am trying to push configuration from Thingsboard CE to gateway. I  am getting following.
```
Jun 25 12:04:01 raspberrypi python3[2041]: ""2020-06-25 12:04:01" - INFO - [tb_device_mqtt.py] - tb_device_mqtt - 194 - connection SUCCESS"
Jun 25 12:04:01 raspberrypi python3[2041]: ""2020-06-25 12:04:01" - ERROR - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - 79 - 'utf-8' codec can't decode byte 0x89 in position 0: invalid start byte"
Jun 25 12:04:01 raspberrypi python3[2041]: Traceback (most recent call last):
Jun 25 12:04:01 raspberrypi python3[2041]:   File "/usr/lib/python3/dist-packages/thingsboard_gateway/gateway/tb_gateway_remote_configurator.py", line 59, in process_configuration
Jun 25 12:04:01 raspberrypi python3[2041]:     self.__new_configuration = loads(decoded_configuration)
Jun 25 12:04:01 raspberrypi python3[2041]:   File "/usr/lib/python3/dist-packages/simplejson/__init__.py", line 516, in loads
Jun 25 12:04:01 raspberrypi python3[2041]:     return _default_decoder.decode(s)
Jun 25 12:04:01 raspberrypi python3[2041]:   File "/usr/lib/python3/dist-packages/simplejson/decoder.py", line 373, in decode
Jun 25 12:04:01 raspberrypi python3[2041]:     s = s.decode(self.encoding)
Jun 25 12:04:01 raspberrypi python3[2041]: UnicodeDecodeError: 'utf-8' codec can't decode byte 0x89 in position 0: invalid start byte
```
Extensions are not synced.

zbeacon, 2020-06-25:
Hi @hasikp1 

Please tell more, how do you try to push the configuration?

hasikp1, 2020-06-25:
I am using Thingsboard CE and I have created a device and set it as gateway. From the extensions in the gateway properties I have configured modbus RTU extension with required settings such as polling time, id, register addresses and saved it.

[image]

zbeacon, 2020-06-25:
It is old remote configuration, it will be deprecated in the future versions. The dashboard for remote gateway configuration is available in version 2.5 and above (If you install demo data).

closed 2020-06-27 by zbeacon