thingsboard/thingsboard-gateway #328: help with mosquitto to gateway

jackattack21, 2020-06-25:
hey - currently trying to bridge mosquitto server to thingsboard using thingsboard gateway. everything is running on ubutunu 18.04. guides were used to install tb and tb-gateway. 

right now I am just testing with some of the demo data given by thingsboard website. logs and service status show good connection between the gateway and TB on local host 1882 and good connection with mosquitto on local host 1883. 

however, any data published to mosquitto is not reflected in the TB dashboard and there are no errors listed. TB dashboard continues to show value 0 for all gateway telemetry. 

sending mosquitto_pub -h localhost -t /sensor/data -p 1883 -m '“serialNumber”: “SN-001”, “sensorType”: “Thermometer”, “sensorModel”: “T1000”, “temp”: 42, “hum”: 58}' -u username -P password

mosquitto_sub receives message succesfully, tb_gateway shows nothing

**tb_gateway.yaml**
thingsboard:
  host: localhost
  port: 1882
  remoteConfiguration: false
  security:
    accessToken: myaccesstoken
storage:
  type: memory
  read_records_count: 100
  max_records_count: 100000
#  type: file
#  data_folder_path: ./data/
#  max_file_count: 10
#  max_read_records_count: 10
#  max_records_per_file: 10000
connectors:

  -
    name: MQTT Broker Connector
    type: mqtt
    configuration: mqtt.json

**mqtt.json**
{
  "broker": {
    "name":"Default Local Broker",
    "host":"127.0.0.1",
    "port":1883,
...
with rest of the default TB info

**Connector.log**
""2020-06-25 19:52:33" - INFO - [mqtt_connector.py] - mqtt_connector - 226 - MQTT Broker Connector connected to 127.0.0.1:1883 - successfully."
""2020-06-25 19:52:33" - INFO - [mqtt_connector.py] - mqtt_connector - 265 - Connector "MQTT Broker Connector" subscribe to /sensor/data"
""2020-06-25 19:52:33" - INFO - [mqtt_connector.py] - mqtt_connector - 265 - Connector "MQTT Broker Connector" subscribe to /sensor/+/data"
""2020-06-25 19:52:33" - INFO - [mqtt_connector.py] - mqtt_connector - 265 - Connector "MQTT Broker Connector" subscribe to /custom/sensors/+"
""2020-06-25 19:52:33" - INFO - [mqtt_connector.py] - mqtt_connector - 312 - "MQTT Broker Connector" subscription success to topic /sensor/data, subscriptio$
""2020-06-25 19:52:33" - INFO - [mqtt_connector.py] - mqtt_connector - 312 - "MQTT Broker Connector" subscription success to topic /sensor/+/data, subscript$
""2020-06-25 19:52:33" - INFO - [mqtt_connector.py] - mqtt_connector - 312 - "MQTT Broker Connector" subscription success to topic /custom/sensors/+, subscr$
""2020-06-25 19:52:33" - INFO - [mqtt_connector.py] - mqtt_connector - 312 - "MQTT Broker Connector" subscription success to topic sensor/connect, subscript$
""2020-06-25 19:52:33" - INFO - [mqtt_connector.py] - mqtt_connector - 312 - "MQTT Broker Connector" subscription success to topic sensor/+/connect, subscri$
""2020-06-25 19:52:33" - INFO - [mqtt_connector.py] - mqtt_connector - 312 - "MQTT Broker Connector" subscription success to topic sensor/disconnect, subscr$
""2020-06-25 19:52:33" - INFO - [mqtt_connector.py] - mqtt_connector - 312 - "MQTT Broker Connector" subscription success to topic sensor/+/disconnect, subs$

**tb-connection.log**
""2020-06-25 19:47:03" - INFO - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 175 - Subscribed to *|* with id 1"
""2020-06-25 19:47:03" - INFO - [tb_device_mqtt.py] - tb_device_mqtt - 194 - connection SUCCESS"
""2020-06-25 19:52:33" - INFO - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 175 - Subscribed to *|* with id 1"
""2020-06-25 19:52:33" - INFO - [tb_device_mqtt.py] - tb_device_mqtt - 194 - connection SUCCESS"

**converter.log**
empty

pleaseee help

jackattack21, 2020-06-26:
error im getting now in converter.log
2020-06-26 13:36:51" - ERROR - [json_mqtt_uplink_converter.py] - json_mqtt_uplink_converter - 57 - Expecting property name enclosed in double quotes: line$
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/mqtt/json_mqtt_uplink_converter.py", line 34, in convert
    dict_result["deviceName"] = TBUtility.get_value(self.__config.get("deviceNameJsonExpression"), data, expression_instead_none=True)
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/tb_utility/tb_utility.py", line 115, in get_value
    body = loads(body)
  File "/usr/lib/python3/dist-packages/simplejson/__init__.py", line 518, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3/dist-packages/simplejson/decoder.py", line 370, in decode
    obj, end = self.raw_decode(s)
  File "/usr/lib/python3/dist-packages/simplejson/decoder.py", line 400, in raw_decode
    return self.scan_once(s, idx=_w(s, idx).end())
simplejson.errors.JSONDecodeError: Expecting property name enclosed in double quotes: line 1 column 2 (char 1)
""2020-06-26 13:36:51" - ERROR - [json_mqtt_uplink_converter.py] - json_mqtt_uplink_converter - 78 - Expecting property name enclosed in double quotes: line$
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/mqtt/json_mqtt_uplink_converter.py", line 62, in convert
    value = TBUtility.get_value(datatype_config["value"], data, datatype_config["type"], expression_instead_none=True)
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/tb_utility/tb_utility.py", line 115, in get_value
    body = loads(body)
  File "/usr/lib/python3/dist-packages/simplejson/__init__.py", line 518, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3/dist-packages/simplejson/decoder.py", line 370, in decode
    obj, end = self.raw_decode(s)
  File "/usr/lib/python3/dist-packages/simplejson/decoder.py", line 400, in raw_decode
    return self.scan_once(s, idx=_w(s, idx).end())
simplejson.errors.JSONDecodeError: Expecting property name enclosed in double quotes: line 1 column 2 (char 1)

jackattack21, 2020-06-26:
tried sending -m {"deviceName": "Device A", "deviceType": "thermostat", "attributes": {"model": "Model A", "serialNumber": "SN-111"}, telemetry": {"temperature": 42, "humidity": 80}}

still similar errors in converter.log

jackattack21, 2020-06-26:
tried sending -m {"deviceName": "Device A", "deviceType": "thermostat", "attributes": {"model": "Model A", "serialNumber": "SN-111"}, "telemetry": {"temperature": 42, "humidity": 80}}

still similar errors in converter.log

jackattack21, 2020-06-26:
@zbeacon

jackattack21, 2020-06-26:
Walked back through and installed everything from scratch. 
-installed thingsboard and changed default mqtt to 1882
- installed mosquitto
- tested thingsboard with mosquitto_pub to 1882 for a new device with telemetry - everything works good
- installed gateway
- configured gateway to TB by updating ip/port and access token 
- edited mqtt.json with username and password for localhost 1883 for mosquitto
- checked systemctl status
- checked var/log for good connections to broker and tb -everything good, connected and subscribed
- sent test with mosquitto_pub -d -h localhost -t /sensor/data -u username -P password -m '{"serialNumber":"SN-001","sensorType":"Thermometer","sensorModel":"T1000","temp":42,"hum":58}'

gateway stats still 0. no new devices.

jackattack21, 2020-06-26:
same error when using the demo payload with the demo mqtt.json converter

""2020-06-26 16:50:05" - ERROR - [json_mqtt_uplink_converter.py] - json_mqtt_uplink_converter - 57 - Expecting property name enclosed in double quotes: line 1 column 2
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/mqtt/json_mqtt_uplink_converter.py", line 34, in convert
    dict_result["deviceName"] = TBUtility.get_value(self.__config.get("deviceNameJsonExpression"), data, expression_instead_none=True)
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/tb_utility/tb_utility.py", line 115, in get_value
    body = loads(body)
  File "/usr/lib/python3/dist-packages/simplejson/__init__.py", line 518, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3/dist-packages/simplejson/decoder.py", line 370, in decode
    obj, end = self.raw_decode(s)
  File "/usr/lib/python3/dist-packages/simplejson/decoder.py", line 400, in raw_decode
    return self.scan_once(s, idx=_w(s, idx).end())
simplejson.errors.JSONDecodeError: Expecting property name enclosed in double quotes: line 1 column 2 (char 1)
""2020-06-26 16:50:05" - ERROR - [json_mqtt_uplink_converter.py] - json_mqtt_uplink_converter - 78 - Expecting property name enclosed in double quotes: line$
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/mqtt/json_mqtt_uplink_converter.py", line 62, in convert
    value = TBUtility.get_value(datatype_config["value"], data, datatype_config["type"], expression_instead_none=True)
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/tb_utility/tb_utility.py", line 115, in get_value
    body = loads(body)
  File "/usr/lib/python3/dist-packages/simplejson/__init__.py", line 518, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3/dist-packages/simplejson/decoder.py", line 370, in decode
    obj, end = self.raw_decode(s)
  File "/usr/lib/python3/dist-packages/simplejson/decoder.py", line 400, in raw_decode
    return self.scan_once(s, idx=_w(s, idx).end())
simplejson.errors.JSONDecodeError: Expecting property name enclosed in double quotes: line 1 column 2 (char 1)

zbeacon, 2020-06-26:
@jackattack21  

This error means you have a wrong json in the configuration file.

jackattack21, 2020-06-26:
@zbeacon the default mqtt.json files comes with an array for the first attributes section. I left 
{            "type": "string",
            "key": "model",
            "value": "${sensorModel}"
          }
and deleted the one below it - think it looked like this: 
{
            "type": "string",
            "key": "model",
            "value": "On"
          }

it now works. additionally removing the space after the colon in the json payload from mosquitto was required as well

closed 2020-06-26 by jackattack21