thingsboard/thingsboard-gateway #329: will this gateway ever be user friendly?

vanilla-thunder, 2020-06-27:
And by "user" I mean some regular IT/tech affine person without PhD in informatics and 10 years experience in python programming.

My current goal is getting data from a BLE soil sensor into thingsboard.
I got some 3rd party scripts working without any problems, but I wanted to do it "the right way" and make use of the gateway.
After installing the gateway I started reading the connector documentation and at some point it casually says "this is our demo config, you will have to make your own config with all different UUIDs and parameters" as if it were the easiest thing ever and not worth any guidance.
(which might be true, if you are a Python Jedi with a bluetooth sword, but unfortunately I'm not)

Is the gateway (or at least the BLE connector) supposed to be for very advanced users only?
Cheers

zbeacon, 2020-06-27:
Hi @vanilla-thunder,

Thank you for your interest in the ThingsBoard IoT gateway. 

I agree with you, the BLE connector is a little difficult to configure, because you need to know the UUID of the services and the method for these services (read or notify).  Please write your suggestions - how to make it more convenient?

zbeacon, 2020-06-27:
@vanilla-thunder  

The gateway can read default UUIDs, but usually, every BLE device contains custom services with different identifiers which depend on developer company choice. I see the next solution for this issue:

We can develop feature, it will read all custom services and tried to read them values. The gateway will push it to the ThingsBoard with UUIDs as keys and values from services as value of the attributes. 
This will help to find UUIDs, but unfortunately, we won't know which services you need.
What do you think about this solution?

closed 2020-12-06 by zbeacon