thingsboard/thingsboard-gateway #330: OPC UA Connector connected but no data is received on Dashboard[HELP]

sayyam1994, 2020-06-27:
**Describe the issue**
I am trying to connect Thingsboard to Kepserverex v6.7 but I am not getting any data on Dashboard even after successfully connecting to the opc connector.
I think the issue is with incorrect mapping in opcua.json file.

opcua.json file:-

{
  "server": {
    "name": "Thingsboard OPC-UA client",
    "url": "127.0.0.1:49320",
    "timeoutInMillis": 5000,
    "scanPeriodInMillis": 5000,
    "disableSubscriptions":false,
    "subCheckPeriodInMillis": 100,
    "showMap": false,
    "security": "None",
    "identity": {
      "type": "anonymous"
    },
    "mapping": [
      {
        "deviceNodePattern": "Channel1\\.Device\\d+$",
        "deviceNamePattern": "Device ${_System._DeviceId}",
        "attributes": [
          {
            "key": "Tag1",
            "path": "${ns=2;s=Channel1.Device1.Tag1}"
          }
        ],
        "timeseries": [
          {
            "key": "Tag2",
            "path": "${ns=2;s=Channel1.Device1.Tag2}"
          }
        ]
      }
    ]
  }
}

I am using thingsboard-gateway 2.4.
Kindly help me in getting the right configuration

Versions:-
 - OS: [e.g. Windows 7]
 - Thingsboard IoT Gateway version [2.4]
 - Python version [3.7]

zbeacon, 2020-06-29:
Hi @sayyam1994 

Thank you for your interest in the ThingsBoard IoT gateway. Please set path in the parameters from values like "ns=2;s=Channel1.Device1.Tag1" to "Channel1\\.Device1\\.Tag1". Node path delimiters are "\\." instead of "\." or ".".

The configuration example is:

```

{
  "server": {
    "name": "Thingsboard OPC-UA client",
    "url": "127.0.0.1:49320",
    "timeoutInMillis": 5000,
    "scanPeriodInMillis": 5000,
    "disableSubscriptions": false,
    "subCheckPeriodInMillis": 100,
    "showMap": false,
    "security": "None",
    "identity": {
      "type": "anonymous"
    },
    "mapping": [
      {
        "deviceNodePattern": "Channel1\\.Device\\d+$",
        "deviceNamePattern": "Device ${Root\\._System\\._DeviceId}",
        "attributes": [
          {
            "key": "Tag1",
            "path": "${Channel1\\.Device1\\.Tag1}"
          }
        ],
        "timeseries": [
          {
            "key": "Tag2",
            "path": "${Channel1\\.Device1\\.Tag2}"
          }
        ]
      }
    ]
  }
}

```

sayyam1994, 2020-06-29:
Thanks for your quick response.

I tried with the above configuration but still I am not getting any data on dashboard.

ocpua.json

{
  "server": {
    "name": "Thingsboard OPC-UA client",
    "url": "127.0.0.1:49320",
    "timeoutInMillis": 5000,
    "scanPeriodInMillis": 5000,
    "disableSubscriptions": false,
    "subCheckPeriodInMillis": 100,
    "showMap": false,
    "security": "None",
    "identity": {
      "type": "anonymous"
    },
    "mapping": [
      {
        "deviceNodePattern": "Channel1\\.Device\\d+$",
        "deviceNamePattern": "Device ${Root\\._System\\._DeviceId}",
        "attributes": [
          {
            "key": "Tag1",
            "path": "${Channel1\\.Device1\\.Tag1}"
          }
        ],
        "timeseries": [
          {
            "key": "Tag2",
            "path": "${Channel1\\.Device1\\.Tag2}"
          }
        ]
      }
    ]
  }
}

tb_gateway.yaml
thingsboard:
  host: 127.0.0.1
  port: 1883
  remoteConfiguration: false
  security:
    accessToken: pLd2WQ1TcnmEsPsX4fUY
storage:
  type: memory
  read_records_count: 100
  max_records_count: 100000
#  type: file
#  data_folder_path: ./data/
#  max_file_count: 10
#  max_read_records_count: 10
#  max_records_per_file: 10000
connectors:
#  -
#    name: MQTT Broker Connector
#    type: mqtt
#    configuration: mqtt.json
#
#  -
#    name: Modbus Connector
#    type: modbus
#    configuration: modbus.json
#
#  -
#    name: Modbus Connector
#    type: modbus
#    configuration: modbus_serial.json
#
  -
    name: OPC-UA Connector
    type: opcua
    configuration: opcua.json

#  -
#    name: BLE Connector
#    type: ble
#    configuration: ble.json
#
#  -
#    name: REQUEST Connector
#    type: request
#    configuration: request.json
#
#  -
#    name: CAN Connector
#    type: can
#    configuration: can.json
#
#  -
#    name: BACnet Connector
#    type: bacnet
#    configuration: bacnet.json
#
#  -
#    name: ODBC Connector
#    type: odbc
#    configuration: odbc.json
#
#  -
#    name: REST Connector
#    type: rest
#    configuration: rest.json
#
#  -
#    name: Custom Serial Connector
#    type: serial
#    configuration: custom_serial.json
#    class: CustomSerialConnector

My device tree looks like this:-
Channel1
               Device1
                            Tag1
                            Tag2

Screenshot of my Kepserver Configuration:-
[image]

sayyam1994, 2020-06-29:
Thanks for your quick response.

I tried with the above configuration but still I am not getting any data on dashboard.

ocpua.json

{
  "server": {
    "name": "Thingsboard OPC-UA client",
    "url": "127.0.0.1:49320",
    "timeoutInMillis": 5000,
    "scanPeriodInMillis": 5000,
    "disableSubscriptions": false,
    "subCheckPeriodInMillis": 100,
    "showMap": false,
    "security": "None",
    "identity": {
      "type": "anonymous"
    },
    "mapping": [
      {
        "deviceNodePattern": "Channel1\\.Device\\d+$",
        "deviceNamePattern": "Device ${Root\\._System\\._DeviceId}",
        "attributes": [
          {
            "key": "Tag1",
            "path": "${Channel1\\.Device1\\.Tag1}"
          }
        ],
        "timeseries": [
          {
            "key": "Tag2",
            "path": "${Channel1\\.Device1\\.Tag2}"
          }
        ]
      }
    ]
  }
}

Screenshot of my Kepserver Configuration:-
[image]

zbeacon, 2020-06-29:
Okay, please attach information from connector.log file in the logs folder.

sayyam1994, 2020-06-29:
""2020-06-28 16:40:37" - INFO - [opcua_connector.py] - opcua_connector - 130 - Reconnected to the OPC-UA server - 127.0.0.1:49320"
""2020-06-29 12:49:44" - INFO - [opcua_connector.py] - opcua_connector - 96 - Starting OPC-UA Connector"
""2020-06-29 12:49:44" - INFO - [opcua_connector.py] - opcua_connector - 121 - OPC-UA connector OPC-UA Connector connected to server 127.0.0.1:49320"
""2020-06-29 13:09:59" - INFO - [opcua_connector.py] - opcua_connector - 96 - Starting OPC-UA Connector"
""2020-06-29 13:10:00" - INFO - [opcua_connector.py] - opcua_connector - 121 - OPC-UA connector OPC-UA Connector connected to server 127.0.0.1:49320"
""2020-06-29 13:15:41" - INFO - [opcua_connector.py] - opcua_connector - 96 - Starting OPC-UA Connector"
""2020-06-29 13:15:41" - INFO - [opcua_connector.py] - opcua_connector - 121 - OPC-UA connector OPC-UA Connector connected to server 127.0.0.1:49320"
""2020-06-29 13:17:30" - INFO - [opcua_connector.py] - opcua_connector - 96 - Starting OPC-UA Connector"
""2020-06-29 13:17:30" - INFO - [opcua_connector.py] - opcua_connector - 121 - OPC-UA connector OPC-UA Connector connected to server 127.0.0.1:49320"
""2020-06-29 13:20:58" - INFO - [opcua_connector.py] - opcua_connector - 96 - Starting OPC-UA Connector"
""2020-06-29 13:20:58" - INFO - [opcua_connector.py] - opcua_connector - 121 - OPC-UA connector OPC-UA Connector connected to server 127.0.0.1:49320"
""2020-06-29 13:26:19" - INFO - [opcua_connector.py] - opcua_connector - 96 - Starting OPC-UA Connector"
""2020-06-29 13:26:19" - INFO - [opcua_connector.py] - opcua_connector - 121 - OPC-UA connector OPC-UA Connector connected to server 127.0.0.1:49320"
""2020-06-29 13:29:39" - INFO - [opcua_connector.py] - opcua_connector - 96 - Starting OPC-UA Connector"
""2020-06-29 13:29:39" - INFO - [opcua_connector.py] - opcua_connector - 121 - OPC-UA connector OPC-UA Connector connected to server 127.0.0.1:49320"
""2020-06-29 13:35:23" - INFO - [opcua_connector.py] - opcua_connector - 96 - Starting OPC-UA Connector"
""2020-06-29 13:35:24" - INFO - [opcua_connector.py] - opcua_connector - 121 - OPC-UA connector OPC-UA Connector connected to server 127.0.0.1:49320"
""2020-06-29 13:37:38" - INFO - [opcua_connector.py] - opcua_connector - 96 - Starting OPC-UA Connector"
""2020-06-29 13:37:38" - INFO - [opcua_connector.py] - opcua_connector - 121 - OPC-UA connector OPC-UA Connector connected to server 127.0.0.1:49320"
""2020-06-29 13:37:54" - INFO - [opcua_connector.py] - opcua_connector - 96 - Starting OPC-UA Connector"
""2020-06-29 13:37:54" - INFO - [opcua_connector.py] - opcua_connector - 121 - OPC-UA connector OPC-UA Connector connected to server 127.0.0.1:49320"
""2020-06-29 13:54:19" - INFO - [opcua_connector.py] - opcua_connector - 96 - Starting OPC-UA Connector"
""2020-06-29 13:54:19" - INFO - [opcua_connector.py] - opcua_connector - 121 - OPC-UA connector OPC-UA Connector connected to server 127.0.0.1:49320"

zbeacon, 2020-06-29:
It cannot connect correctly to the OPC-UA server. Please chenge the level of the logging in the file config\logs.conf from INFO to DEBUG, run the gateway again and send file. Also you can archive the logs and email me ibarkov@thingsboard.io

sayyam1994, 2020-06-29:
I have sent the mail

zbeacon, 2020-06-30:
I saw them, but I didn't see the map of the server. Did you set up the parameter "showMap" to true in config?

sayyam1994, 2020-06-30:
I have sent the updated logs after enabling "showMap = True"

zbeacon, 2020-06-30:
Ok, I saw, please try the following configuration:

```

{
  "server": {
    "name": "Thingsboard OPC-UA client",
    "url": "127.0.0.1:49320",
    "timeoutInMillis": 5000,
    "scanPeriodInMillis": 5000,
    "disableSubscriptions": false,
    "subCheckPeriodInMillis": 100,
    "showMap": false,
    "security": "None",
    "identity": {
      "type": "anonymous"
    },
    "mapping": [
      {
        "deviceNodePattern": "Channel1\\.Device\\d+$",
        "deviceNamePattern": "Device ${Root\\._System\\._DeviceId}",
        "attributes": [
          {
            "key": "Tag1",
            "path": "${Tag1}"
          }
        ],
        "timeseries": [
          {
            "key": "Tag2",
            "path": "${Tag2}"
          }
        ]
      }
    ]
  }
}

```

sayyam1994, 2020-07-05:
I also tried connecting the opc ua server in ubuntu but still same issue. I am not getting any data even after successfully connecting with thingsboard gateway.

closed 2020-09-09 by zbeacon