thingsboard/thingsboard-gateway #331: [HELP] modbus config

jdevoo, 2020-06-27:
**Describe the issue**
I have a modbus server started on the LAN, port 5020 and I use the configs below for the GW
```
thingsboard:
  host: localhost
  port: 1883
  security:
    accessToken: some_token

storage:
  type: memory
  read_records_count: 100
  max_records_count: 100000

connectors:
  -
    name: Modbus Connector
    type: modbus
    configuration: modbus.json
```

modbus is configured with the following file

```
{
  "server": {
    "host": "192.168.1.109",
    "port": 5020,
    "devices": [
      {
        "unitId": 1,
        "deviceName": "fio_server",
        "sendDataOnlyOnChange": false,
        "attributes": [
          {
            "tag": "at_entry", "type": "bits", "functionCode": 1, "objectsCount": 1, "address": 1
          },
          {
            "tag": "at_exit", "type": "bits", "functionCode": 1, "objectsCount": 1, "address": 2
          }
        ],
        "rpc": [
          {
            "tag": "entry_conveyor", "type": "bits", "functionCode": 5, "objectsCount": 1, "address": 0
          },
          {
            "tag": "buffer_conveyor", "type": "bits", "functionCode": 5, "objectsCount": 1, "address": 1
          }
        ]
      }
    ]
  }
}
```

When i launch the gateway though, the modbus config is causing troubles with a NoneType - see below

```
""2020-06-27 14:27:46" - INFO - [tb_gateway_service.py] - tb_gateway_service - 76 - Gateway starting..."
""2020-06-27 14:27:46" - INFO - [tb_gateway_service.py] - tb_gateway_service - 81 - ThingsBoard IoT gateway version: 2.4.0"
""2020-06-27 14:27:46" - INFO - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 175 - Subscribed to *|* with id 1"
Modbus library not found - installing...
""2020-06-27 14:27:47" - INFO - [tb_device_mqtt.py] - tb_device_mqtt - 194 - connection SUCCESS"
Collecting pymodbus>=2.3.0
  Downloading pymodbus-2.3.0-py2.py3-none-any.whl (138 kB)
Requirement already satisfied: six>=1.11.0 in /usr/lib/python3/dist-packages (from pymodbus>=2.3.0) (1.14.0)
Requirement already satisfied: pyserial>=3.4 in /usr/lib/python3/dist-packages (from pymodbus>=2.3.0) (3.4)
Installing collected packages: pymodbus
  WARNING: The script pymodbus.console is installed in '/root/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
Successfully installed pymodbus-2.3.0
""2020-06-27 14:27:48" - ERROR - [tb_gateway_service.py] - tb_gateway_service - 297 - 'NoneType' object is not callable"
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/gateway/tb_gateway_service.py", line 289, in _connect_with_connectors
    connector = self._implemented_connectors[connector_type](self, connector_config["config"][config],
TypeError: 'NoneType' object is not callable
""2020-06-27 14:27:48" - INFO - [tb_gateway_service.py] - tb_gateway_service - 138 - Gateway started."
```

**Versions (please complete the following information):**
 - OS: Docker image based on Ubuntu 20.04
 - Thingsboard IoT Gateway version 2.4.0 (checkout tag/2.4.0)
 - Python version 3.8.2

zbeacon, 2020-06-29:
Hi @jdevoo,

It is an issue with docker image, could you try to download the latest image from dockerhub and try again?

closed 2020-08-03 by zbeacon

faywong, 2021-07-14:
> I have a modbus server started on the LAN, port 5020 and I use the configs below for the GW

@zbeacon @jdevoo I apologize for not related with the topic in advance. My question is what's the most used modbus server, is it a [hardware component](https://comtrol.com/resources/product-resources-white-papers/devicemaster-resources/devicemaster-up-modbus-solutions/modbus-server) or a software emulation library like [EasyModbusTCP.NET](https://github.com/rossmann-engineering/EasyModbusTCP.NET) ?