thingsboard/thingsboard-gateway #332: [HELP] How set QoS on thingsboard gateway?

pehedima, 2020-06-29:
Hi together,
it is possible to set QoS at gateway (MQTT) and if yes how to do it?
What is the default?

zbeacon, 2020-06-30:
Hi @pehedima 

There is no ability to change the QoS using config. To change it, you can change the source code of the library.
Default is 1. Also you can set 0 in the gateway service.

viktorkho, 2020-06-30:
Еhis is a suitable candidate for placement in the [Configuration properties](https://thingsboard.io/docs/user-guide/install/config/), IMHO.

DanielRichart, 2020-08-13:
Hello,

Where can I change the source code of the library and how can I set qos to 0 in the service?

Thanks,
Daniel

zbeacon, 2020-09-01:
Hi to change the quality of server in the gateway please download the version from the master branch and check the tb_gateway.yaml file. There is in the section "thingsboard" is appeared new parameter with name "qos". You can put there QoS, you prefer. The gateway supports 0 or 1 qos.

closed 2020-09-01 by zbeacon