thingsboard/thingsboard-gateway #334: [HELP] BLE Connector random address type

zwells, 2020-06-30:
**Describe the issue**
Configuring the ble connector, how do you set the addrType as random in the ble.json?  Is there a .py file that needs to be modified?

I can't connect to my devices through bluepy unless specified as random address type.

**Connector name (If you need help with some connector/converter):**
- BLE Connector

**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 18.04]
 - Thingsboard IoT Gateway version [e.g. 2.2.4]
 - Python version[e.g. 3.7]

zwells, 2020-06-30:
I think I'll just need to manually insert it as a parameter like peripheral = Peripheral(MAC, "random") in the __get_services_and_chars(self): function in ble_connector.py

Will test and close if it resolves the issue.

zbeacon, 2020-07-01:
Hi @zwells,

Thank you for your investigations. You are right, there was no addrType parameter. But I added it in commit bcb9b87 .
You can put parameter "addrType" into device configuration, near the MACAddress and the gateway will use it instead of public address type.

closed 2020-07-03 by zbeacon