thingsboard/thingsboard-gateway #335: [BUG] ThingsBoard IoT Gateway stalled on uppercase letters in some topics parts

viktorkho, 2020-07-01:
**Describe the bug**
ThingsBoard IoT Gateway with custom converter (the same as in #299) stalled if I publish messages containing spaces.

In `mqtt.json`:
```
"topicFilter": "+/devices/+/controls/+"
```

How to reproduce:
```
mosquitto_pub -t ABC123DE/devices/test/controls/any -m 1      # INFO Successfully converted
mosquitto_pub -t ABC123DE/devices/TEST/controls/any -m 1      # Nothing in log
mosquitto_pub -t ABC123DE/devices/test/controls/any -m 1      # Nothing in log
```
Expected behaviour:
Rather puzzling question. MQTT topicas are (at least in v3.1.1 of spec). But what is clear:
- IoT gateway SHOULD NOT crash in case related cases;
- if uppercese part of topic hits to wildcard IoT Gateway SHOULD pass it to converter.

**Connector name (If bug in the some connector):**
MQTT Connector

**Error traceback (If available):**
Nothing in docker container's output after message published to the topic (to which it is subscribed)

**Versions (please complete the following information):**
- docker-compose version 1.17.1
- image: thingsboard/tb-gateway:latest

zbeacon, 2020-07-01:
Hi @viktorkho 

Do you use the default converter?
The gateway receives and process the data. It looks like an issue with converter.

viktorkho, 2020-07-01:
Hi, @zbeacon I use customized converter

Yes, seems I have some error in my converter, which I cannot localize because it becomes stalled on any error inside `try` block.

Bu the way, with your default converter from 6b47a6b my Gateway stalled immediately after start:
```
...
DEBUG - tb_gateway_mqtt - 65 - Service subscription to topic v1/gateway/rpc - successfully completed.'
DEBUG - tb_gateway_service - 179 - Received data: {}'
DEBUG - tb_gateway_service - 180 - (None,)'
```

I guess it is related to exteptions handling in converter or at a higher level, you can test this:
```
class TestMqttConverter(MqttUplinkConverter):
    def __init__(self, config):
        self.__config = config.get('converter')
        self.dict_result = {}

    def convert(self, topic, body):
        try:
            self.dict_result["telemetry"] = [{"test": "123"}]
            log.debug(self.dict_result)
            raise ZeroDivisionError("DEADBEEF")
            return self.dict_result

        except Exception as e:
            log.exception('Error in converter, for config: \n%s\n and message: \n%s\n', dumps(self.__config), body)
            log.exception(e)
```

Output:
```
...
DEBUG - tb_gateway_mqtt - 65 - Service subscription to topic v1/gateway/attributes/response - successfully completed.'
DEBUG - tb_gateway_mqtt - 65 - Service subscription to topic v1/gateway/rpc - successfully completed.'
DEBUG - tb_gateway_service - 179 - Received data: {}'
DEBUG - tb_gateway_service - 180 - (None,)'
DEBUG - wirenboard_mqtt_uplink_converter - 60 - {'telemetry': [{'test': '123'}]}'
^^^ this is the last message from the Gateway until restart
```

viktorkho, 2020-07-01:
I think I should open new issue about this.

closed 2020-07-01 by viktorkho

viktorkho, 2020-07-01:
> I think I should open new issue about this.

Here it is: #336