thingsboard/thingsboard-gateway #336: [BUG] ThingsBoard IoT Gateway stalled on any error inside custom converter

viktorkho, 2020-07-01:
**Describe the bug**

You can try to reproduce it with this code:
```
class TestMqttConverter(MqttUplinkConverter):
    def __init__(self, config):
        self.__config = config.get('converter')
        self.dict_result = {}

    def convert(self, topic, body):
        try:
            self.dict_result["telemetry"] = [{"test": "123"}]
            log.debug(self.dict_result)
            raise ZeroDivisionError("DEADBEEF")
            return self.dict_result

        except Exception as e:
            log.exception('Error in converter, for config: \n%s\n and message: \n%s\n', dumps(self.__config), body)
            log.exception(e)
```

Publish any message into any suitable topic.

Output:
```
...
DEBUG - tb_gateway_mqtt - 65 - Service subscription to topic v1/gateway/attributes/response - successfully completed.'
DEBUG - tb_gateway_mqtt - 65 - Service subscription to topic v1/gateway/rpc - successfully completed.'
DEBUG - tb_gateway_service - 179 - Received data: {}'
DEBUG - tb_gateway_service - 180 - (None,)'
DEBUG - wirenboard_mqtt_uplink_converter - 60 - {'telemetry': [{'test': '123'}]}'
^^^ this is the last message from the Gateway until restart
```

You can comment `raise ZeroDivisionError...` string to make it work. **Any exception in this block cannot be debugged**.

**Connector name (If bug in the some connector):**
MQTT Connector - Custom Converter

**Versions (please complete the following information):**
- docker-compose version 1.17.1
- image: thingsboard/tb-gateway:latest

viktorkho, 2020-07-01:
I think most ofl my issues could have been called differently if not for this bug..

viktorkho, 2020-07-01:
Upd:
```
        except Exception as e:
            log.exception('Error in converter')
            log.exception(e)
```
outputs:
```
tgate        | ''2020-07-01 12:33:45' - ERROR - test_mqtt_uplink_converter - 65 - Error in converter'
tgate        | Traceback (most recent call last):
tgate        |   File "/var/lib/thingsboard_gateway/extensions/mqtt/test_mqtt_uplink_converter.py", line 61, in convert
tgate        |     raise ZeroDivisionError("DEADBEEF")
tgate        | ZeroDivisionError: DEADBEEF
tgate        | ''2020-07-01 12:33:45' - ERROR - test_mqtt_uplink_converter - 66 - DEADBEEF'
tgate        | Traceback (most recent call last):
tgate        |   File "/var/lib/thingsboard_gateway/extensions/mqtt/test_mqtt_uplink_converter.py", line 61, in convert
tgate        |     raise ZeroDivisionError("DEADBEEF")
tgate        | ZeroDivisionError: DEADBEEF
tgate        | ''2020-07-01 12:33:45' - ERROR - mqtt_connector - 351 - Cannot find converter for the topic:"123ABC34/devices/test/controls/any"! Client: <paho.mqtt.client.Client object at 0x7fa6e56e8ef0>, User data: None'
```

zbeacon, 2020-07-27:
Hi @viktorkho ,

I added the error handling for the converters in the Mqtt connector.It is fixed in 2.5.0

closed 2020-07-27 by zbeacon