thingsboard/thingsboard-gateway #338: [HELP]How to map data of this format in mqtt.json

sayyam1994, 2020-07-02:
Hello Again,

I want to how can the data of following format be mapped in MQTT.json

Data:- {"timestamp":1593698672213,"values":[{"id":"Channel1.Device1.Tag1","v":1121,"q":true,"t":1593698671342},{"id":"Channel1.Device1.Tag2","v":0,"q":true,"t":1593698671342}]}

**Versions (please complete the following information):**
 - OS: Windows 7
 - Thingsboard IoT Gateway version 2.4
 - Python version 3.7

zbeacon, 2020-07-03:
Hi @sayyam1994 ,

To parse payloads like this in the MQTT connector you can use JSONpath Expressions, e.g.
To get first id, you can use the following configuration:

```
{
  "type":"string",
  "key":"id",
  "value":"${$.values[0].id}"
}
```

Unfortunately, there is no ability to parse an array of the objects in the payload. It is not implemented yet in the default JSONUplink converter. But you can create the custom converter, if you will need this option.

sayyam1994, 2020-07-03:
Thanks for the help.

closed 2020-07-03 by sayyam1994