thingsboard/thingsboard-gateway #344: [BUG] Latest (2.4.1) docker image is not working

Tikarei, 2020-07-13:
**Describe the bug**
The latest (2.4.1) docker image does not have the default directories listed on this guide: https://thingsboard.io/docs/iot-gateway/configuration/

What am I missing or doing wrong?

**Error traceback (If available):**
```
root@f5df766f692b:/# cd /etc/thingsboard-gateway/config
bash: cd: /etc/thingsboard-gateway/config: No such file or directory
root@f5df766f692b:/# cd /var/lib/thingsboard_gateway/extensions 
bash: cd: /var/lib/thingsboard_gateway/extensions: No such file or directory
root@f5df766f692b:/# cd /var/log/thingsboard-gateway
bash: cd: /var/log/thingsboard-gateway: No such file or directory
```

**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 18.04]
 - Thingsboard IoT Gateway version: Latest docker image 2.4.1

zbeacon, 2020-07-13:
Hi @Tikarei ,

Thank you for your interest in the ThingsBoard IoT gateway.
You are right, we are changed the base image for the gateway in the docker, and paths also were changed. We will add information about new paths into the documentation.

New paths in the docker container are:

config - `/thingsboard_gateway/config`
extensions - `/thingsboard_gateway/extensions`
logs - `/logs`

Tikarei, 2020-07-13:
Thank you very much for the quick response. I'll close this case when the documentation is updated so others can see this issue in the meantime.

zbeacon, 2020-08-03:
Hi @Tikarei ,

The documentation are updated, I close this issue.

If you will have any questions about the gateway please let me know.

closed 2020-08-03 by zbeacon

Tikarei, 2020-08-05:
I'm still seeing the old config directories listed `/etc/thingsboard-gateway/config` on this page - https://thingsboard.io/docs/iot-gateway/configuration/

zbeacon, 2020-08-05:
@Tikarei 

The paths for the installation from the package were not changed. But in installation guides for docker paths were changed:
[Install ThingsBoard IoT Gateway using Docker for Linux](https://thingsboard.io/docs/iot-gateway/install/docker-installation/)
[Install ThingsBoard IoT Gateway using Docker for Windows](https://thingsboard.io/docs/iot-gateway/install/docker-windows/)

Tikarei, 2020-08-05:
Thank you for clarifying!