thingsboard/thingsboard-gateway #346: [HELP]

e60newbie, 2020-07-17:
**Describe the issue**
I want install TB-Gateway on my Debian 9 Module; Unfortunately my disk space is really restricted (total 1Gb, where 600Mb free).
Is there a possibility to install the gateway without neccessary Modules? I need to use only MQTT and the Can.
I tried to install with all sources (PIP, Docker, from sources) but in all cases I get compiling errors or not enough disk space because i need to install several Modules to use TB-GW.
For any help I was happy.

**Connector name (If you need help with some connector/converter):**
MQTT-Connector
Can-Connector

**Versions (please complete the following information):**
 - OS: Debian 9
 - Thingsboard IoT Gateway version 2.4.1

zbeacon, 2020-07-20:
Hi @e60newbie 

Thank you for your interest in the ThingsBoard IoT gateway. 
In the latest release (2.5.0), I moved some libraries from installation during general installation to the installation on connector start. Unfortunately, I'm not sure that this action will decrease the size of the gateway to the required size, but you can try to use it.

e60newbie, 2020-07-21:
I will test if I can get run the Gateway.

zbeacon, 2020-08-03:
Hi @e60newbie ,

I removed not required packages from installation, the best way is get the version from the sources, configure it and then install using setup.py . Also I recommend to use the virtual environment for this to get the information about size of the gateway package. 
To decrease required storage place you also can use the "memory" storage.

closed 2020-09-11 by zbeacon