thingsboard/thingsboard-gateway #347: BLE telemetry data via thingsboard-gateway is not updating in thingsboard

perseon, 2020-07-20:
I tried everywhere for a solution but I'm stuck.

My setup is the following:

ESP32 uses BLE GATT NOTIFICATION characteristic to push temperature data via thingsboard gateway into thingsboard. Once the BLE connection is established the first telemetry package is shown in the freshly created device's 'latest telemetry' area. If i turn on gateway debugging I can see further notifications reaching thingboard like this:

**Configuration** 

{
    "name": "BLE Connector",
    "rescanIntervalSeconds": 100,
    "checkIntervalSeconds": 10,
    "scanTimeSeconds": 5,
    "passiveScanMode": true,
    "devices": [
        {
            "name": "Temperature and humidity sensor",
            "MACAddress": "24:62:AB:F3:43:72",
            "telemetry": [
                {
                    "key": "temperature",
                    "method": "notify",
                    "characteristicUUID": "0972EF8C-7613-4075-AD52-756F33D4DA91",
                    "byteFrom": 0,
                    "byteTo": -1
                }
            ],
            "attributes": [
                {
                    "key": "name",
                    "characteristicUUID": "00002A00-0000-1000-8000-00805F9B34FB",
                    "method": "read",
                    "byteFrom": 0,
                    "byteTo": -1
                }
            ],
            "attributeUpdates": [
                {
                    "attributeOnThingsBoard": "sharedName",
                    "characteristicUUID": "00002A00-0000-1000-8000-00805F9B34FB"
                }
            ]
        }
    ]
}

but the temperature won't update, once first value received. 

**Connector name:**
BLE Connector

**This is the log of one package collected from thingsboard when I've enabled debugging:**
```
{"LOGS":"2020-07-20 02:04:19,640 - DEBUG - [ble_connector.py] - ble_connector - 321 - Notification received from device {'device_config': {'name': 'Esp32 v2.2', 'MACAddress': '24:62:AB:F3:43:72', 'telemetry': [{'key': 'temperature', 'method': 'notify', 'characteristicUUID': '0972EF8C-7613-4075-AD52-756F33D4DA91', 'byteFrom': 0, 'byteTo': -1}], 'attributes': [{'key': 'name', 'characteristicUUID': '00002A00-0000-1000-8000-00805F9B34FB', 'method': 'read', 'byteFrom': 0, 'byteTo': -1}], 'attributeUpdates': [{'attributeOnThingsBoard': 'sharedName', 'characteristicUUID': '00002A00-0000-1000-8000-00805F9B34FB'}], 'serverSideRpc': [{'methodRPC': 'rpcMethod1', 'withResponse': True, 'characteristicUUID': '00002A00-0000-1000-8000-00805F9B34FB', 'methodProcessing': 'read'}]}, 'interest_uuid': {'00002A00-0000-1000-8000-00805F9B34FB': [{'section_config': {'key': 'name', 'characteristicUUID': '00002A00-0000-1000-8000-00805F9B34FB', 'method': 'read', 'byteFrom': 0, 'byteTo': -1}, 'type': 'attributes', 'converter': <thingsboard_gateway.connectors.ble.bytes_ble_uplink_converter.BytesBLEUplinkConverter object at 0xb4427eb0>}], '0972EF8C-7613-4075-AD52-756F33D4DA91': [{'section_config': {'key': 'temperature', 'method': 'notify', 'characteristicUUID': '0972EF8C-7613-4075-AD52-756F33D4DA91', 'byteFrom': 0, 'byteTo': -1}, 'type': 'telemetry', 'converter': <thingsboard_gateway.connectors.ble.bytes_ble_uplink_converter.BytesBLEUplinkConverter object at 0xb4427eb0>}]}, 'scanned_device': <bluepy.btle.ScanEntry object at 0xb443a290>, 'is_new_device': False, 'peripheral': <bluepy.btle.Peripheral object at 0xb58f0070>, 'services': {'00001801-0000-1000-8000-00805F9B34FB': {'00002A05-0000-1000-8000-00805F9B34FB': {'characteristic': <bluepy.btle.Characteristic object at 0xb443a210>, 'handle': 2}}, '00001800-0000-1000-8000-00805F9B34FB': {'00002A00-0000-1000-8000-00805F9B34FB': {'characteristic': <bluepy.btle.Characteristic object at 0xb443a270>, 'handle': 21}, '00002A01-0000-1000-8000-00805F9B34FB': {'characteristic': <bluepy.btle.Characteristic object at 0xb443a1d0>, 'handle': 23}, '00002AA6-0000-1000-8000-00805F9B34FB': {'characteristic': <bluepy.btle.Characteristic object at 0xb443a2b0>, 'handle': 25}}, 'AB0828B1-198E-4351-B779-901FA0E0371E': {'0972EF8C-7613-4075-AD52-756F33D4DA91': {'characteristic': <bluepy.btle.Characteristic object at 0xb443a6b0>, 'handle': 41}, '4AC8A682-9736-4E5D-932B-E9B31405049C': {'characteristic': <bluepy.btle.Characteristic object at 0xb443a5f0>, 'handle': 44}}}} handle: 42, data: b'25.00'"} 
```

**This is the ESP32 code used to create BLE notiffications**
https://pastebin.com/NqMfxsK6
Notiffications do update in nRF Conect

**Versions:**
 - OS: Raspbian GNU/Linux 10 on armv7l
 - Thingsboard IoT Gateway version 2.5.0
 - Python 3.7.3

I realise maybe the issue is not in gateway and might not even be an issue, it might be something I've overlooked. Any kind of help much appreciated. I've being wrestling with this the entire weekend.

zbeacon, 2020-07-20:
Hi @perseon 

Thank you for your interest in the ThingsBoard IoT gateway. I will try to repeat the issue and write here.

perseon, 2020-07-20:
Thank you very much for your help. I'll be happy to provide you any more information upon request.

zbeacon, 2020-07-20:
It looks like the gateway receives the incoming data from the device, but by some reasons doesn't send it to the ThingsBoard. Could you set the logging level from INFO to DEBUG in the logs.conf file, repeat the issue and send the logs from folder /var/log/thingsboard-gateway/ to me using email (ibarkov@thingsboard.io)?

vinod405, 2020-08-02:
Hi,
even i am also facing the same issue. i suspect the function below is causing this issue.
def __get_services_and_chars(self):

it is called based on the interval set ie. checkIntervalSeconds in config.
so for the first connection it is fine and able get the data and sending to the server. But for the second time when again it is getting called this function again try to initiate a connect to the device which is already connected and getting hang there itself.

zbeacon, 2020-08-03:
Hi @vinod405, @perseon 

Please try to use the version from the develop/2.4-python branch and let me know about the results.

vinod405, 2020-08-03:
Hi @zbeacon,
yes it is fixed in commit develop/2.4-python. now i am able to see the updated data in server.
Tq

closed 2020-10-09 by zbeacon