thingsboard/thingsboard-gateway #349: Device type of devices always default MQTT connector

jazkiddie, 2020-07-21:
Hello, 

Title says it all, I'm using a mqtt connector and thingsboard gateway, I have checked that my syntax is correct and it is even properly fetched in the corresponding json mqtt uplink converter file as shown in the following screenshots
[image]
and this is what I get when I type `systemctl status thingsboard-gateway.service` on my shell
[image]
If this can't be easily fixed, I put the device type as a client attribute and I would like to know if there is a way to put that as the device type.

zbeacon, 2020-07-23:
Hi @jazkiddie,

Thank for your interest in the ThingsBoard IoT gateway and feedback. This issue caused by first connection of the device. we will fix it.

zbeacon, 2020-07-27:
Hi @jazkiddie,

could you check that the first message from your device contains the device type and you are configure correctly your gateway (deviceTypeJsonExpression or deviceTypeTopicExpression). The gateway cannot change the device type after device was created.

closed 2020-09-01 by zbeacon