thingsboard/thingsboard-gateway #350: Device Limitation

yeduy, 2020-07-23:
Hi,

Can we connect as many devices as we want with the Thingsboard Gateway or is there a device limitation?

Thank you

zbeacon, 2020-07-23:
Hi @yeduy ,

Thank you for your interest in the ThingsBoard IoT gateway. 
The gateway doesn't contain limits for the device, but some ThingsBoard  subscription plans have limits. 
You can read more about subscription plans [here](https://thingsboard.io/pricing/).

yeduy, 2020-07-23:
Hi @zbeacon ,

Thank you for your answer. We are using Thingsboard PE for our project. But for modbus connections we need to use TB Gateway. Actually the question was about performance about gateway. In the website, it is saying "TB Gateway is designed for Device Number < 1000". So we are wondering what happened if use 10.000 device or 1.000.000 device. Is it dependong or RAM or CPU usage?

Thank you

zbeacon, 2020-07-23:
@yeduy 

Maximum connected device in tests are 1000, you can try to use more. If you will try to connect more devices please give feedback.

Thank you.

yeduy, 2020-07-23:
@zbeacon 

Okay, thank you very much

closed 2020-07-23 by yeduy