thingsboard/thingsboard-gateway #351: [BUG] Duplicated telemetry

everardogentil, 2020-07-23:
Is there any reason for thingsboard-gateway sending duplicated telemetry for each modbus device connected to it?

[image]

The data is being sent from device to tb-gateway only once every 10 seconds.

zbeacon, 2020-07-24:
Hi @everardogentil 

Thank you for your report, I will investigate this issue next week.

zbeacon, 2020-07-27:
Hi @everardogentil 

Please attach your configuration file for modbus connector.

everardogentil, 2020-07-27:
@zbeacon ,
here is the config file:

[modbus_tcp.json.txt](https://github.com/thingsboard/thingsboard-gateway/files/4982598/modbus_tcp.json.txt)

zbeacon, 2020-08-03:
Hi @everardogentil ,

I found and fixed the bug, please try to use the version from the master branch and let me know about the results.

everardogentil, 2020-08-04:
Thank you, @zbeacon . I will try.

closed 2020-09-01 by zbeacon