thingsboard/thingsboard-gateway #352: [HELP] Modbus connector has been stopped

everardogentil, 2020-07-23:
After a while trying to connect to modbus devices the tb-gateway raise this information at connector.log.

[image]

Now, even the modbus devices are online the tb-gateway does not connect to them. 

How can I restart the modbus connector?

zbeacon, 2020-07-27:
Hi @everardogentil,

The gateway can connect only to Modbus server, currently. We are going to add ability to connect to Modbus devices in one of the future releases.

everardogentil, 2020-08-28:
I think was not so clear. The problem occurs with connection to masters (in this case energy meters). Even with the meters connected to the network the tb-gw does not connect to them. It is like the connector has freezed and needs some restart to connect again.

zbeacon, 2020-09-15:
Hi @everardogentil ,

Please try to use the latest version of the gateway.

closed 2020-09-25 by zbeacon