thingsboard/thingsboard-gateway #354: [HELP]

meghanajose, 2020-07-29:
I'm facing issues with request connector 
When I run Tb_gateway I'm getting
""2020-07-29 11:40:39" - ERROR - [request_connector.py] - request_connector - 204 - Timeout error on request https://127.0.0.1:39320/iotgateway/read?ids=Channel1.Device1.tag1.json."
I'm using Kepserver to send data. I have used REST server , so data from kepserver is pushed to "https://127.0.0.1:39320/iotgateway/read?ids=Channel1.Device1.tag1" and from there I need to get data to thingsboard.
Data in https://127.0.0.1:39320/iotgateway/read?ids=Channel1.Device1.tag1
{"readResults":[{"id":"Channel1.Device1.tag1","s":true,"r":"","v":87,"t":1596002500129},{"id":"Channel1.Device1.tag2","s":true,"r":"","v":0,"t":1596002414945}]}

**Configuration** (Attach your configuration file)
thingsboard:
  host: demo.thingsboard.io
  port: 1883
  remoteShell: false
  remoteConfiguration: false
  security:
    accessToken: access code
storage:
  type: memory
  read_records_count: 10
  max_records_count: 1000
#  type: file
#  data_folder_path: ./data/
#  max_file_count: 10
#  max_read_records_count: 10
#  max_records_per_file: 10000
connectors:
  -
#   name: MQTT Broker Connector
#  type: mqtt
# configuration: mqtt.json

#  -
#    name: Modbus Connector
#    type: modbus
#    configuration: modbus.json
#
#  -
#    name: Modbus Connector
#    type: modbus
#    configuration: modbus_serial.json
#
#  -
#    name: OPC-UA Connector
#    type: opcua
#    configuration: opcua.json
#
#  -
#    name: BLE Connector
#    type: ble
#    configuration: ble.json
#
#  -
     name: REQUEST Connector
     type: request
     configuration: request.json
#
#  -
#    name: CAN Connector
#    type: can
#    configuration: can.json
#
#  -
#    name: BACnet Connector
#    type: bacnet
#    configuration: bacnet.json
#
#  -
#    name: ODBC Connector
#    type: odbc
#    configuration: odbc.json
#
[... 82 lines cut ...]
""2020-07-29 10:57:39" - ERROR - [request_connector.py] - request_connector - 204 - Timeout error on request https://127.0.0.1:39320/iotgateway/read?ids=Channel1.Device1.tag1.json."
""2020-07-29 10:58:39" - ERROR - [request_connector.py] - request_connector - 204 - Timeout error on request https://127.0.0.1:39320/iotgateway/read?ids=Channel1.Device1.tag1.json."
""2020-07-29 10:59:39" - ERROR - [request_connector.py] - request_connector - 204 - Timeout error on request https://127.0.0.1:39320/iotgateway/read?ids=Channel1.Device1.tag1.json."
""2020-07-29 11:00:39" - ERROR - [request_connector.py] - request_connector - 204 - Timeout error on request https://127.0.0.1:39320/iotgateway/read?ids=Channel1.Device1.tag1.json."
""2020-07-29 11:01:39" - ERROR - [request_connector.py] - request_connector - 204 - Timeout error on request https://127.0.0.1:39320/iotgateway/read?ids=Channel1.Device1.tag1.json."
""2020-07-29 11:02:39" - ERROR - [request_connector.py] - request_connector - 204 - Timeout error on request https://127.0.0.1:39320/iotgateway/read?ids=Channel1.Device1.tag1.json."
""2020-07-29 11:03:39" - ERROR - [request_connector.py] - request_connector - 204 - Timeout error on request https://127.0.0.1:39320/iotgateway/read?ids=Channel1.Device1.tag1.json."
""2020-07-29 11:04:39" - ERROR - [request_connector.py] - request_connector - 204 - Timeout error on request https://127.0.0.1:39320/iotgateway/read?ids=Channel1.Device1.tag1.json."
""2020-07-29 11:05:39" - ERROR - [request_connector.py] - request_connector - 204 - Timeout error on request https://127.0.0.1:39320/iotgateway/read?ids=Channel1.Device1.tag1.json."
""2020-07-29 11:06:39" - ERROR - [request_connector.py] - request_connector - 204 - Timeout error on request https://127.0.0.1:39320/iotgateway/read?ids=Channel1.Device1.tag1.json."
""2020-07-29 11:07:39" - ERROR - [request_connector.py] - request_connector - 204 - Timeout error on request https://127.0.0.1:39320/iotgateway/read?ids=Channel1.Device1.tag1.json."
""2020-07-29 11:08:39" - ERROR - [request_connector.py] - request_connector - 204 - Timeout error on request https://127.0.0.1:39320/iotgateway/read?ids=Channel1.Device1.tag1.json."
""2020-07-29 11:09:39" - ERROR - [request_connector.py] - request_connector - 204 - Timeout error on request https://127.0.0.1:39320/iotgateway/read?ids=Channel1.Device1.tag1.json."
""2020-07-29 11:10:39" - ERROR - [request_connector.py] - request_connector - 204 - Timeout error on request https://127.0.0.1:39320/iotgateway/read?ids=Channel1.Device1.tag1.json."
""2020-07-29 11:11:39" - ERROR - [request_connector.py] - request_connector - 204 - Timeout error on request https://127.0.0.1:39320/iotgateway/read?ids=Channel1.Device1.tag1.json."
""2020-07-29 11:12:39" - ERROR - [request_connector.py] - request_connector - 204 - Timeout error on request https://127.0.0.1:39320/iotgateway/read?ids=Channel1.Device1.tag1.json."
""2020-07-29 11:13:39" - ERROR - [request_connector.py] - request_connector - 204 - Timeout error on request https://127.0.0.1:39320/iotgateway/read?ids=Channel1.Device1.tag1.json."
""2020-07-29 11:14:39" - ERROR - [request_connector.py] - request_connector - 204 - Timeout error on request https://127.0.0.1:39320/iotgateway/read?ids=Channel1.Device1.tag1.json."

**Versions (please complete the following information):**
 - Os: Windows 10 pro
 - Thingsboard IoT Gateway version [e.g. 2.5.0]
 - Python version[e.g. 3.8.3
[service.log](https://github.com/thingsboard/thingsboard-gateway/files/4992975/service.log)
]

zbeacon, 2020-07-29:
Hi @meghanajose ,

Thank you for your interest in the ThingsBoard IoT gateway.
Are you sure, that you have a https on your localhost?
Please try to use http instead of https in the endpoint address and re-run the gateway.

meghanajose, 2020-07-29:
Yes, I noticed now, it was http and then I had Re-run gateway but now the errors are
"2020-07-29 14:29:01" - ERROR - [json_request_uplink_converter.py] - json_request_uplink_converter - 52 - Cannot find the value from the input data"
""2020-07-29 14:29:01" - ERROR - [json_request_uplink_converter.py] - json_request_uplink_converter - 52 - Cannot find the value from the input data"
""2020-07-29 14:29:01" - ERROR - [tb_utility.py] - tb_utility - 62 - No telemetry and attributes in data: {"deviceName": "Test device", "deviceType": "default", "attributes": [], "telemetry": []}"
""2020-07-29 14:29:01" - ERROR - [tb_gateway_service.py] - tb_gateway_service - 314 - Data from REQUEST Connector connector is invalid."

this is my Request.json file
{
  "host": "http://127.0.0.1:39320",
  "SSLVerify": true,
  "security": {
    "type": "anonymous"
  },
  "mapping": [
    {
      "url": "iotgateway/read?ids=Channel1.Device1.tag1",
      "httpMethod": "GET",
      "httpHeaders": {
        "ACCEPT": "application/json"
      },
      "allowRedirects": true,
      "timeout": 0.5,
      "scanPeriod": 60,
      "converter": {
        "type": "json",
        "deviceNameJsonExpression": "device",
        "deviceTypeJsonExpression": "default",
        "attributes": [
          {
            "type": "double",
            "key": "channel1.device1.tag1",
            "value": "${v}"
          }
        ],
        "telemetry": [
          {
            "type": "double",
            "key": "channel1.device1.tag1",
            "value": "${v}"
          }
        ]
      }
    }
  ]
}

zbeacon, 2020-08-03:
Hi @meghanajose,

To solve this error, you should use the JSONPathExpression for the telemetry or attributes value parameters, because you have an array in the response for example for serialNumber key you can use the following configuration:

```
"key": "serialNumber",
"type": "string",
"value": "${$.readResults[0].id}"
```

meghanajose, 2020-08-03:
Hi @zbeacon , 
Thank you so much for correcting  the error.  After making the changes  I'm receiving data in telemetry.

zbeacon, 2020-08-03:
I'm glad to hear this. please let me know, if you will have any questions about the gateway.
Also, I close this issue.

closed 2020-08-03 by zbeacon

meghanajose, 2020-08-03:
Yeah, If I do have later on I'll do ask. Thank you